facebookresearch/habitat-sim #1818: Habitat_sim installs failing in Colab

srkiyengar, 2022-07-19:
Recently, habitat_sim  colab installs seem to fail in "import habitat-sim" after successful installtion. It seems to be a complaining about missing library:
ImportError: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /usr/local/lib/python3.7/dist-packages/scipy/optimize/_group_columns.cpython-37m-x86_64-linux-gnu.so)

The two colab habitat-sim install that I have tried are,
1) https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/main/examples/tutorials/colabs/ECCV_2020_Interactivity.ipynb

2) !curl -L https://raw.githubusercontent.com/facebookresearch/habitat-sim/master/examples/colab_utils/colab_install.sh | NIGHTLY=false bash -s

**Error details while importing habitat-sim in  [ECCV_2020_Interactivity.ipynb](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/main/examples/tutorials/colabs/ECCV_2020_Interactivity.ipynb)**

ImportError                               Traceback (most recent call last)
[<ipython-input-2-cc10888dea5c>](https://localhost:8080/#) in <module>()
     17 from PIL import Image
     18 
---> 19 import habitat_sim
     20 from habitat_sim.utils import common as ut
     21 from habitat_sim.utils import viz_utils as vut

11 frames
[/usr/local/lib/python3.7/dist-packages/scipy/optimize/_numdiff.py](https://localhost:8080/#) in <module>()
      6 from scipy.sparse.linalg import LinearOperator
      7 from ..sparse import issparse, csc_matrix, csr_matrix, coo_matrix, find
----> 8 from ._group_columns import group_dense, group_sparse
      9 
     10 

ImportError: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /usr/local/lib/python3.7/dist-packages/scipy/optimize/_group_columns.cpython-37m-x86_64-linux-gnu.so)

**Error details of "import habitat-sim"  after install from "https://raw.githubusercontent.com/facebookresearch/habitat-sim/master/examples/colab_utils/colab_install.sh | NIGHTLY=false bash -s"**

[/content/oreo-pybullet/agent_oreo.py](https://localhost:8080/#) in <module>()
----> 1 import habitat_sim
      2 import numpy as np
      3 import quaternion
      4 import math
      5 import cv2

[/usr/local/lib/python3.7/dist-packages/habitat_sim/__init__.py](https://localhost:8080/#) in <module>()
     11 if not getattr(builtins, "__HSIM_SETUP__", False):
     12     # TODO: kept only for compatibiliy with existing code. Please gradually remove
---> 13     from habitat_sim import (
     14         agent,
     15         attributes,

[/usr/local/lib/python3.7/dist-packages/habitat_sim/agent/__init__.py](https://localhost:8080/#) in <module>()
      7 # type: ignore
      8 
----> 9 from .agent import *  # noqa: F403
     10 from .controls import *  # noqa: F403
     11 

[/usr/local/lib/python3.7/dist-packages/habitat_sim/agent/agent.py](https://localhost:8080/#) in <module>()
     29 )
     30 
---> 31 from .controls import ActuationSpec, ObjectControls
     32 
     33 __all__ = ["ActionSpec", "SixDOFPose", "AgentState", "AgentConfiguration", "Agent"]

[/usr/local/lib/python3.7/dist-packages/habitat_sim/agent/controls/__init__.py](https://localhost:8080/#) in <module>()
      8 from habitat_sim.agent.controls.default_controls import *  # noqa: F401, F403
      9 from habitat_sim.agent.controls.object_controls import ObjectControls
---> 10 from habitat_sim.agent.controls.pyrobot_noisy_controls import PyRobotNoisyActuationSpec
     11 
     12 __all__ = [

[/usr/local/lib/python3.7/dist-packages/habitat_sim/agent/controls/pyrobot_noisy_controls.py](https://localhost:8080/#) in <module>()
     18 import magnum as mn
     19 import numpy as np
---> 20 import scipy.stats
     21 from attr import Attribute
     22 from numpy import ndarray

[/usr/local/lib/python3.7/dist-packages/scipy/stats/__init__.py](https://localhost:8080/#) in <module>()
    439 """
    440 
--> 441 from .stats import *
    442 from .distributions import *
    443 from .morestats import *

[/usr/local/lib/python3.7/dist-packages/scipy/stats/stats.py](https://localhost:8080/#) in <module>()
     41 import scipy.special as special
     42 from scipy import linalg
---> 43 from . import distributions
     44 from . import mstats_basic
     45 from ._stats_mstats_common import (_find_repeats, linregress, theilslopes,

[/usr/local/lib/python3.7/dist-packages/scipy/stats/distributions.py](https://localhost:8080/#) in <module>()
      6 #       instead of `git blame -Lxxx,+x`.
      7 #
----> 8 from ._distn_infrastructure import (rv_discrete, rv_continuous, rv_frozen)
      9 
     10 from . import _continuous_distns

[/usr/local/lib/python3.7/dist-packages/scipy/stats/_distn_infrastructure.py](https://localhost:8080/#) in <module>()
     22 # for root finding for continuous distribution ppf, and max likelihood
     23 # estimation
---> 24 from scipy import optimize
     25 
     26 # for functions of continuous distributions (e.g. moments, entropy, cdf)

[/usr/local/lib/python3.7/dist-packages/scipy/optimize/__init__.py](https://localhost:8080/#) in <module>()
    398 """
    399 
--> 400 from .optimize import *
    401 from ._minimize import *
    402 from ._root import *

[/usr/local/lib/python3.7/dist-packages/scipy/optimize/optimize.py](https://localhost:8080/#) in <module>()
     34                          line_search_wolfe2 as line_search,
     35                          LineSearchWarning)
---> 36 from ._numdiff import approx_derivative
     37 from scipy._lib._util import getfullargspec_no_self as _getfullargspec
     38 from scipy._lib._util import MapWrapper

[/usr/local/lib/python3.7/dist-packages/scipy/optimize/_numdiff.py](https://localhost:8080/#) in <module>()
      6 from scipy.sparse.linalg import LinearOperator
      7 from ..sparse import issparse, csc_matrix, csr_matrix, coo_matrix, find
----> 8 from ._group_columns import group_dense, group_sparse
      9 
     10 

ImportError: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /usr/local/lib/python3.7/dist-packages/scipy/optimize/_group_columns.cpython-37m-x86_64-linux-gnu.so)

Hopefully the details are sufficient to troubleshoot the colab environment working.The installs used to work nicely the last time I checked 3-4 weeks ago.
Thanks.

dhruvbatra, 2022-07-19:
@Skylion007 -- thoughts?

srkiyengar, 2022-07-22:
Strings command of /usr/lib/x86_64-linux-gnu/libstdc++.so.6 for the colab install.
[image]

arjunmajum, 2022-07-22:
Here's temporary fix that worked for me from the second solution [here](https://stackoverflow.com/a/65494217).

Run this one time:

```
!add-apt-repository ppa:ubuntu-toolchain-r/test -y
!apt-get update --fix-missing
!apt-get dist-upgrade
```

Then, restart the runtime -- or -- modify the imports to automatically restart:

```
try:
  import habitat_sim
except ImportError:
  exit()
```

srkiyengar, 2022-07-23:
Thank you @arjunmajum.

Skylion007, 2022-07-23:
> @Skylion007 -- thoughts?

Trying to find a long term fix here that doesn't require a Colab restart. I am really confused where the new LIBCXX is being installed / overwritten from.

srkiyengar, 2022-08-02:
@arjunmajum's suggestion of restart worked well and was very usable. But as of last evening, executing the install script to install habitat-sim is running into another issue. 

PackagesNotFoundError: The following packages are not available from current channels:

  - numba=0.56.0

Any suggestion or work around will be appreciated.

-----------
!curl -L https://raw.githubusercontent.com/facebookresearch/habitat-sim/master/examples/colab_utils/colab_install.sh | NIGHTLY=false bash -s

--2022-08-02 13:43:47--  https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh
Resolving repo.continuum.io (repo.continuum.io)... 104.18.200.79, 104.18.201.79, 2606:4700::6812:c94f, ...
Connecting to repo.continuum.io (repo.continuum.io)|104.18.200.79|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh [following]
--2022-08-02 13:43:47--  https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh
Resolving repo.anaconda.com (repo.anaconda.com)... 104.16.130.3, 104.16.131.3, 2606:4700::6810:8203, ...
Connecting to repo.anaconda.com (repo.anaconda.com)|104.16.130.3|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 76607678 (73M) [application/x-sh]
Saving to: ‘Miniconda3-latest-Linux-x86_64.sh’

Miniconda3-latest-L 100%[===================>]  73.06M   234MB/s    in 0.3s    

2022-08-02 13:43:48 (234 MB/s) - ‘Miniconda3-latest-Linux-x86_64.sh’ saved [76607678/76607678]

PREFIX=/usr/local
Unpacking payload ...
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /usr/local

  added / updated specs:
    - _libgcc_mutex==0.1=main
    - _openmp_mutex==4.5=1_gnu
    - brotlipy==0.7.0=py39h27cfd23_1003
    - ca-certificates==2022.3.29=h06a4308_1
    - certifi==2021.10.8=py39h06a4308_2
    - cffi==1.15.0=py39hd667e15_1
    - charset-normalizer==2.0.4=pyhd3eb1b0_0
    - colorama==0.4.4=pyhd3eb1b0_0
    - conda-content-trust==0.1.1=pyhd3eb1b0_0
    - conda-package-handling==1.8.1=py39h7f8727e_0
    - conda==4.12.0=py39h06a4308_0
    - cryptography==36.0.0=py39h9ce1e76_0
    - idna==3.3=pyhd3eb1b0_0
    - ld_impl_linux-64==2.35.1=h7274673_9
    - libffi==3.3=he6710b0_2
    - libgcc-ng==9.3.0=h5101ec6_17
    - libgomp==9.3.0=h5101ec6_17
    - libstdcxx-ng==9.3.0=hd4cf53a_17
    - ncurses==6.3=h7f8727e_2
    - openssl==1.1.1n=h7f8727e_0
    - pip==21.2.4=py39h06a4308_0
    - pycosat==0.6.3=py39h27cfd23_0
    - pycparser==2.21=pyhd3eb1b0_0
    - pyopenssl==22.0.0=pyhd3eb1b0_0
    - pysocks==1.7.1=py39h06a4308_0
    - python==3.9.12=h12debd9_0
    - readline==8.1.2=h7f8727e_1
    - requests==2.27.1=pyhd3eb1b0_0
    - ruamel_yaml==0.15.100=py39h27cfd23_0
    - setuptools==61.2.0=py39h06a4308_0
    - six==1.16.0=pyhd3eb1b0_1
    - sqlite==3.38.2=hc218d9a_0
    - tk==8.6.11=h1ccaba5_0
[... 65 lines cut ...]
PackagesNotFoundError: The following packages are not available from current channels:

  - numba=0.56.0

Current channels:

  - https://conda.anaconda.org/aihabitat/linux-64
  - https://conda.anaconda.org/aihabitat/noarch
  - https://conda.anaconda.org/conda-forge/linux-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.anaconda.com/pkgs/main/linux-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/linux-64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org/

and use the search bar at the top of the page.

An error occured during the installation of Habitat-sim or Habitat-Lab.

-------------------------

pkachana3, 2022-08-04:
@srkiyengar Were you able to get this installation issue resolved? I am having the same problem as well.

srkiyengar, 2022-08-06:
No @pkachana3.

aclegg3, 2022-08-12:
Thanks for the temproary solutions here, folks! With these I'm able to run the colabs successfully. @Skylion007 and I will work on a real fix.

For now, you can:
1.  modify the install cell to use the installer on [a branch](https://github.com/facebookresearch/habitat-sim/tree/colab-numba-version) with the change suggested by @srkiyengar:
```
https://raw.githubusercontent.com/facebookresearch/habitat-sim/d36c1de8723606245290eb4da271b5582395da8a/examples/colab_utils/colab_install.sh
```

2. apply the fix suggested by @arjunmajum in [this comment](https://github.com/facebookresearch/habitat-sim/issues/1818#issuecomment-1192960577):

> Run this one time:
>```
>!add-apt-repository ppa:ubuntu-toolchain-r/test -y
>!apt-get update --fix-missing
>!apt-get dist-upgrade
>```
>Then, restart the runtime

rpartsey, 2022-12-15:
@aclegg3 thank you for advice! 
works for me

aclegg3, 2024-01-16:
Note we have stopped supporting colab and converted tutorials to local Jupyter notebooks instead.

closed 2024-01-16 by aclegg3