facebookresearch/habitat-sim #1822: How should I get one object in the camera's perspective without considering other objects to block the projection area?

DaLi-Jack, 2022-07-30:
## Habitat-Sim version
v0.2.2

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/
Yes

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/
Yes

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
In my work I need to get the proportion of an object that is occluded.
Take the following picture as an example, in the picture, the monkey head is occluded by the cuboid in front.
I want to get the object projection with orange outline
<div align=center>
[image]
</div>
But in **Semantic Sensor**, I can only get the projection after removing the red line part in the picture below (the red line part is blocked by the front cuboid), as shown below
<div align=center>
[image]
</div>
Any help would be appreciated.Thanks a lot!

aclegg3, 2022-08-01:
Hey @DaLi-Jack,

We don't support this use case natively, but I have a couple of ideas for you. Which one you choose will probably depend on your specific use case, performance requirements, etc...

1. You could move both the object and camera very far from the scene with a uniform translation (e.g. 1000 meters up the X axis) and then render. I may not recommended this during active physics simulation, but for static states it would be fine.

2. You could save the camera and object state, load a new scene with an empty background (scene_id="NONE"), replace the object and set the camera state before rendering. This is more of an offline approach if you are generating a dataset.

3. You could hack the code a bit to create a custom [SceneNodeType](https://aihabitat.org/docs/habitat-sim/habitat_sim.scene.html#SceneNodeType) and set it for your object's visual SceneNodes. Then create a custom sensor rendering routine which culls all objects without that type before rendering. This would be the most performant solution if you need to use the feature during RL and physical simulation, but would be the most costly to implement.

DaLi-Jack, 2022-08-02:
Thank you very much for your detailed answer！
The 2nd method may best suit my needs.Also thanks for the other two methods, they may help other users !
I have another question to ask.
**How to save and load the state of the camera conveniently？** 
Any help would be appreciated.
Thanks again for your answer.

closed 2022-08-02 by aclegg3

DaLi-Jack, 2022-08-02:
Thank you very much ！