facebookresearch/habitat-sim #1823: python setup.py install   returned non-zero exit status 1.

WWenMing, 2022-08-02:
**1. environment**
python version = 3.8
cmake = 3.14.0
Habitat-sim has cloned,and pip install -r requirements.txt has done

**2. here is the log with bug:**
setup.py:459: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  assert StrictVersion(
setup.py:461: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  ) >= StrictVersion("3.7"), "Must use python3.7 or newer"
running install
/home/sim/.conda/envs/habitat3.8/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/home/sim/.conda/envs/habitat3.8/lib/python3.8/site-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running bdist_egg
running egg_info
writing src_python/habitat_sim.egg-info/PKG-INFO
writing dependency_links to src_python/habitat_sim.egg-info/dependency_links.txt
writing requirements to src_python/habitat_sim.egg-info/requires.txt
writing top-level names to src_python/habitat_sim.egg-info/top_level.txt
reading manifest file 'src_python/habitat_sim.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
adding license file 'LICENSE'
writing manifest file 'src_python/habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build/lib.linux-x86_64-cpython-38
creating build/lib.linux-x86_64-cpython-38/habitat_sim
copying src_python/habitat_sim/metadata.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying src_python/habitat_sim/__init__.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying src_python/habitat_sim/errors.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying src_python/habitat_sim/attributes.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying src_python/habitat_sim/registry.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying src_python/habitat_sim/scene.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying src_python/habitat_sim/physics.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying src_python/habitat_sim/geo.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying src_python/habitat_sim/sensor.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying src_python/habitat_sim/simulator.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying src_python/habitat_sim/logging.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying src_python/habitat_sim/attributes_managers.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying src_python/habitat_sim/sim.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying src_python/habitat_sim/gfx.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
creating build/lib.linux-x86_64-cpython-38/habitat_sim/sensors
copying src_python/habitat_sim/sensors/sensor_suite.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/sensors
copying src_python/habitat_sim/sensors/__init__.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/sensors
creating build/lib.linux-x86_64-cpython-38/habitat_sim/bindings
copying src_python/habitat_sim/bindings/__init__.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/bindings
creating build/lib.linux-x86_64-cpython-38/habitat_sim/utils
copying src_python/habitat_sim/utils/viz_utils.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/utils
copying src_python/habitat_sim/utils/__init__.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/utils
copying src_python/habitat_sim/utils/datasets_download.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/utils
copying src_python/habitat_sim/utils/validators.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/utils
copying src_python/habitat_sim/utils/compare_profiles.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/utils
copying src_python/habitat_sim/utils/common.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/utils
copying src_python/habitat_sim/utils/manager_utils.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/utils
copying src_python/habitat_sim/utils/gfx_replay_utils.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/utils
copying src_python/habitat_sim/utils/collect_env.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/utils
copying src_python/habitat_sim/utils/profiling_utils.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/utils
creating build/lib.linux-x86_64-cpython-38/habitat_sim/agent
copying src_python/habitat_sim/agent/agent.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/agent
copying src_python/habitat_sim/agent/__init__.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/agent
creating build/lib.linux-x86_64-cpython-38/habitat_sim/nav
copying src_python/habitat_sim/nav/__init__.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/nav
copying src_python/habitat_sim/nav/greedy_geodesic_follower.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/nav
creating build/lib.linux-x86_64-cpython-38/habitat_sim/robots
copying src_python/habitat_sim/robots/__init__.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/robots
copying src_python/habitat_sim/robots/fetch_robot.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/robots
copying src_python/habitat_sim/robots/mobile_manipulator.py -> build/lib.linux-x86_64-cpython-38/habitat_sim/robots
[... 583 lines cut ...]
    self.distribution.run_command(command)
  File "/home/sim/.conda/envs/habitat3.8/lib/python3.8/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/home/sim/.conda/envs/habitat3.8/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 992, in run_command
    cmd_obj.run()
  File "/home/sim/.conda/envs/habitat3.8/lib/python3.8/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/home/sim/.conda/envs/habitat3.8/lib/python3.8/site-packages/setuptools/_distutils/command/install_lib.py", line 112, in build
    self.run_command('build_ext')
  File "/home/sim/.conda/envs/habitat3.8/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 319, in run_command
    self.distribution.run_command(command)
  File "/home/sim/.conda/envs/habitat3.8/lib/python3.8/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/home/sim/.conda/envs/habitat3.8/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 992, in run_command
    cmd_obj.run()
  File "setup.py", line 271, in run
    self.build_extension(ext)
  File "setup.py", line 372, in build_extension
    subprocess.check_call(
  File "/home/sim/.conda/envs/habitat3.8/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/media/Gibson3.8/habitat-sim/build/lib.linux-x86_64-cpython-38/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/home/sim/.conda/envs/habitat3.8/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DREL_BUILD_RPATH=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-DBUILD_GUI_VIEWERS=OFF', '-DTARGET_HEADLESS=ON', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBUILD_WITH_VHACD=OFF', '-DBUILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=OFF', '-DBUILD_BASIS_COMPRESSOR=OFF', '-DBUILD_WITH_AUDIO=OFF', '/media/Gibson3.8/habitat-sim/src']' returned non-zero exit status 1.

**3.question**
I don't know how to deal with the problem, and I generally hope you can give me some advices , thanks a lot!

aclegg3, 2022-08-02:
Hey @WWenMing,

Looks like CMAKE is unable to find GLFW libraries on your system. Can you install with `--headless` successfully? If so, you may have some stale EGL headers. See this possibly related issue #499.

WWenMing, 2022-08-03:
I have with --headless ，but the log is the same

aclegg3, 2022-08-03:
Maybe you can provide some system information? Looks like you're on Linux of some kind, I guess it could be that your system isn't supported or has some quirks.

I see issues finding the Recast and Detour dependencies:
```
CMake Error at cmake/dependencies.cmake:109 (add_subdirectory):
add_subdirectory given source
"/media/Gibson3.8/habitat-sim/src/cmake/../deps/recastnavigation/Recast"
which is not an existing directory.
...
CMake Error at cmake/dependencies.cmake:110 (add_subdirectory):
add_subdirectory given source
"/media/Gibson3.8/habitat-sim/src/cmake/../deps/recastnavigation/Detour"
which is not an existing directory.
```
I also see issues with system install of required graphics libraries:
```
-- Could NOT find GLFW (missing: GLFW_LIBRARY GLFW_INCLUDE_DIR)
-- Found SDL2: /usr/lib/x86_64-linux-gnu/libSDL2-2.0.so
-- Could NOT find Magnum (missing: GlfwApplication Sdl2Application)
```

I think these are the primary issues you'd need to address.

aclegg3, 2022-08-31:
Hey @WWenMing, haven't heard from you in awhile. I'll close the issue as stale, but you can re-open if you'd like to pursue further support.

closed 2022-08-31 by aclegg3