facebookresearch/habitat-sim #1824: no example/viewer.py

WWenMing, 2022-08-03:
I have done :

conda create -n habitat python=3.7 cmake=3.14.0
conda activate habitat
conda install habitat-sim withbullet -c conda-forge -c aihabitat

My problem is the habitat-sim which installed by conda is where , "path/to/habitat" --> what means the "path"
and I find one hatibta-sim in "/home/.conda/envs/habi/lib/python3.7/site-packets",but it doesn't include the folder "example",so I can't find the example/viewer.py to test
[image]

I appreciately you can help me , thank you .

aclegg3, 2022-08-03:
Hey @WWenMing, if you want the examples you'll need to clone the repo and run them from that directory.
You can run the examples from a conda install, but as you noted they don't come with the conda package.

closed 2022-08-12 by aclegg3

dwei-k, 2022-10-24:
hello, @aclegg3 
when I run the `python examples/viewer.py --scene Data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
` I met 
Traceback (most recent call last):
  File "examples/viewer.py", line 19, in <module>
    from magnum import shaders, text
ImportError: cannot import name 'text' from 'magnum'

jemo-mh, 2023-03-29:
> hello, @aclegg3 when I run the `python examples/viewer.py --scene Data/scene_datasets/habitat-test-scenes/skokloster-castle.glb ` I met Traceback (most recent call last): File "examples/viewer.py", line 19, in from magnum import shaders, text ImportError: cannot import name 'text' from 'magnum'

How did you solve this problem?

dwei-k, 2023-03-29:
@jemo-mh hi, it has been a long time for me, I remeber I didn't clone the stable tag of this repo at that time, when I clone the right branch, it was solved.

jemo-mh, 2023-03-29:
> @jemo-mh hi, it has been a long time for me, I remeber I didn't clone the stable tag of this repo at that time, when I clone the right branch, it was solved.

Thanks for your help! I will try it.
Which method did you install habitat-sim??
Did you use pip install?,  Anaconda? or setup.py method?

Youmna-H, 2023-09-04:
Any updates on how to solve the  ImportError: cannot import name 'text' from 'magnum' error? I am facing the same issue.