facebookresearch/habitat-sim #1841: New Features for the Python based Object Visualizer

ykarmesh, 2022-08-28:
## 🚀 Feature
John Reyna recently created the python based object visualizer which lies on the [viewer-py-object-qa-application](https://github.com/facebookresearch/habitat-sim/tree/viewer-py-object-qa-application) branch. The visualizer will be used to find out issues in the object models. Before we start using it on a larger scale, I have the following feature request: 
- Visualize the bounding box of the object
- Add a small display on the top left of the visualizer that:
    - name of the object
    - sim step time w/ physics
    - process memory before/after importing a new object.

## Motivation

Adding these features will help us in reducing the time required to filter out the objects. The earlier we catch issues in the objects, the faster we will be able to use them in Habitat.

ykarmesh, 2022-08-28:
CC:  @aclegg3 @ASzot

jrreyna, 2022-08-28:
For this part:
"(1) Add a small display on the top left of the visualizer that:
 (2) name of the object
 (3) sim step time w/ physics
 (4) process memory before/after importing a new object."

For point (1), As far as I know, and what Alex said, was that displaying text in the application window isn't a feature yet for our python code, just C++. would it be ok to just print that info in the terminal, or is this a dealbreaker?

For point (3), what do you mean by this? Just how long the last frame took to process and render? Or JUST the physics step, and not the rendering?

And for point (4), what do you mean by "process memory?" Do you mean the amount of CPU and RAM memory that is currently being used in the process? Or GPU memory as well?

ykarmesh, 2022-08-29:
> would it be ok to just print that info in the terminal, or is this a dealbreaker
I think we would be adding further information to the visualizer, some of which will be changing as we simulate (for eg: sim step time), so publishing this information in the terminal isnt ideal. How long do you think it would take to create this new module?

> what do you mean by this? Just how long the last frame took to process and render? Or JUST the physics step, and not the rendering?
Having both the rendering time and the physics step time displayed independently would be ideal. If possible, it would be great if the color of the text could change from green to red when the stepping time passed a certain threshold. For starters you could use 10ms as the threshold.

> what do you mean by "process memory?"
@ASzot @aclegg3 Can you chime in here? What did you wanted to see with this?

jrreyna, 2022-08-29:
I'm not sure how long it would take to write code that writes to the application window. This is the first time I have worked with python, so I'd have to really dig into how the magnum Application module works.

As for process memory. I was able to calculate the GPU memory usage using the nvidia_smi library. An example output is:
=====================================================
GPU Info
=====================================================
Device 0: b'NVIDIA GeForce RTX 2070 with Max-Q Design'
Memory Info: 81.45% free
8,589,934,592 (total)
6,996,754,432 (free)
1,593,180,160 (used)

Is that the kind of thing you would want? I also wrote something to get the CPU and RAM memory usage, but I'm not sure it is right

jrreyna, 2022-08-29:
I was able to print out the CPU usage. This is an example print statement:

CPU Memory
    CPU memory usage: 23.20%
CPU Stats  
    Context switches: 554,137,115
    Interrupts: 404,366,682
    Software interrupts: 194,285,948
    System calls: 0
CPU Frequency
    Current frequency: 2,704.47 MHz
    Min frequency: 800.0 MHz
    Max frequency: 4,500.0 MHz

I also was able to draw the bounding boxes. Press 'k' to toggle on and off. you were right, some of them are not axis aligned

aclegg3, 2022-08-29:
@mosra can comment on writing text to the python window here. I understand there are a couple of technical hurdles, maybe related to sharing the render buffer or glContext with the text writer. Likely this would need to be worked on from the Magnum side.

aclegg3, 2022-08-29:
@jrreyna Some of this is already implemented in a tutorial script. Check out [examples/tutorials/physics_benchmarking.py](https://github.com/facebookresearch/habitat-sim/blob/main/examples/tutorials/physics_benchmarking.py).

aclegg3, 2022-08-29:
I added an issue for the GUI text specifically as it has broader scope than this application. #1842

aclegg3, 2022-08-29:
> @aclegg3 Regarding the GPU memory usage - In Macbooks, does Habitat use the Intel graphics card? Currently John is using `nvidia_smi` library in his PR and it gave me an error on my macbook.

Yeah, integrated graphics on mac. @jrreyna we should have a fallback if the user does not have nvidia hardware.

mosra, 2022-08-29:
Re text -- answered on #1842.

Re GPU memory usage / `nvidia_smi` -- this feels like something that should be directly inside Magnum itself, so you don't need to worry about portability. I'll see what can be done.

jrreyna, 2022-08-29:
@mosra 
"Re text -- answered on https://github.com/facebookresearch/habitat-sim/issues/1842.

Re GPU memory usage / nvidia_smi -- this feels like something that should be directly inside Magnum itself, so you don't need to worry about portability (and GL vs Vulkan, once we are there). I'll see what can be done."

So I should just focus on CPU and RAM memory usage?