facebookresearch/habitat-sim #1842: GUI Text for viewer.py

aclegg3, 2022-08-29:
## 🚀 Feature
Add a Text GUI interface to viewer.py.

## Motivation

Currently, we use `Magnum::ImGuiIntegration` to render text to the window for viewer.cpp (C++ viewer application). We'd like to have a similar interface for adding text to frames in viewer.py (Python viewer application)

## Additional context

At some point we (@mosra) discussed this and determined there may be some blockers implementing this feature in Python. If so, let's record them here for tracking.

mosra, 2022-08-29:
I don't remember any blockers for implementing this, really. Except that I'd have to expose a few more Magnum APIs to Python, which is nothing complex on its own. I could do that.

Back then we used ImGui because it allowed to have a tiny FPS counter done with a single line of code ... but at the cost of a massive new dependency, of which we use not even 1%. Because ImGui is a large library on its own, exposing all of it to Python is not simple. There's [PyImGui](https://pyimgui.readthedocs.io/en/latest/) but for that I'd first need to make it work with the Magnum backend, and that's currently an unknown amount of work. Instead, I'd really like to use the far more lightweight `Magnum::Text` library, for both C++ and Python.

aclegg3, 2022-08-29:
> I don't remember any blockers for implementing this, really. Except that I'd have to expose a few more Magnum APIs to Python, which is nothing complex on its own. I could do that.

That would be great. If the investment is small, I'd like to get this working.

> I'd really like to use the far more lightweight Magnum::Text library, for both C++ and Python.

That sounds fine to me. Remove a dependency and have a better mirror between the C++ and python implementations. Is that ready to use for C++? Looks like we'd need bindings to support python.

mosra, 2022-08-29:
Yes, that's what I meant -- the `Text` library is ready to be used from C++, it just needs to have its essential functionality exposed to Python. I'll reach out to you on Slack to discuss priorities.

closed 2022-09-27 by jrreyna