facebookresearch/habitat-sim #1851: Questions about GreedyGeodesicFollower

yaofeng11, 2022-09-02:
Hi there:
 I wonder that if greedygeodescifollower as a planner that take all actions to step or just a predictor.
I mean that
`agent_1_state = habitat_sim.AgentState()
agent_1_state.position = np.array([0.5, 0.0, 1])
arr = np.array([8.0, 0.0, 4.0])

follower = GreedyGeodesicFollower(pathfinder=sim.pathfinder, agent=agent_1)
path = follower.find_path(goal_pos=arr)
for item in path:
    print(item)
    observations_1 = sim.step({1: item})
    cv2.imshow("RGB1", transform_rgb_bgr(observations_1[0]["color_sensor"]))
    observations_1_list.append(transform_rgb_bgr(observations_1[0]["color_sensor"]))
    time.sleep(2)`
All I need is to given two points, the start point and end point. And I just want the frames after each step was taken. So I try the code I show. The path can be successfully print out, which I mean that is the agent is already get the final goal ? or the action list path is just a predict list ? How should I gain the frames after each step is taken  when using greedygeodescifollower?

aclegg3, 2022-09-02:
Hey @yaofeng11,

Yes, the path is produced in advance and contains valid agent states. You will need to place the agent at each state to get an image. 

There is a complete tutorial of this process with example code in the "Pathfinding Queries on NavMesh" subsection of the "Habitat-Sim Basics for Navigation" segment of our [ECCV 2020 Tutorial Series](https://aihabitat.org/tutorial/2020/). Feel free to borrow the code from there.

closed 2022-09-02 by aclegg3

yaofeng11, 2022-09-02:
Hi @aclegg3 ,
Thanks for your quick apply.I know that "Pathfinding Queries on NavMesh", but what I need is every frame after each step.The agent may take action like turn_left, turn_right, it does cause frame update not like place agent at each state. Thats why I use greedygeodescifollower.
Besides, I run the number I show ,the cv2.imshow shows nothing

aclegg3, 2022-09-02:
I see, so you want to use the greedy follower to get to observation at every frame. Sure, then you want to iteratively call [GreedyGeodesicFollower.next_action_along](https://aihabitat.org/docs/habitat-sim/habitat_sim.nav.GreedyGeodesicFollower.html#next_action_along) and call `step(action)` for the actions it returns.

You can see an example of this in the unit test `tests/test_greedy_follower.py`