facebookresearch/habitat-sim #1852: How to add new objects?

yaofeng11, 2022-09-02:
Hi there:
I see the example you show in colab. And here is my code
`obj_attr_mgr = sim.get_object_template_manager()
obj_attr_mgr.load_configs(str(os.path.join(data_path, "objects/example_objects")))
obj_attr_mgr.load_configs(str(os.path.join(data_path, "objects/locobot_merged")))
prim_attr_mgr = sim.get_asset_template_manager()
stage_attr_mgr = sim.get_stage_template_manager()
# Manager providing access to rigid objects
rigid_obj_mgr = sim.get_rigid_object_manager()
file_obj_handles = obj_attr_mgr.get_file_template_handles()
prim_obj_handles = obj_attr_mgr.get_synth_template_handles()
prim_asset_handles = prim_attr_mgr.get_template_handles()
sel_file_obj_handle = file_obj_handles[0]
sel_prim_obj_handle = prim_obj_handles[0]
sel_asset_handle = prim_asset_handles[0]
obj_template_handle = sel_file_obj_handle
asset_tempalte_handle = sel_asset_handle
object_type = "File-based"  # @param ["File-based","Primitive-based"]
obj_template_handle = sel_file_obj_handle
offset_x = 0.5  # @param {type:"slider", min:-2, max:2, step:0.1}
offset_y = 1.4  # @param {type:"slider", min:0, max:3.0, step:0.1}
offset_z = -1.5  # @param {type:"slider", min:-3, max:0, step:0.1}
offset = np.array([offset_x, offset_y, offset_z])
orientation_x = 0  # @param {type:"slider", min:-180, max:180, step:1}
orientation_y = 0  # @param {type:"slider", min:-180, max:180, step:1}
orientation_z = 0  # @param {type:"slider", min:-180, max:180, step:1}
rotation_x = mn.Quaternion.rotation(mn.Deg(orientation_x), mn.Vector3(1.0, 0, 0))
rotation_y = mn.Quaternion.rotation(mn.Deg(orientation_y), mn.Vector3(1.0, 0, 0))
rotation_z = mn.Quaternion.rotation(mn.Deg(orientation_z), mn.Vector3(1.0, 0, 0))
orientation = rotation_z * rotation_y * rotation_x
obj_1 = rigid_obj_mgr.add_object_by_template_handle(obj_template_handle)
`
The offset location is from agent body node.So my question is as for search task, I need to put the objects for example I need to put the banana(maybe cellphone) under the bed, and I know the beds location, so what should I do for the offset location? I mean is there exist any globe location to set the objects where is not from agent body node?

closed 2022-09-02 by aclegg3

yaofeng11, 2022-09-02:
Hi @aclegg3 
Sorry for my too many questions.I checked the link
`obj_templates_mgr = sim.get_object_template_manager()
banana_template_id = obj_templates_mgr.load_configs(str(os.path.join(data_path, "test_assets/objects/banana")))[0]
rigid_obj_mgr = sim.get_rigid_object_manager()
banana_obj = rigid_obj_mgr.add_object_by_template_id(banana_template_id)
banana_obj.translation = [0.5, 0.0, 1.0]`
I just download the file at your link and put it in the path.And it says
    banana_template_id = obj_templates_mgr.load_configs(str(os.path.join(data_path, "test_assets/objects/banana")))[0]
IndexError: list index out of range
I dont know why

aclegg3, 2022-09-02:
This indicates that you don't have the assets. 
`python -m habitat_sim.utils.datasets_download --uids habitat_example_objects`