facebookresearch/habitat-sim #1854: Error when building from source with

doudoudou1999, 2022-09-06:
## Habitat-Sim version
vx.x.x or main?

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? The bug you are about to report may already be fixed in the latest version.

Main branch contains 'bleeding edge' code, but we do appreciate bug reports for it!

## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:

1.
2.
3.

 Please note that without a minimal working example to reproduce the bug, we may not be able to help you.

## Expected behavior

## Additional context

## System Info

Please copy and paste the output from the
[environment collection script](https://github.com/facebookresearch/habitat-sim/tree/main/habitat_sim/utils/collect_env.py)
(or fill out the checklist below manually).

You can run the script with:
```
# For security purposes, please check the contents of collect_env.py before running it.
python src_python/habitat_sim/utils/collect_env.py
```

 - OS (e.g., Linux):
 - How you installed PyTorch (`conda`, `pip`, source):
 - Build command you used (if compiling from source):
 - Python version:
 - GPU models and configuration:
 - CUDA version:
 - PIP/conda dependencies packages versions
 - Any other relevant information:

dhruvbatra, 2022-09-06:
@ChanganVR

aclegg3, 2022-09-06:
Hey @doudoudou1999,
Looks like your system is having trouble finding the CUDA compiler. It could be that you need developer libraries which are missing, or your compiler system setup is non-standard.

Can you install correctly without `--with-cuda`? What about without `--audio`?

Inverse-function, 2025-10-31:
你好，我也遇见了这个问题，请问你是怎么解决的