facebookresearch/habitat-sim #1864: Using ReplicaCAD to programmatically create different room configurations

purvaten, 2022-09-14:
Hi,

I am trying to use ReplicaCAD to create a variety of scene instances with different configurations (e.g., half-opened cupboards) and different objects lying around. I had some questions for this:

1. Can you point me to the sampling function to add an arbitrary object on receptacles in ReplicaCAD?
2. Is there a way to Open/Close interactive receptacles such as drawers, fridge, cabinet doors etc programmatically?
3. Is there an API call which allows me to store the entire scene as a mesh (`.ply` / `.obj`)?

More context for the above questions respectively:
1. I found the [`sample_object_state()`](https://github.com/facebookresearch/habitat-sim/blob/a0061df551e8a3159238e51f14c1a946d4b70adc/examples/tutorials/nb_python/ECCV_2020_Interactivity.py#L277) but it looks like NavMeshes mostly only consist of the floor and stairs. Is there a function like this for other surfaces or will I just have to simulate a fall?
2. I believe this involves using the URDF files which will also be used in data for the [Re-arrangement Challenge](https://aihabitat.org/challenge/2022_rearrange/) but I didn't find starter code online if I had to create such data myself (say for half-opened drawers).
3. [`sim.get_sensor_observations()`](https://github.com/facebookresearch/habitat-sim/blob/a0061df551e8a3159238e51f14c1a946d4b70adc/examples/tutorials/nb_python/ECCV_2020_Interactivity.py#L608) returns an image, but is there a function that can return the scene mesh?

Thanks!

aclegg3, 2022-09-15:
Hey @purvaten,

Hopefully I can help answer your questions and provide some pointers. I'm interested to know a bit more about your use-case to know if there are additional suggestions which may be useful.

1. You found the navmesh sampling example which works reasonably well for large furniture. As you mentioned, this isn't really setup to generate small clutter in the scene. We have [a system for this in Habitat-lab](https://github.com/facebookresearch/habitat-lab/tree/main/habitat/datasets/rearrange) which we use to generating arrangements of items for robotic rearrangement tasks. The interesting sampling code is [here](https://github.com/facebookresearch/habitat-lab/tree/main/habitat/datasets/rearrange/samplers). The basic idea is that we annotate Receptacles (bounding boxes) and then construct configuration files which define how we want particular sets of objects to be spawned within the receptacles.

2. I think what you are hoping for are functions like object.open() and object.close(). Essentially this would involve annotating the joint positions of these open or closed states and then setting the [joint_positions](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.ManagedBulletArticulatedObject.html#joint_positions) attribute to corresponding value. You could easily do this in your user code by caching the correct values. A heuristic may be the maximum and minimum joint position, the [joint_position_limits](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.ManagedBulletArticulatedObject.html#joint_position_limits) which are defined for most joints.

3. Unfortunately we don't have an exporter for the entire scene as a mesh at this time. I believe this is in development. However, we do have some tooling for [loading replays in Blender,](https://github.com/facebookresearch/habitat-sim/pull/1759) there are C++ functions which[ join the entire scene into a mesh](https://github.com/facebookresearch/habitat-sim/blob/8cbc1204866137a566cf656b25179cd517ecb901/src/esp/sim/Simulator.cpp#L839) and you could merge an obj writer to that. We also have the ability to [export scene_instance.json files](https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#save_current_scene_config-a306e) so you can re-load an edited scene within Habitat.

purvaten, 2022-09-15:
Hi @aclegg3,

Thank you for the detailed reply. This is super helpful!

closed 2022-09-15 by purvaten