facebookresearch/habitat-sim #1875: Interactive tests shows nothing [WSL2 Ubuntu 20.04]

xEvheMary, 2022-09-29:
### I'm using
* WSL2 Windows 11
* Anaconda 3

I've installed both habitat-lab and habitat-sim (conda install) following the README up until the tests. The non-interactive tests in the habitat-lab are working fine. But the Interactive tests of both habitat-lab (interactive_play) and habitat-sim (habitat_viewer) show black. I've read that habitat doesn't support Windows or WSL (don't know how it is for VMWare) but it installed fine.

[image]

Any insight is welcome

aclegg3, 2022-09-29:
Looks like 0ns GPU duration means either nothing to render, or no rendering happening. Console logs would be necessary to know more.

xEvheMary, 2022-09-29:
It somehow runs correctly, very very rarely. I don't see any apparent difference between failing and successful logs.
It automatically uses my intel graphic as the renderer even after i change it to nvidia through MESA (though i'm not an expert myself so i might have missed something), but i doubt this is the reason.
This is the console log of the habitat viewer.
```
Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to physical DPI
Renderer: D3D12 (Intel(R) UHD Graphics 770) by Microsoft Corporation
OpenGL version: 4.1 (Core Profile) Mesa 22.2.0 - kisak-mesa PPA
Using optional features:
    GL_ARB_vertex_array_object
    GL_ARB_ES2_compatibility
    GL_ARB_separate_shader_objects
    GL_ARB_robustness
    GL_ARB_texture_storage
    GL_ARB_invalidate_subdata
    GL_ARB_texture_storage_multisample
    GL_ARB_multi_bind
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_texture_filter_anisotropic
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    mesa-dsa-createquery-except-pipeline-stats
    mesa-forward-compatible-line-width-range
[04:32:38:779433]:[Metadata] AttributesManagerBase.h(380)::createFromJsonOrDefaultInternal : <Dataset>: Proposing JSON name : default.scene_dataset_config.json from original name : default| This file does not exist.
[04:32:38:779497]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (capsule3DSolid:capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
[04:32:38:779530]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (capsule3DWireframe:capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
[04:32:38:779554]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (coneSolid:coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
[04:32:38:779575]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (coneWireframe:coneWireframe_segments_32_halfLen_1.25) created and registered.
[04:32:38:779594]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cubeSolid:cubeSolid) created and registered.
[04:32:38:779612]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cubeWireframe:cubeWireframe) created and registered.
[04:32:38:779633]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cylinderSolid:cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
[04:32:38:779654]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cylinderWireframe:cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
[04:32:38:779675]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (icosphereSolid:icosphereSolid_subdivs_1) created and registered.
[04:32:38:779694]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (icosphereWireframe:icosphereWireframe_subdivs_1) created and registered.
[04:32:38:779714]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (uvSphereSolid:uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
[04:32:38:779733]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (uvSphereWireframe:uvSphereWireframe_rings_16_segments_32) created and registered.
[04:32:38:779753]:[Metadata] AssetAttributesManager.cpp(110)::AssetAttributesManager : Built default primitive asset templates : 12
[04:32:38:779873]:[Metadata] SceneDatasetAttributesManager.cpp(35)::createObject : File (default) not found, so new default dataset attributes created  and registered.
[04:32:38:779886]:[Metadata] MetadataMediator.cpp(120)::createSceneDataset : Dataset default successfully created.
[04:32:38:779903]:[Metadata] AttributesManagerBase.h(380)::createFromJsonOrDefaultInternal : <Physics Manager>: Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json| This file does not exist.
[04:32:38:779925]:[Metadata] PhysicsAttributesManager.cpp(26)::createObject : File (./data/default.physics_config.json) not found, so new default physics manager attributes created and registered.
[04:32:38:779939]:[Metadata] MetadataMediator.cpp(203)::setActiveSceneDatasetName : Previous active dataset  changed to default successfully.
[04:32:38:779952]:[Metadata] MetadataMediator.cpp(175)::setCurrPhysicsAttributesHandle : Old physics manager attributes  changed to ./data/default.physics_config.json successfully.
[04:32:38:779957]:[Metadata] MetadataMediator.cpp(66)::setSimulatorConfiguration : Set new simulator config for scene/stage :  and dataset : default which is currently active dataset.
[04:32:38:785609]:[Metadata] MetadataMediator.cpp(66)::setSimulatorConfiguration : Set new simulator config for scene/stage : Download/scene_datasets/habitat-test-scenes/apartment_1.glb and dataset : default which is currently active dataset.
[04:32:38:785651]:[Metadata] AttributesManagerBase.h(321)::loadAllTemplatesFromPathAndExt : <Object Template> : ParsingObject Template: Cannot find data/objects/example_objects as directory or data/objects/example_objects.object_config.json as config file. Aborting parse.
[... cut ...]
[04:32:50:514005]:[Sensor] Sensor.cpp(69)::~Sensor : Deconstructing Sensor
[04:32:50:514874]:[Sensor] Sensor.cpp(69)::~Sensor : Deconstructing Sensor
[04:32:50:516273]:[Sensor] Sensor.cpp(69)::~Sensor : Deconstructing Sensor
[04:32:50:516939]:[Sensor] Sensor.cpp(69)::~Sensor : Deconstructing Sensor
[04:32:50:518751]:[Sensor] Sensor.cpp(69)::~Sensor : Deconstructing Sensor
[04:32:50:518807]:[Scene] SemanticScene.h(47)::~SemanticScene : Deconstructing SemanticScene
[04:32:50:519684]:[Gfx] Renderer.cpp(72)::~Impl : Deconstructing Renderer

aclegg3, 2022-09-29:
Are you looking around when it doesn't work? The mesh isn't just behind the camera?
Not sure if I know how to debug this issue if it isn't happening consistently.

xEvheMary, 2022-09-30:
Indeed, I moved when it successfully ran, the mesh is correct I suppose. I'm also not sure what caused this, since it's not like the program failed to run, but it just outputs weird things. I'll see what I can do and try