facebookresearch/habitat-sim #1894: Texture mesh and Semantic mesh visualization for hm3d and mp3d datasets

Jthon-lab, 2022-10-07:
## Habitat-Sim version
0.2.2
## Docs and Tutorials
Did you read the docs?  Yes
Did you check out the tutorials?  Yes
## ❓ Questions and Help
I am trying to visualize the both semantic and textured 3D mesh of hm3d and mp3d datasets with open3d. In the mp3d dataset directory, I get xxx.glb, xxx.house, xxx.navmesh, and xxx_semantic.ply. I can get the textured mesh by reading xxx_semantic.ply as shown below. In the simulation, the observations provide the first-person view segmentation results, but I was wondering how to get the global semantic mesh? 
[image]

In the hm3d dataset directory, I get xxx.basis.glb, xxx.basis.navmesh, xxx.semantic.glb, xxx.semantic.txt.  By reading xxx.semantic.glb, I can visualize the global semantic mesh as shown below, so how to visualize the global texture mesh similar to  mp3d? 
[image]

aclegg3, 2022-10-07:
Hey @Jthon-lab,

There are instructions for using both datasets in the README [here](https://github.com/facebookresearch/habitat-sim/blob/main/DATASETS.md#habitat-matterport-3d-research-dataset-hm3d).

More details:
Both datasets have semantic and RGB assets as you have mentioned. If you load the datasets through the SceneDataset config files as described in the documentation then you can use both color and semantic sensors to get renders of each one in Habitat. 

For MP3D, the semantics are per-vertex color embeddings in the ply file. For HM3D, the semantics are textured. You won't be able to load both of these simultaneously in another engine (e.g. open3D) because they have separate mesh geometry. You'll also need to be sure you grab the glb versions without basis compression (which will appear black) for most editors.

Jthon-lab, 2022-10-07:
So, in the mp3d dataset, xxx.glb stores the mesh with rgb texture color and xxx_semantic.ply stores the segmentation information, while in the hm3d dataset, I need to download the glb with the link (https://api.matterport.com/resources/habitat/hm3d-train-habitat.tar) to get the rgb texture?

Jthon-lab, 2022-10-07:
Thank you for your reply @aclegg3. 
So, in the mp3d dataset, xxx.glb stores the mesh with rgb texture color and xxx_semantic.ply stores the segmentation information, while in the hm3d dataset, I need to download the glb with the link (https://api.matterport.com/resources/habitat/hm3d-train-glb.tar) to get the rgb texture?

aclegg3, 2022-10-07:
@Jthon-lab 
Yes, that sounds correct. In both cases, the semantic and rgb assets are separate.

In HM3D the only difference is the texture set, so you could possibly create a blended material or swap the textures on the mesh.

In mp3d, there are also different mesh geometries, so you need to treat these as separate assets.

Jthon-lab, 2022-10-08:
@aclegg3 By the way, what is the difference in the coordination system between the habitat simulators and the mp3d or hm3d dataset?

aclegg3, 2022-10-11:
Seems like the issue is resolved.

closed 2022-10-11 by aclegg3