facebookresearch/habitat-sim #1905: Failed to load HM3D semantic annotations

XHRlyb, 2022-10-20:
I tried the example code about loading HM3D semantic annotations provided in datasets.md, but failed. Below is the results:
```
I1020 18:14:09.477332 10972 AttributesManagerBase.h:262] AttributesManager::buildCfgPathsFromJSONAndLoad : 123 paths specified in JSON doc for Scene Instance templates.
I1020 18:14:09.477349 10972 SceneDatasetAttributesManager.cpp:23] JSON Configuration File (data/hm3d/hm3d_annotated_basis.scene_dataset_config.json) based dataset attributes created and registered.
I1020 18:14:09.477355 10972 MetadataMediator.cpp:47] MetadataMediator::createDataset : Dataset data/hm3d/hm3d_annotated_basis.scene_dataset_config.json successfully created.
I1020 18:14:09.482674 10972 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Physics Manager) : Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json | This file  does not exist.
I1020 18:14:09.482738 10972 PhysicsAttributesManager.cpp:27] File (./data/default.physics_config.json) not found, so new default physics manager attributes created and registered.
I1020 18:14:09.482748 10972 AbstractObjectAttributesManagerBase.h:175] AbstractObjectAttributesManager<T>::createObject  (Stage) : Making attributes with handle : data/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.glb
I1020 18:14:09.482757 10972 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : data/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.glb changed to proposed JSON configuration filename : data/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.stage_config.json
I1020 18:14:09.482769 10972 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Stage) : Proposing JSON name : data/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.stage_config.json from original name : data/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.glb | This file  does not exist.
I1020 18:14:09.482851 10972 AbstractObjectAttributesManagerBase.h:181] AbstractObjectAttributesManager<T>::createObject  (Stage) : Done making attributes with handle : data/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.glb
I1020 18:14:09.482856 10972 AbstractObjectAttributesManagerBase.h:188] File (data/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.glb) exists but is not a recognized config filename extension, so new default Stage attributes created and registered.
I1020 18:14:09.482877 10972 Simulator.cpp:156] Loading navmesh from data/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.navmesh
I1020 18:14:09.483250 10972 Simulator.cpp:158] Loaded.
I1020 18:14:09.483270 10972 SceneGraph.h:93] Created DrawableGroup: 
Renderer: NVIDIA GeForce RTX 2080 Ti/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 510.85.02
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-forward-compatible-core-context
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
W1020 18:14:09.620342 10972 ResourceManager.cpp:228] ResourceManager::loadStage : Not loading semantic mesh - File Name : data/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis_semantic.ply does not exist.
I1020 18:14:09.620363 10972 ResourceManager.cpp:262] ResourceManager::loadStage : start load render asset data/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.glb.
I1020 18:14:09.620368 10972 ResourceManager.cpp:569] ResourceManager::loadStageInternal : Attempting to load stage data/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.glb 
I1020 18:14:09.620390 10972 ResourceManager.cpp:1119] Importing Basis files as BC7 for TEEsavR23oF.basis.glb
W1020 18:14:10.353278 10972 Simulator.cpp:248] :
---
 The active scene does not contain semantic annotations. 
---
I1020 18:14:10.354135 10972 simulator.py:213] Loaded navmesh data/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis.navmesh
I1020 18:14:10.535151 10972 PhysicsManager.cpp:33] Deconstructing PhysicsManager
I1020 18:14:10.535192 10972 SemanticScene.h:41] Deconstructing SemanticScene
I1020 18:14:10.535199 10972 SceneManager.h:25] Deconstructing SceneManager
I1020 18:14:10.535204 10972 SceneGraph.h:26] Deconstructing SceneGraph
I1020 18:14:10.535370 10972 Sensor.h:87] Deconstructing Sensor
I1020 18:14:10.539736 10972 Renderer.cpp:34] Deconstructing Renderer
I1020 18:14:10.539780 10972 WindowlessContext.h:17] Deconstructing WindowlessContext
I1020 18:14:10.544560 10972 Simulator.cpp:49] Deconstructing Simulator
```
It tried to find a non-exist "data/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.basis_semantic.ply", and said "The active scene does not contain semantic annotations."
So how can I fix it and get semantic labels? Thanks!

aclegg3, 2022-10-24:
Hey @XHRlyb,

If you used the dataset downloader util on `main` to get the data, you may be picking up v0.1 still.

See #1910 for an updated downloader script. Should be tested and merged into `main` soon. 

You should be looking for `data/hm3d/minival/00800-TEEsavR23oF/TEEsavR23oF.semantic.glb`, not a .ply file. This file is available in the minival zip file for v0.2 at this time.

aclegg3, 2022-11-02:
Downloader on main is updated now. Please try that and open a new issue if problems persist.