facebookresearch/habitat-sim #1907: Simulator.semantic_scene got empty aabbs of regions&objects in HM3DSem dataset

XHRlyb, 2022-10-24:
## Habitat-Sim version
v0.2.2

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/  
Yes.

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/
Yes.

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
I tried habitat_sim.Simulator.semantic_scene with habitat-sim v0.2.2 to get HM3DSem semantic annotations. I can get regions and object categories, but when it comes to bboxes, both region aabbs and object aabbs are empty (all data are 0.). So I just get category hierarchy in the scene, but don't where something is. 
SO, MY QUESTION IS:
- Does habitat-sim v0.2.2 support getting region and object bboxes of HM3DSem v0.1/v0.2? 
- How can I get correct instance semantic labels in HM3DSem scenes? 
Thanks!

aclegg3, 2022-10-24:
Hey @XHRlyb 

We have not exported semantic bounding boxes with the HM3D Semantics dataset at this time.

The semantic assets do contain per-vertex colors baked from the semantic textures. Therefore, it is possible to compute these bounding boxes from the source mesh data. For example, we have a prototype of this in the [viewer.cpp application](https://github.com/facebookresearch/habitat-sim/blob/e98c12498f3712381702466aff676f3c2a9fef92/src/utils/viewer/viewer.cpp#L1314) (e.g. `generateAndSaveSemanticCCReport`). I suggest you take a look here and see if there is a way to pre-compute the metadata you need.

aclegg3, 2022-10-25:
Hey @PeihaoChen,

I think the issue here is the color type exposed by open3d. Looks like they expose vertex colors as float64 array. The embedded colors from glTF should be byte data which would correctly convert back to hex, but maybe the intermediate float representation gets in the way.

For example, in habitat-sim C++ we are directly mapping the data into `uint32_t` [here](https://github.com/facebookresearch/habitat-sim/blob/e98c12498f3712381702466aff676f3c2a9fef92/src/esp/assets/GenericSemanticMeshData.cpp#L184).