facebookresearch/habitat-sim #1909: ImportError: cannot import name 'text' from 'magnum'

dwei-k, 2022-10-24:
when I have done 
`conda install habitat-sim withbullet -c conda-forge -c aihabitat`
and run 
`python examples/viewer.py --scene Data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`
I met 
Traceback (most recent call last):
  File "examples/viewer.py", line 19, in <module>
    from magnum import shaders, text
ImportError: cannot import name 'text' from 'magnum'

Looking forward to your reply thank you very much

aclegg3, 2022-10-24:
Hey @dwei-k,

Looks like you cloned the `main` branch of habitat-sim and then installed from the last stable conda release instead of nightly. Your build will therefore be missing some features and you shouldn't expect it to run correctly. Either pull the last stable tag locally or update your conda to a nightly build: `-c aihabitat-nightly`

Let me know if this resolves your issue.

dwei-k, 2022-10-27:
sorry, I do clone the `main` branch and didn't `conda install` with the right paramter
and after I pulled the last stable tag locally, `python examples/viewer.py --scene Data/scene_datasets/habitat-test-scenes/skokloster-castle.glb` works successfully
thank you very much, your suggetion helps me a lot

closed 2022-10-27 by aclegg3

Yuchen-Song, 2023-01-12:
@dwei-k Which specific version of habitat-sim did you install? I encountered the same problem with [stable version v0.1.7](https://github.com/facebookresearch/habitat-sim/releases/tag/stable) and [latest version v0.2.3](https://github.com/facebookresearch/habitat-sim/releases/tag/v0.2.3), my habitat-sim and habitat-lab packages are both at v0.2.3

dwei-k, 2023-01-13:
@Yuchen-Song ,hi
I remember I install v0.2.2 of habitat-sim which is the stable version at that time.