facebookresearch/habitat-sim #1922: Error when building habitat-sim on the RLRAudioPropogationUpdate branch

sreeharshaparuchur1, 2022-10-31:
Hi,

I'm trying to build habitat-sim with the audio functionality to use SoundSpaces 2.0. However, I run into this error when trying to build the project on the mentioned branch.

```
make[2]: *** [esp/sensor/CMakeFiles/sensor.dir/build.make:102: esp/sensor/CMakeFiles/sensor.dir/SensorFactory.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [esp/sensor/CMakeFiles/sensor.dir/build.make:154: esp/sensor/CMakeFiles/sensor.dir/AudioSensor.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:5875: esp/sensor/CMakeFiles/sensor.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
Traceback (most recent call last):
  File "setup.py", line 485, in <module>
    include_package_data=True,
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/__init__.py", line 87, in setup
    return distutils.core.setup(**attrs)
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/_distutils/core.py", line 185, in setup
    return run_commands(dist)
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
    dist.run_commands()
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 973, in run_commands
    self.run_command(cmd)
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 992, in run_command
    cmd_obj.run()
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/command/install.py", line 74, in run
    self.do_egg_install()
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/command/install.py", line 123, in do_egg_install
    self.run_command('bdist_egg')
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/_distutils/cmd.py", line 319, in run_command
    self.distribution.run_command(command)
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 992, in run_command
    cmd_obj.run()
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 165, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 151, in call_command
    self.run_command(cmdname)
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/_distutils/cmd.py", line 319, in run_command
    self.distribution.run_command(command)
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 992, in run_command
    cmd_obj.run()
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/_distutils/command/install_lib.py", line 112, in build
    self.run_command('build_ext')
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/_distutils/cmd.py", line 319, in run_command
    self.distribution.run_command(command)
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 992, in run_command
    cmd_obj.run()
  File "setup.py", line 271, in run
    self.build_extension(ext)
  File "setup.py", line 385, in build_extension
    [osp.join(CMAKE_BIN_DIR, "cmake"), "--build", self.build_temp] +  build_args #, self.build_temp] + build_args
  File "/home/ruddra/anaconda3/envs/ss20/lib/python3.7/subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2.

```

These errors are proceeded by errors specific to the AudioSensor, starting with: 
```
/home/chieftwit/SoundSpaces20/habitat-sim/src/esp/sensor/AudioSensor.h:36:3: error: ‘RLRA_ContextConfiguration’ does not name a type
   36 |   RLRA_ContextConfiguration acousticsConfig_;

```

Kindly help me out with the same as I was able to build habitat-sim multiple times before following the specific instructions [here](https://github.com/facebookresearch/sound-spaces/blob/main/README.md#installation). 

Thanks in advance!

dhruvbatra, 2022-10-31:
@ChanganVR

aclegg3, 2022-11-01:
@ChanganVR 

The submodule in [src/deps/](https://github.com/facebookresearch/habitat-sim/tree/main/src/deps) is  synced to a specific commit. I've opened a PR #1923 to update this to `main`.

closed 2022-11-02 by aclegg3

sreeharshaparuchur1, 2022-11-03:
Hi @aclegg3 ,

Thanks a lot for the speedy support! Following the instructions given in the soundspaces readme, I was able to build the requirements for SoundSpaces 2.0 successfully. I'd like to point out that the README still points to the faulty habitat-sim branch at [this line](https://github.com/facebookresearch/sound-spaces/blob/main/README.md?plain=1#L42). In accordance to where you pushed the fix for my issue, this line should be updated to point to the main branch of habitat-sim to prevent the above mentioned issue thwarting the setup of habitat-lab.

Thank you!

aclegg3, 2022-11-03:
Glad this is working for you now on `main`, @sreeharshaparuchur1.

I'll defer to @carlschisslerfb regarding the correct branch to work on at this time.

sreeharshaparuchur1, 2022-11-04:
Thanks for pushing the update to the RLRAudioPropogation branch asap. The build now works there too. 

Your support is super appreciated :)

YYX666660, 2023-02-08:
@sreeharshaparuchur1 @ChanganVR I encountered a similar problem when build habitat-sim with the audio functionality to use SoundSpaces 1.0. There are my errors when I tried to build:
```
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/esp/sensor/AudioSensor.cpp:389:51: error: expected ‘,’ before ‘)’ token
   static_assert( sizeof(vec3f) == 3*sizeof(float) );
                                                   ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/esp/sensor/AudioSensor.cpp:389:51: error: expected string-literal before ‘)’ token
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/esp/sensor/AudioSensor.cpp: In member function ‘void esp::sensor::AudioSensor::loadMesh(esp::sim::Simulator&)’:
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/esp/sensor/AudioSensor.cpp:503:51: error: expected ‘,’ before ‘)’ token
   static_assert( sizeof(vec3f) == 3*sizeof(float) );
                                                   ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/esp/sensor/AudioSensor.cpp:503:51: error: expected string-literal before ‘)’ token
esp/sensor/CMakeFiles/sensor.dir/build.make:153: recipe for target 'esp/sensor/CMakeFiles/sensor.dir/AudioSensor.cpp.o' failed
make[2]: *** [esp/sensor/CMakeFiles/sensor.dir/AudioSensor.cpp.o] Error 1
CMakeFiles/Makefile2:5677: recipe for target 'esp/sensor/CMakeFiles/sensor.dir/all' failed
make[1]: *** [esp/sensor/CMakeFiles/sensor.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
Traceback (most recent call last):
  File "/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/setup.py", line 468, in <module>
    setup(
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/__init__.py", line 87, in setup
    return distutils.core.setup(**attrs)
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 185, in setup
    return run_commands(dist)
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
    dist.run_commands()
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
    self.run_command(cmd)
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/dist.py", line 1208, in run_command
    super().run_command(command)
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/command/install.py", line 74, in run
    self.do_egg_install()
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/command/install.py", line 123, in do_egg_install
    self.run_command('bdist_egg')
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/dist.py", line 1208, in run_command
    super().run_command(command)
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/command/bdist_egg.py", line 165, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/command/bdist_egg.py", line 151, in call_command
    self.run_command(cmdname)
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/dist.py", line 1208, in run_command
    super().run_command(command)
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/_distutils/command/install_lib.py", line 112, in build
    self.run_command('build_ext')
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/dist.py", line 1208, in run_command
    super().run_command(command)
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/setup.py", line 271, in run
    self.build_extension(ext)
  File "/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/setup.py", line 383, in build_extension
    subprocess.check_call(
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2.
```

I tried to clone ```rlr-audio-propagation``` repo and copy its ```RLRAudioPropagationPkg/``` folder to overwrite the folder in ```src/deps/rlr-audio-propagation/```, but it does not work. And the errors are as above. It seems like the error comes from the ```AudioSensor.cpp```, so could you help me with this issue?

ChanganVR, 2023-02-08:
@YYX666660 please follow this step-by-step installation guide: https://github.com/facebookresearch/sound-spaces/blob/main/INSTALLATION.md

YYX666660, 2023-02-10:
@ChanganVR Thank you for your speedy support! I'm sure that I followed the guild and run the instructions as follows:
```
conda create -n ss python=3.9 cmake=3.14.0 -y
conda activate ss

git clone https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim
git checkout RLRAudioPropagationUpdate
python setup.py install --headless --audio
```
but when it comes to ```python setup.py install --headless --audio```, the building process went wrong. I even tried multiple times to build the environment but unfortunately it appeared the same error as above **(the above last comments of mine).** I do not know what's wrong with my ```AudioSensor.cpp```, because the error indicates ```make[2]: *** [esp/sensor/CMakeFiles/sensor.dir/AudioSensor.cpp.o] Error 1```. And the building process was broken at 74%:
```
[ 70%] Built target AssimpImporter
[ 70%] Built target PrimitiveImporter
[ 70%] Built target MagnumDebugTools
[ 72%] Built target magnum
[ 73%] Built target sim
[ 74%] Building CXX object esp/sensor/CMakeFiles/sensor.dir/AudioSensor.cpp.o
In file included from /media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/esp/sensor/AudioSensor.h:16:0,
                 from /media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/esp/sensor/AudioSensor.cpp:9:
``` 

I'm very confused with this error and I'd appreciate it if you can kindly help me out with the same as I was able to build habitat-sim.

Thanks in advance!

ChanganVR, 2023-02-10:
@YYX666660 can you show the full log here? the log above does not show any error message

YYX666660, 2023-02-10:
The whole error messages are as follows:
```
[ 70%] Built target MagnumDebugTools
[ 72%] Built target magnum
[ 73%] Built target sim
[ 74%] Building CXX object esp/sensor/CMakeFiles/sensor.dir/AudioSensor.cpp.o
In file included from /media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/esp/sensor/AudioSensor.h:16:0,
                 from /media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/esp/sensor/AudioSensor.cpp:9:
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:576:1: warning: type attributes ignored after type is already defined [-Wattributes]
 };
 ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:597:1: warning: type attributes ignored after type is already defined [-Wattributes]
 };
 ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:628:91: warning: ‘Configuration’ is deprecated [-Wdeprecated-declarations]
 RLRA_DEPRECATED std::ostream& operator << ( std::ostream& os, const Configuration& config );
                                                                                           ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:654:98: warning: ‘ChannelLayout’ is deprecated [-Wdeprecated-declarations]
 RLRA_DEPRECATED std::ostream& operator << ( std::ostream& os, const ChannelLayout& channelLayout );
                                                                                                  ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:683:43: warning: ‘Simulator’ is deprecated [-Wdeprecated-declarations]
     Simulator( const Simulator& other ) = delete;
                                           ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:684:38: warning: ‘Simulator’ is deprecated [-Wdeprecated-declarations]
     Simulator( Simulator&& other ) = delete;
                                      ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:676:23: note: declared here
 class RLRA_DEPRECATED Simulator
                       ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:685:56: warning: ‘Simulator’ is deprecated [-Wdeprecated-declarations]
     Simulator& operator = ( const Simulator& other ) = delete;
                                                        ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:676:23: note: declared here
 class RLRA_DEPRECATED Simulator
                       ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:685:56: warning: ‘Simulator’ is deprecated [-Wdeprecated-declarations]
     Simulator& operator = ( const Simulator& other ) = delete;
                                                        ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:686:51: warning: ‘Simulator’ is deprecated [-Wdeprecated-declarations]
     Simulator& operator = ( Simulator&& other ) = delete;
                                                   ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:676:23: note: declared here
 class RLRA_DEPRECATED Simulator
                       ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:686:51: warning: ‘Simulator’ is deprecated [-Wdeprecated-declarations]
     Simulator& operator = ( Simulator&& other ) = delete;
                                                   ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:676:23: note: declared here
 class RLRA_DEPRECATED Simulator
                       ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:691:69: warning: ‘Configuration’ is deprecated [-Wdeprecated-declarations]
     RLRA_DEPRECATED ErrorCodes Configure(const Configuration& config);
                                                                     ^
/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/src/deps/rlr-audio-propagation/RLRAudioPropagationPkg/headers/RLRAudioPropagation.h:704:97: warning: ‘VertexData’ is deprecated [-Wdeprecated-declarations]
     RLRA_DEPRECATED ErrorCodes LoadMeshData(const VertexData& vertices, const IndexData& indices);
                                                                                                 ^
[... cut ...]
  File "/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/setup.py", line 271, in run
    self.build_extension(ext)
  File "/media/data1/yeyx/move2hear-active-AV-separation/habitat-sim/setup.py", line 383, in build_extension
    subprocess.check_call(
  File "/home/yeyx/miniconda3/envs/ss/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2.
```

Thank you for your patient!!!

ChanganVR, 2023-02-10:
@YYX666660 this is a bizarre error and I don't know why it's happening (assuming you didn't modify any of the codes). All I could think of is to do clean install again (removing and re-cloning).

@aclegg3 any idea?

YYX666660, 2023-02-11:
@ChanganVR **Thank you for your advice!** I'm sure I did remove and re-clone the repository and did'n modify the original codes, then I tried multiple times and got the same error... so confused about it...

carlschisslerfb, 2023-02-13:
@YYX666660 The errors seem like your C++ compiler is not configured correctly. Does your compiler support C++17? It complains about static_assert() missing a string parameter, but that is valid code in C++17. Make sure that you have C++17 enabled.

YYX666660, 2023-02-19:
@carlschisslerfb Thx for your suggestion! I tried to update my gcc and g++ to 8.4.0 for satisfying the C++17 version. But when I run the ```python setup.py install --headless --audio```, the same error appeared. So this may not work...

aizhiqi-work, 2023-08-07:
@YYX666660 @ChanganVR    Has the problem been solved? I have been running soundspace2 recently, and this compilation took two days to do many times, which is also the problem

ChanganVR, 2023-08-12:
@aizhiqi-work did you run into the same error or was it just the compilation time took too long?