facebookresearch/habitat-sim #1929: wrong category in mp3d

pioneer-innovation, 2022-11-03:
## Habitat-Sim version
v0.2.2

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/
yes!
Did you check out the tutorials? https://aihabitat.org/tutorial/2020/
yes!
Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
I find that there are some wrong category semantic masks in the mp3d. There are so many mistakes that I suspect it is my own problem.

I use this code to convert instance_id to semantic_id:
```python
{int(obj.id.split("_")[-1]): obj.category.index() for obj in scene_meta.objects}
```

In scene rqfAleAoiTq.glb; position: [7.8854465, 0.6306739, 2.4871054]; rotation: (1, 0, 0, 0)

A part of ceiling is annotated as table:

[image]

In scene ZMojNkEp431.glb; position: [ 9.613549  ,  0.09357202, -2.3997178 ]; rotation: (1, 0, 0, 0) 

A furniture is annotated as sofa:

[image]

There are a lot of such mistakes in other scenes. I'm not sure whether the mp3d dataset itself has such errors or my own program. This makes it very difficult for me to fine-tune models.

pioneer-innovation, 2022-11-03:
In scene VLzqgDo317F.glb; position: [1.9985332, 3.0382738, 7.885516 ]; rotation: (1, 0, 0, 0)

Floor is annotated as curtain:

[image]

In scene B6ByNegPMKs.glb; position: [11.165708,  0.080204, 31.584179]; rotation: (1, 0, 0, 0)

Table and floor are annotated as wall:

[image]

There are many such mistakes！T-T

dhruvbatra, 2022-11-03:
I can't really tell what's going on in your visuals, but these are likely annotation errors in MP3D (and there's nothing Habitat can do about that). Consider switching to https://aihabitat.org/datasets/hm3d-semantics/

closed 2022-11-03 by dhruvbatra