facebookresearch/habitat-sim #1939: Unable to dynamically add a non-audio sensor to sim

sreeharshaparuchur1, 2022-11-11:
## Habitat-Sim Branch
```RLRAudioPropogationUpdate```

-> It's possible that this isn't a bug in the main branch but only in the above mentioned branch, hence this bug report. I am using said branch as it offers the capability to run the SoundSpaces 2.0 simulator.

## 🐛 Bug

I am unable to dynamically add a non-audio sensor such as ```habitat_sim.CameraSensorSpec()``` to the sim object. I am, however, able to add said sensor to the sim object if I declare it in a ```make_simple_cfg``` function as given [in this tutorial](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/main/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb#scrollTo=OVPVSxPOTsAt) , thus, it's not a pressing issue but I thought I'd report it nonetheless. A related discussion is [this](https://github.com/facebookresearch/habitat-sim/issues/856).

## Steps to Reproduce

Steps to reproduce the behavior:

1. I have executed the first cell in [this notebook](https://github.com/facebookresearch/sound-spaces/blob/main/examples/soundspaces2_quick_tutorial.ipynb).
2. I have added the following code in a subsequent cell: 
```
rgb_sensor_spec = habitat_sim.CameraSensorSpec()
rgb_sensor_spec.uuid = "color_sensor"
rgb_sensor_spec.sensor_type = habitat_sim.SensorType.COLOR
rgb_sensor_spec.resolution = [256, 256]
rgb_sensor_spec.position = [0.0, 1.5, 0.0]
sim.add_sensor(rgb_sensor_spec)
```
3. I execute the above cell.

The output is the following error message:

```
ValueError: Data for SensorType.COLOR sensor was not loaded during Simulator init. Cannot dynamically add a SensorType.COLOR sensor unless one already exists.
```

## Expected behavior

I expected to be able to add the RGB sensor in the same was as the referenced notebook added the audio sensor without encountering an error.

## Additional context

The error seems to indicate that I am unable to dynamically add a SensorType.COLOR sensor. To test if this is the case, I added multiple audio sensors (habitat_sim.sensor.RLRAudioPropagationChannelLayoutType.Binaural) with unique uuid's to the sim object and they were reflected in the config without an issue. 

To check if it's an issue with the SensorType as a whole, I created the following ```make_simple_cfg``` function that initialized a sim with a solitary RGB sensor.

```
def make_simple_cfg(sim_settings):
    # simulator backend
    sim_cfg = habitat_sim.SimulatorConfiguration()
    sim_cfg.scene_id = "/home/chieftwit/Datasets/mini_mp3d_ddppo/X7HyMhZNoso/X7HyMhZNoso.glb"
    sim_cfg.scene_dataset_config_file = "/home/chieftwit/SoundSpaces20/sound-spaces/data/scene_datasets/mp3d/mp3d.scene_dataset_config.json"
    sim_cfg.load_semantic_mesh = True
    sim_cfg.enable_physics = False

    agent_cfg = habitat_sim.agent.AgentConfiguration()
    rgb_sensor_spec = habitat_sim.CameraSensorSpec()
    rgb_sensor_spec.uuid = "color_sensor"
    rgb_sensor_spec.sensor_type = habitat_sim.SensorType.COLOR
    rgb_sensor_spec.resolution = [sim_settings["height"], sim_settings["width"]]
    rgb_sensor_spec.position = [0.0, sim_settings["sensor_height"], 0.0]

    agent_cfg.sensor_specifications = [rgb_sensor_spec]

    return habitat_sim.Configuration(sim_cfg, [agent_cfg])
```
The simulator was initialized without an error and I was able to dynamically add an audio sensor to the sim object without having to restart the simulator. Thus, showing me that the issue is not inherent to the SensorType but because of the inability to dynamically add it to the sim object.

## System Info

 - OS: Linux
 - How you installed PyTorch (`conda`, `pip`, source): pip
 - Build command you used (if compiling from source): python setup.py install --with-cuda --audio
 - Python version: 3.7.13

Skylion007, 2022-11-12:
This is a known limitation not a bug. Essentially, you need to have at least 1RGB camera loaded when the simulator is instantiated. We haven an optimization where textures are not loaded if no sensors are using them (allowing for much lower memory requirements and faster scene loading).

sreeharshaparuchur1, 2022-11-14:
Thank you for the insight.

closed 2023-02-27 by sreeharshaparuchur1