facebookresearch/habitat-sim #1940: Patchy semantic map and no depth data

pfmephisto, 2022-11-11:
## Problem
I seem to be getting a patchy segmentation map form my sematic sensor as well as no data form the depth sensor and I am not sure as to why.

[image]

## Habitat-Sim version
```
➤ conda list | grep habitat
habitat-sim               0.2.2                    pypi_0    pypi
habitat-sim-mutex         1.0              display_bullet    aihabitat
withbullet                2.0                           0    aihabitat
```

I'm try to using the habitat-matterport-3dresearch example dataset `GLAQ4DNUx5U.glb` 
and am running this from inside WSL2.

Any ideas as to what might be causing those issues?

## My configs:

```python
sim_settings = {
    "scene": test_scene,  # Scene path
    "scene_dataset": hm3d_scene_dataset,  # the scene dataset configuration files
    "default_agent": 0,
    "sensor_height": 1.5,  # Height of sensors in meters
    "width": 512,  # Spatial resolution of the observations
    "height": 512,
    "hfov": 90,
}

def make_simple_cfg(settings):

    # simulator backend
    sim_cfg = habitat_sim.SimulatorConfiguration()
    sim_cfg.gpu_device_id = 0
    sim_cfg.create_renderer = True
    sim_cfg.enable_physics = False
    sim_cfg.scene_id = settings["scene"]
    sim_cfg.scene_dataset_config_file = settings["scene_dataset"]
    sim_cfg.load_semantic_mesh = True
    sim_cfg.requires_textures = True

    # agent
    agent_cfg = habitat_sim.agent.AgentConfiguration()

    sensor_position = [0.0, settings["sensor_height"], 0.0]
    sensor_resolution = [settings["height"], settings["width"]]

    color_sensor_spec  = habitat_sim.CameraSensorSpec()
    color_sensor_spec.uuid = "color_sensor"
    color_sensor_spec.hfov = settings["hfov"]
    color_sensor_spec.sensor_type = habitat_sim.SensorType.COLOR
    color_sensor_spec.resolution = sensor_resolution
    color_sensor_spec.position = sensor_position
    color_sensor_spec.sensor_subtype = habitat_sim.SensorSubType.PINHOLE
    color_sensor_spec.noise_model = "None"

    depth_sensor_spec = habitat_sim.CameraSensorSpec()
    depth_sensor_spec.uuid = "depth_sensor"
    depth_sensor_spec.hfov = settings["hfov"]
    depth_sensor_spec.sensor_type = habitat_sim.SensorType.DEPTH
    depth_sensor_spec.resolution = sensor_resolution
    depth_sensor_spec.position = sensor_position
    depth_sensor_spec.sensor_subtype = habitat_sim.SensorSubType.PINHOLE
    depth_sensor_spec.noise_model = "None"

    semantic_sensor_spec = habitat_sim.CameraSensorSpec()
    semantic_sensor_spec.uuid = "semantic_sensor"
    semantic_sensor_spec.hfov = settings["hfov"]
[... 178 lines cut ...]
[10:35:23:925113]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 41:Flat.
[10:35:23:925130]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 42:Flat.
[10:35:23:925225]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 43:Flat.
[10:35:23:925267]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 44:Flat.
[10:35:23:925287]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 45:Flat.
[10:35:23:925319]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 46:Flat.
[10:35:23:925355]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 47:Flat.
[10:35:23:925570]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 48:Flat.
[10:35:23:925613]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 49:Flat.
[10:35:23:925643]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 50:Flat.
[10:35:23:925669]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 51:Flat.
[10:35:23:925693]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 52:Flat.
[10:35:23:925716]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 53:Flat.
[10:35:23:925738]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 54:Flat.
[10:35:23:925763]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 55:Flat.
[10:35:23:925793]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 56:Flat.
[10:35:24:534659]:[Sim] Simulator.cpp(439)::instanceStageForSceneAttributes : Successfully loaded stage named : /home/mephisto/repos/habitat-sim/data/habitat-matterport-3dresearch/example/00861-GLAQ4DNUx5U/GLAQ4DNUx5U.glb
[10:35:24:534754]:[Sim] Simulator.cpp(205)::reconfigure : CreateSceneInstance success == true for active scene name : /home/mephisto/repos/habitat-sim/data/habitat-matterport-3dresearch/example/00861-GLAQ4DNUx5U/GLAQ4DNUx5U.glb  with renderer.
[10:35:24:553539]:[Nav] PathFinder.cpp(410)::build : Building navmesh with 283 x 215 cells
[10:35:24:742313]:[Nav] PathFinder.cpp(681)::build : Created navmesh with 566 vertices 273 polygons
[10:35:24:742396]:[Sim] Simulator.cpp(835)::recomputeNavMesh : reconstruct navmesh successful
```

</p>
</details>

aclegg3, 2022-11-22:
Hey @pfmephisto, very odd issue here. 

I'm not sure why the depth image is empty. It may need to be normalized to the clipping range (e.g. 10 m). 

The semantic issue looks a bit like there is some render clipping bug on your system. My local version works fine on `main` and I don't see any red flags in these logs.

My first suggestion is try the nightly Habitat conda build or build locally from source on `main` and ensure you have HM3D Semantics v0.2.

pfmephisto, 2022-11-29:
Hey @aclegg3,

Still no luck.
With the nightly build running: 
`habitat-viewer /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`

give me a black screen with updating stats but now image.

Otherwise running my scrip results in an `unable to find EGL` error although I did not install it with headless.
```
Platform::WindowlessEglApplication::tryCreateContext(): unable to find EGL device for CUDA device 0     │
WindowlessContext: Unable to create windowless context
```
Would there be an issue with my graphic card?
I am using a NVIDIA Quadro P4200 Mobile

Or does the Linux distro matter?
I am on Arch