facebookresearch/habitat-sim #1941: Habitat Sim Demo not working

Southyang, 2022-11-16:
## Habitat-Sim version
v0.2.2

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

## ❓ Questions and Help
When I run the habitat sim demo(https://aihabitat.org/docs/habitat-api/habitat-sim-demo.html), the log file is as follows:
```
[10:14:54:190506]:[Metadata] AttributesManagerBase.h(380)::createFromJsonOrDefaultInternal : <Dataset>: Proposing JSON name : default.scene_dataset_config.json from original name : default| This file does not exist.
[10:14:54:190590]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (capsule3DSolid:capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
[10:14:54:190620]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (capsule3DWireframe:capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
[10:14:54:190640]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (coneSolid:coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
[10:14:54:190658]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (coneWireframe:coneWireframe_segments_32_halfLen_1.25) created and registered.
[10:14:54:190673]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cubeSolid:cubeSolid) created and registered.
[10:14:54:190685]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cubeWireframe:cubeWireframe) created and registered.
[10:14:54:190699]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cylinderSolid:cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
[10:14:54:190716]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cylinderWireframe:cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
[10:14:54:190730]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (icosphereSolid:icosphereSolid_subdivs_1) created and registered.
[10:14:54:190743]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (icosphereWireframe:icosphereWireframe_subdivs_1) created and registered.
[10:14:54:190757]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (uvSphereSolid:uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
[10:14:54:190770]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (uvSphereWireframe:uvSphereWireframe_rings_16_segments_32) created and registered.
[10:14:54:190780]:[Metadata] AssetAttributesManager.cpp(110)::AssetAttributesManager : Built default primitive asset templates : 12
[10:14:54:190961]:[Metadata] SceneDatasetAttributesManager.cpp(35)::createObject : File (default) not found, so new default dataset attributes created  and registered.
[10:14:54:190973]:[Metadata] MetadataMediator.cpp(120)::createSceneDataset : Dataset default successfully created.
[10:14:54:190981]:[Metadata] AttributesManagerBase.h(380)::createFromJsonOrDefaultInternal : <Physics Manager>: Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json| This file does not exist.
[10:14:54:190992]:[Metadata] PhysicsAttributesManager.cpp(26)::createObject : File (./data/default.physics_config.json) not found, so new default physics manager attributes created and registered.
[10:14:54:190999]:[Metadata] MetadataMediator.cpp(203)::setActiveSceneDatasetName : Previous active dataset  changed to default successfully.
[10:14:54:191004]:[Metadata] MetadataMediator.cpp(175)::setCurrPhysicsAttributesHandle : Old physics manager attributes  changed to ./data/default.physics_config.json successfully.
[10:14:54:191012]:[Metadata] MetadataMediator.cpp(66)::setSimulatorConfiguration : Set new simulator config for scene/stage : /home/ros/southyang/data/mp3d/17DRP5sb8fy/17DRP5sb8fy.glb and dataset : default which is currently active dataset.
Renderer: SVGA3D; build: RELEASE;  LLVM; by VMware, Inc.
OpenGL version: 3.3 (Core Profile) Mesa 20.0.8
Using optional features:
    GL_ARB_vertex_array_object
    GL_ARB_ES2_compatibility
    GL_ARB_separate_shader_objects
    GL_ARB_robustness
    GL_ARB_texture_storage
    GL_ARB_invalidate_subdata
    GL_ARB_texture_storage_multisample
    GL_ARB_multi_bind
    GL_ARB_direct_state_access
    GL_ARB_texture_filter_anisotropic
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    svga3d-gettexsubimage-oob-write
    svga3d-broken-dsa-bufferdata
    mesa-dsa-createquery-except-pipeline-stats
    mesa-forward-compatible-line-width-range
    svga3d-texture-upload-slice-by-slice
[10:14:54:298197]:[Metadata] MetadataMediator.cpp(297)::getSceneInstanceAttributesByName : Dataset : default has no preloaded SceneInstanceAttributes or StageAttributes named : /home/ros/southyang/data/mp3d/17DRP5sb8fy/17DRP5sb8fy.glb so loading/creating a new StageAttributes with this name, and then creating a SceneInstanceAttributes with the same name that references this stage.
[10:14:54:298226]:[Metadata] AttributesManagerBase.h(380)::createFromJsonOrDefaultInternal : <Stage Template>: Proposing JSON name : /home/ros/southyang/data/mp3d/17DRP5sb8fy/17DRP5sb8fy.stage_config.json from original name : /home/ros/southyang/data/mp3d/17DRP5sb8fy/17DRP5sb8fy.glb| This file does not exist.
[10:14:54:298275]:[Metadata] AbstractObjectAttributesManagerBase.h(184)::createObject : File (/home/ros/southyang/data/mp3d/17DRP5sb8fy/17DRP5sb8fy.glb) exists but is not a recognized config filename extension, so new default Stage Template attributes created and registered.
[10:14:54:298313]:[Metadata] SceneDatasetAttributes.cpp(47)::addNewSceneInstanceToDataset : Dataset : 'default' : Stage Attributes '/home/ros/southyang/data/mp3d/17DRP5sb8fy/17DRP5sb8fy.glb' specified in Scene Attributes exists in dataset library.
[10:14:54:298324]:[Metadata] SceneDatasetAttributes.cpp(84)::addNewSceneInstanceToDataset : Dataset : 'default' : Lighting Layout Attributes 'no_lights' specified in Scene Attributes but does not exist in dataset, so creating.
[10:14:54:298333]:[Metadata] AttributesManagerBase.h(380)::createFromJsonOrDefaultInternal : <Lighting Layout>: Proposing JSON name : no_lights.lighting_config.json from original name : no_lights| This file does not exist.
[10:14:54:298346]:[Metadata] LightLayoutAttributesManager.cpp(36)::createObject : File (no_lights) not found, so new default light layout attributes created and registered.
[10:14:54:298359]:[Sim] Simulator.cpp(237)::createSceneInstance : Navmesh file location in scene instance : /home/ros/southyang/data/mp3d/17DRP5sb8fy/17DRP5sb8fy.navmesh
[10:14:54:298372]:[Sim] Simulator.cpp(242)::createSceneInstance : Loading navmesh from /home/ros/southyang/data/mp3d/17DRP5sb8fy/17DRP5sb8fy.navmesh
[10:14:54:298387]:[Nav] PathFinder.cpp(852)::loadNavMesh : NavMeshSettings aren't present, guessing that they are the default
[10:14:54:298456]:[Sim] Simulator.cpp(244)::createSceneInstance : Navmesh Loaded.
[10:14:54:298475]:[Scene] SceneGraph.h(85)::createDrawableGroup : Created DrawableGroup: 
[10:14:54:298484]:[Assets] ResourceManager.cpp(297)::loadSemanticSceneDescriptor : SceneInstance : /home/ros/southyang/data/mp3d/17DRP5sb8fy/17DRP5sb8fy.glb proposed Semantic Scene Descriptor filename : /home/ros/southyang/data/mp3d/17DRP5sb8fy/17DRP5sb8fy.house
[10:14:54:318638]:[Assets] ResourceManager.cpp(308)::loadSemanticSceneDescriptor : SSD with SceneInstanceAttributes-provided name  /home/ros/southyang/data/mp3d/17DRP5sb8fy/17DRP5sb8fy.house successfully found and loaded
[10:14:54:318700]:[Assets] ResourceManager.cpp(1641)::ConfigureImporterManagerGLExtensions : Importing Basis files as ASTC 4x4.
[10:14:54:318802]:[Sim] Simulator.cpp(302)::createSceneInstance : Using scene instance-specified Light key : - no_lights -
[10:14:54:318816]:[Metadata] MetadataMediator.cpp(66)::setSimulatorConfiguration : Set new simulator config for scene/stage : /home/ros/southyang/data/mp3d/17DRP5sb8fy/17DRP5sb8fy.glb and dataset : default which is currently active dataset.
[... 18 lines cut ...]
[10:14:58:573453]:[Assets] ResourceManager.cpp(2194)::loadMaterials : Idx 05:Flat.
[... cut ...]
[10:14:58:630113]:[Sim] Simulator.cpp(205)::reconfigure : CreateSceneInstance success == true for active scene name : /home/ros/southyang/data/mp3d/17DRP5sb8fy/17DRP5sb8fy.glb  with renderer.
[10:14:58:630229]:[Sensor] SensorFactory.cpp(56)::createSensors : Unreachable code : Cannot add the specified non-visual sensorType: 1
Reached unreachable code at /home/ros/southyang/coding/habitat-sim/src/esp/sensor/SensorFactory.cpp:59
已放弃 (核心已转储)
```
My purpose is to get three kinds of images in the navigation process, namely rgb, semantic, depth.

Southyang, 2022-11-16:
### Wrong part in make_cfg() function
1.Code in demo：
```
sim_cfg.scene.id = settings["scene"]
```
**Correction：**
```
sim_cfg.scene_id = settings["scene"]
```
2.Code in demo：
```
sensor_spec = habitat_sim.SensorSpec()
```
**Correction：**
```
sensor_spec = habitat_sim.CameraSensorSpec()
```
now I can get the result like demo

Baozao-99, 2022-11-17:
I met the same issue when I tried to use the ReplicaCAD.
But I can open the viewer by C++ ,while python found a error.

aclegg3, 2022-11-22:
Thanks @Baozao-99,

Thanks for pointing this out. This is clearly an outdated tutorial as your correction to the `scene_id` indicates an update to the API.
The problem here is that the demo notebook is hosted on fbaipublic files and therefore not tested by the repo for breaking changes. 
@mathfac @vincentpierre this is part of the Habitat-lab web documentation. We should probably remove this from the documentation or add it to the repo for testing.

I suggest you look at `Habitat-Sim Basics for Navigation` from our [ECCV tutorials](https://aihabitat.org/tutorial/2020/) for updated API examples.

**Short answer to your issue**: you should set the `sim_cfg.scene_dataset_config_file = "./data/scene_datasets/mp3d_example/mp3d.scene_dataset_config.json"` (or your equivalent path) to correctly capture the orientation of the semantic scene asset.

Southyang, 2022-11-23:
Thanks！ @aclegg3 
The result is correct after setting the `sim_cfg.scene_dataset_config_file`.

closed 2022-11-23 by Southyang

chris-naknak, 2023-07-24:
Thank you for your specific question. @Southyang 
I wonder where you wrote '''sim_cfg.scene_dataset_config_file="path/to/config/" '''in the code.

jeezrick, 2023-11-15:
> Thanks @Baozao-99,
>
> Thanks for pointing this out. This is clearly an outdated tutorial as your correction to the `scene_id` indicates an update to the API. The problem here is that the demo notebook [...]

As far as I tried, a lot of the tutorial code is outdated, some of them are just unrunable.

Star-down, 2024-09-05:
@aclegg3 
hello! where can i get file"/data/scene_datasets/mp3d_example/mp3d.scene_dataset_config.json",I used the -- habitat parameter when downloading the MP3D dataset. After downloading, it seems that my file is missing the /data/scene_datasets/mp3d_example/mp3d.scene_dataset_config.json file.