facebookresearch/habitat-sim #1944: `python setup.py install --headless --with-cuda --cmake`   error

dwei-k, 2022-11-21:
I run `rm -rf build` and  `python setup.py install --headless --with-cuda --cmake`
and errors:
setup.py:419: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  assert StrictVersion(
setup.py:421: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  ) >= StrictVersion("3.6"), "Must use python3.6 or newer"
running install
/opt/conda/envs/poni/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/opt/conda/envs/poni/lib/python3.8/site-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
adding license file 'LICENSE'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build/lib.linux-x86_64-cpython-38
creating build/lib.linux-x86_64-cpython-38/examples
copying examples/settings.py -> build/lib.linux-x86_64-cpython-38/examples
copying examples/ab_test.py -> build/lib.linux-x86_64-cpython-38/examples
copying examples/__init__.py -> build/lib.linux-x86_64-cpython-38/examples
copying examples/example.py -> build/lib.linux-x86_64-cpython-38/examples
copying examples/benchmark.py -> build/lib.linux-x86_64-cpython-38/examples
copying examples/demo_runner.py -> build/lib.linux-x86_64-cpython-38/examples
creating build/lib.linux-x86_64-cpython-38/tests
copying tests/test_profiling_utils.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_sensors.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_greedy_follower.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_pyrobot_noisy_controls.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_light_setup.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_semantic_scene.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_agent.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_robot_wrapper.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_snap_point.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_random_seed.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_physics.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_navmesh.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/conftest.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_utils.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_simulator.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/__init__.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_configs.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_physics_benchmarking.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_compare_profiles.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_examples.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_data_extraction.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_attributes_managers.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_noise_models.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_gfx.py -> build/lib.linux-x86_64-cpython-38/tests
copying tests/test_controls.py -> build/lib.linux-x86_64-cpython-38/tests
creating build/lib.linux-x86_64-cpython-38/habitat_sim
copying habitat_sim/metadata.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying habitat_sim/sim.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying habitat_sim/scene.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying habitat_sim/__init__.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying habitat_sim/errors.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying habitat_sim/simulator.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying habitat_sim/geo.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying habitat_sim/registry.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying habitat_sim/logging.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying habitat_sim/attributes.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
copying habitat_sim/gfx.py -> build/lib.linux-x86_64-cpython-38/habitat_sim
[... 297 lines cut ...]
  File "/opt/conda/envs/poni/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/opt/conda/envs/poni/lib/python3.8/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/opt/conda/envs/poni/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/opt/conda/envs/poni/lib/python3.8/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/opt/conda/envs/poni/lib/python3.8/site-packages/setuptools/_distutils/command/install_lib.py", line 112, in build
    self.run_command('build_ext')
  File "/opt/conda/envs/poni/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/opt/conda/envs/poni/lib/python3.8/site-packages/setuptools/dist.py", line 1217, in run_command
    super().run_command(command)
  File "/opt/conda/envs/poni/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "setup.py", line 247, in run
    self.build_extension(ext)
  File "setup.py", line 332, in build_extension
    subprocess.check_call(
  File "/opt/conda/envs/poni/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/root/Projects/PONI/dependencies/habitat-sim/build/lib.linux-x86_64-cpython-38/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/opt/conda/envs/poni/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-GNinja', '-DBUILD_GUI_VIEWERS=OFF', '-DTARGET_HEADLESS=ON', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBUILD_WITH_VHACD=OFF', '-DBUILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=ON', '-DBUILD_BASIS_COMPRESSOR=OFF', '/root/Projects/PONI/dependencies/habitat-sim/src']' returned non-zero exit status 1.

Really need help, thank you very much

aclegg3, 2022-11-21:
Hey @dwei-k,

Looks like a couple of system configuration issues at first glance.

1. Looks like OpenGL libraries not found and may not be installed or exposed:
```
Could NOT find OpenGL (missing: OPENGL_opengl_LIBRARY OPENGL_glx_LIBRARY
OPENGL_INCLUDE_DIR)
```
You might try (2) from [the install steps](https://github.com/facebookresearch/habitat-sim/blob/main/BUILD_FROM_SOURCE.md#build-from-source) or equivalent on your system. 

2. Looks like some CMAKE checks are failing. Honestly not sure how significant this is:
```
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
...
-- Check size of unsigned int16 - failed
...
-- Performing Test HAVE_NO_UNNAMED_TYPE_TEMPLATE_ARGS - Failed
...
-- Performing Test HAVE_RWLOCK - Failed
...
``` and a few more

dwei-k, 2022-11-23:
thank you very much 
I think my docker's configuration was not correct, I'll try to solve it with your advice

closed 2022-12-22 by dwei-k