facebookresearch/habitat-sim #1948: tools/gen_gibson_semantics.sh: line 28: tools/../build/utils/datatool/datatool: No such file or directory

dwei-k, 2022-11-24:
'git clone --branch stable https://github.com/facebookresearch/habitat-sim.git'

'cd habitat'

'pip install -r requirements.txt'

'python setup.py install --headless --with-cuda'

'tools/gen_gibson_semantic.sh <path> <path> <path>'

then encountered `tools/gen_gibson_semantics.sh: line 28: tools/../build/utils/datatool/datatool: No such file or directory`

I found that there is no `utils` in `build` and there is a file `utils/datatool` in `src`

closed 2022-11-25 by dwei-k

LIUQI-creat, 2022-12-21:
Hello, I performed the same operation and I also encountered this problem. Have you solved it ？Thank you very much! @dwei-k 
@likethesky

dwei-k, 2022-12-22:
git clone https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim
conda activate your_env
python setup.py build --build-datatool

LIUQI-creat, 2022-12-22:
> 

Thanks for your timely reply! I will follow your advice.