facebookresearch/habitat-sim #1979: Finding global coordinates of a point in a scene to place objects

adrshsrvstv, 2022-12-30:
I am trying to initialize a scene with different objects at different (but fixed) points several times for a benchmark. How do I find the global coordinates in the scene so I could add an object there - for example, the coordinates of the table top in one of the apartment scenes?

Ukias, 2023-01-01:
If you are using ReplicaCAD, then one possibility is to look up the translation of an object in the respective JSON-scene configuration file in the folder "configs/scenes". However, I would also be interested in getting the global coordinates and the scale (for example of a table mesh) from the simulation, e.g. in order to place objects on a tabletop. The scale is not contained in the JSON configuration files.

aclegg3, 2023-01-11:
Hey @adrshsrvstv, 

@Ukias is correct that the transform information in the JSON could be used to get object transform information outside of the simulator API.

Depending on the specifics of your situation, there are a few options to get this kind of information within the simulator:

1. If you want details of an object's transform, you can query it from the ObjectManager. See the [RigidObjects tutorial](https://aihabitat.org/docs/habitat-sim/managed-rigid-object-tutorial.html) for details of the API.
2. You can query the object's bounding box (in local space) to compute points of interest such as the top, bottom, and corner of the object. First get the [object's SceneNode](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.ManagedBulletRigidObject.html#root_scene_node) and then query the [mesh_bb](https://aihabitat.org/docs/habitat-sim/habitat_sim.scene.SceneNode.html#mesh_bb).
4. If you are using the Viewer application, you could print the [raycast hit position from a mouse click](https://github.com/facebookresearch/habitat-sim/blob/4ed8d04021ec7c0c4ad33ad441a3a46163f029ff/src/utils/viewer/viewer.cpp#L2054) (right click with `--enable-physics` to add a primitive at the click location. Add a line of code to print that location).

Ukias, 2023-03-27:
output for object's scene node: mesh_bb = Range({0, 0, 0}, {0, 0, 0}).
(Not working)

transform is not precise enough. I need the top of a table.

Ukias, 2023-04-23:
For me solved by adding the maximum of root_scene_node.cumulative__bb to the translation of the root_scene_node.