facebookresearch/habitat-sim #1980: Depth observation in viewer.py has overlapping with previous frames

moyutianque, 2023-01-04:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/
yes
Did you check out the tutorials? https://aihabitat.org/tutorial/2020/
yes
Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
I am playing with examples/viewer.py
I added sim_settings['depth_sensor'] = True and use self.sim.get_sensor_observations() after each agent.act(x) to get depth observation and color observation.
The color observation is correct without error, but it seems the buffer of depth observation is not cleared. There is a huge overlapping between the current frame and the previous frames. You can find the effect in the attached images.

dhruvbatra, 2023-01-04:
That's odd. If you give us a minimal reproducible example script (with details of the scene), we can look at reproducing it.

moyutianque, 2023-01-04:
Please find the testing data and script [here](https://drive.google.com/file/d/1eLMQZ5wclMkq2EUvMSpzJkVzSMmFoKQN/view?usp=sharing)
just cd to the directory and 
```bash
sh run_my_inter_replica.sh
```
Some of my previous outputs can be found in ```test_output```

dhruvbatra, 2023-01-04:
Can we get a github repository where we can actually look at code?

moyutianque, 2023-01-05:
sure I also uploaded it to https://github.com/moyutianque/habitat-viewer-test

0mdc, 2023-01-06:
@moyutianque 

This appears to be caused by the depth buffer not being cleared between draws.

Can you add this code immediately before `self.sim.get_sensor_observations()`:
```
mn.gl.default_framebuffer.clear(mn.gl.FramebufferClear.COLOR | mn.gl.FramebufferClear.DEPTH)
```

moyutianque, 2023-01-06:
> @moyutianque
> 
> This appears to be caused by the depth buffer not being cleared between draws.
> 
> Can you add this code immediately before `self.sim.get_sensor_observations()`:
> 
> ```
> mn.gl.default_framebuffer.clear(mn.gl.FramebufferClear.COLOR | mn.gl.FramebufferClear.DEPTH)
> ```

I tried to add this line in multiple places, but it seems the depth buffer is still not being cleared. It is strange that rgb and semantic observations do not have these overlapping effects.

moyutianque, 2023-01-09:
Update, it seems without using ```from magnum.platform.glfw import Application``` as the interactive interface, the depth rendering does not have problems. I tried to keep settings the same but directly visualize observations by PIL.

DummyNodeHead, 2023-02-21:
Hi. Have you guys solved this problem? Same problem with me and any response will be greatly appreciated.

moyutianque, 2023-02-21:
> Hi. Have you guys solved this problem? Same problem with me and any response will be greatly appreciated.

not solved. I give up using this interactive version

DummyNodeHead, 2023-02-21:
I managed to "solve" this in a residual way.

```
if self.has_prev_depth:
                render_depth = obs['depth_sensor'] - self.prev_depth
            else:
                self.has_prev_depth = True
                render_depth = obs['depth_sensor']
                
            self.prev_depth = obs['depth_sensor']
```

And the render_depth is the depth value I get. Although this method doesnt tackle the root cause, it helps me get the depth observation.

VladimirYugay, 2023-04-25:
@DummyNodeHead how did you run the viewer for the replica dataset? How to find the proper json file for that? 
For me it never works.

DummyNodeHead, 2023-04-26:
> @DummyNodeHead how did you run the viewer for the replica dataset? How to find the proper json file for that? For me it never works.

Hi. I met this problem before but I almost forget it now. Maybe you can show some details so I can help you. :)

Now this command works for me.
`./build/viewer --dataset /PATH/TO/REPLICA/replica.scene_dataset_config.json -- frl_apartment_0`