facebookresearch/habitat-sim #1988: NavMesh Error with certain HM3D scenes

Michael-Equi, 2023-01-17:
## Habitat-Sim version
v0.2.1

I am having issues navigating on some of the HM3D v0.2 scenes. I have provided code below that demonstrates the issue. 612 of the 800 training scenes work fine but the remaining 188 cause the following error: `RuntimeError: NavMesh has no navigable area, this indicates an issue with the NavMesh`

Any suggestions on what the issue might be and how I can fix it?

Thanks

```
import habitat_sim
import habitat_sim.agent
import random
from settings import default_sim_settings, make_cfg

settings = default_sim_settings.copy()
settings["seed"] = 5
settings["save_png"] = True
settings["color_sensor"] = True
settings["sensor_height"] = 0.5
settings["width"] = 224
settings["height"] = 224
settings['scene'] = "/raid/michaelequi/hm3dv2/scene_datasets/hm3d/train/00386-b3WpMbPFB6q/b3WpMbPFB6q.basis.glb"

_cfg = make_cfg(settings)

_sim = habitat_sim.Simulator(_cfg) 
_sim.seed(settings["seed"])
random.seed(settings["seed"])

agent = _sim.initialize_agent(settings["default_agent"])
greedy_follower = _sim.make_greedy_follower(agent_id=settings["default_agent"])

start_state = agent.get_state()
start_state.position = _sim.pathfinder.get_random_navigable_point()
```

srama2512, 2023-01-18:
@Michael-Equi - I could not find the file settings. I'm assuming you're referring to `habitat_sim.utils.settings`? I tried running this script after changing the import and updating the scene path. It works as expected. I'm using the habitat-sim nightly build installed today.
```
conda install habitat-sim headless -c conda-forge -c aihabitat-nightly
```

Michael-Equi, 2023-01-18:
It seems to have been an issue with habitat sim v0.2.1. v0.2.3 from nightly works fine. Thanks!