facebookresearch/habitat-sim #1992: Issues after installation using conda

jcarlosgm30, 2023-01-30:
Good morning,

I've installed habitat-sim using conda and tested the scripts described [here](https://github.com/facebookresearch/habitat-sim#testing). The c++ programs are working correctly but I've experimented with some issues with the python scripts. 

In the following I show the tracebacks:

```
python habitat-sim/examples/viewer.py --dataset data/replica_cad/replicaCAD.scene_dataset_config.json --scene apt_1
Traceback (most recent call last):
  File "habitat-sim/examples/viewer.py", line 19, in <module>
    from magnum import shaders, text
ImportError: cannot import name 'text' from 'magnum' (/opt/conda/envs/habitat/lib/python3.7/site-packages/magnum/__init__.py)
```
```
python habitat-sim/examples/example.py --scene data/scene_datasets/habitat-test-scenes/skokloster-castle.glb --enable_physics
Traceback (most recent call last):
  File "habitat-sim/examples/example.py", line 10, in <module>
    import demo_runner as dr
  File "/home/josecarlos/object_goal_navigation/habitat-sim/examples/demo_runner.py", line 15, in <module>
    from settings import default_sim_settings, make_cfg
  File "/home/josecarlos/object_goal_navigation/habitat-sim/examples/settings.py", line 6, in <module>
    from habitat_sim.utils.settings import default_sim_settings
ModuleNotFoundError: No module named 'habitat_sim.utils.settings'
```
And these are the habitat python packages installed in conda:

```
conda list | grep habitat
# packages in environment at /opt/conda/envs/habitat:
habitat-baselines         0.2.3                     dev_0    <develop>
habitat-lab               0.2.3                     dev_0    <develop>
habitat-sim               0.2.2                    pypi_0    pypi
habitat-sim-mutex         1.0              display_bullet    aihabitat
withbullet                2.0                           0    aihabitat
```

I really appreciate any help you can provide.

BR,
Carlos

aclegg3, 2023-01-31:
Hey @jcarlosgm30,

Looks to me like you may have a version synchronization issue here.
I see `habitat-sim 0.2.2` and `habitat-lab 0.2.3`. Habitat-sim should also be v0.2.3. 
Try re-installing from conda to get v0.2.3 and then running these tests again.

jcarlosgm30, 2023-02-02:
Hi @aclegg3,

You are right, the issue was the version was mismatched. I've installed the `habitat-sim v0.2.3` explicitly and it is working now!

I leave here the way to install the version explained in case someone needs it:
`conda install habitat-sim=0.2.3 withbullet -c conda-forge -c aihabitat`

Thank you again.

BR, 
Carlos

closed 2023-02-02 by jcarlosgm30