facebookresearch/habitat-sim #1998: enableMaterials result in crashes when simulating

YYDreamzure, 2023-02-07:
## Habitat-Sim version
v0.2.3

I have issue running the Habitat-sim simulator with materials. 
Sample code I was using: [minimal.example.py ](https://github.com/facebookresearch/sound-spaces/blob/main/examples/minimal_example.py). 
The original setting is:
audio_sensor_spec.enableMaterials = False
I had no issues running this successfully
But when I changed it to:
audio_sensor_spec.enableMaterials = False
with
audio_sensor.setAudioMaterialsJSON("/path/to/data/mp3d_material_config.json")

The jupyter kernel died and when I run in the terminal, the following error was reported:
Assertion resourceDict_.count(filename) > 0 failed at /home/dto/habitat-sim/src/esp/assets/ResourceManager.cpp:2995
Aborted (core dumped)

Please advice how I can solve this issue. Thank you in advance!

## ❓ Questions and Help

sreeharshaparuchur1, 2023-02-08:
Hey @YYDreamzure, 

I have faced this issue too and recently the maintainer of the soundspaces repo has listed the same as a known issue over [here](https://github.com/facebookresearch/sound-spaces/blob/main/INSTALLATION.md). 

The best solution would be to set that variable to False.

ChanganVR, 2023-02-09:
Hi @YYDreamzure this is mostly because you don't have the semantic files for the dataset you are using. Please check if you could find the required semantic files.

YYDreamzure, 2023-02-13:
@ChanganVR Many thanks for the advice! I used the hm3d sample and the semantic files came in a separate folder. The problem went away after I put them together with the .glb file.

closed 2023-02-13 by YYDreamzure