facebookresearch/habitat-sim #2002: unable to find EGL device for CUDA device 0

uahic, 2023-02-10:
## Habitat-Sim version
version: main

I know this issue has been reported several times and is often solved by installing the GPU drivers properly. In my case, I'd like to run Habitat in a Ubuntu 20 or 22 container *without* GPU support but for some reason  I dont get it working anymore without habitat asking for CUDA devices for libEGL. But I had habitat once running on my machine using Docker. I dont know whether I put dependencies in my apt-install list that shouldnt be there (related to gl in some way) or compiled Habitat in a different way.

My laptop contains an Intel onboard graphics and a dedicated Nvidia RTX A2000 GPU.  The container is meant to be used for my students who do not have a dedicated GPU.

this is how my docker-compose file looks like:
```

version: '3'
services:
  simulation:
    image: habitat
    build:
      context: .
      dockerfile: Dockerfile
    stdin_open: true
    tty: true
    env_file:
      - .env
    environment:
      - "DISPLAY=${DISPLAY}"
      - "XAUTHORITY=${XAUTHORITY}"
    volumes:
      - /tmp/.X11-unix:/tmp/.X11-unix:rw
      - ~/.Xauthority:/root/.Xauthority:rw
    network_mode: "host"
    devices:
      - "/dev/dri:/dev/dri"

```

This is how the head of the dockerfile looks like:

```
FROM ubuntu:20.04

# Set TimeZone
ENV TZ=America/Los_Angeles
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone
ARG DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get install -y \
    build-essential \
    apt-utils \
    git \
    curl \
    vim \
    ca-certificates \
    libjpeg-dev \
    libpng-dev \
    libglfw3-dev \
    libglm-dev \
    # libx11-dev \
    libomp-dev \
    x11-apps \
    libegl1-mesa-dev \
    pkg-config \
    wget \
    tzdata \
    zip \
    tk-dev \
    python3-tk \
    python3-pip \
    unzip \
    libxrandr-dev \ 
    libxinerama-dev \ 
    libxcursor-dev \
    libasound2 \
[... 112 lines cut ...]
[06:54:19:303567]:[Metadata] AssetAttributesManager.cpp(123)::createObject : Asset attributes (uvSphereWireframe:uvSphereWireframe_rings_16_segments_32) created and registered.
[06:54:19:303576]:[Metadata] AssetAttributesManager.cpp(112)::AssetAttributesManager : Built default primitive asset templates : 12
[06:54:19:303772]:[Metadata] SceneDatasetAttributesManager.cpp(37)::createObject : File (default) not found, so new default dataset attributes created  and registered.
[06:54:19:303776]:[Metadata] MetadataMediator.cpp(120)::createSceneDataset : Dataset default successfully created.
[06:54:19:303786]:[Metadata] AttributesManagerBase.h(380)::createFromJsonOrDefaultInternal : <Physics Manager>: Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json| This file does not exist.
[06:54:19:303795]:[Metadata] PhysicsAttributesManager.cpp(26)::createObject : File (./data/default.physics_config.json) not found, so new default physics manager attributes created and registered.
[06:54:19:303800]:[Metadata] MetadataMediator.cpp(203)::setActiveSceneDatasetName : Previous active dataset  changed to default successfully.
[06:54:19:303824]:[Metadata] AttributesManagerBase.h(380)::createFromJsonOrDefaultInternal : <Physics Manager>: Proposing JSON name : data/default.physics_config.json from original name : data/default.physics_config.json| This file does not exist.
[06:54:19:303831]:[Metadata] PhysicsAttributesManager.cpp(26)::createObject : File (data/default.physics_config.json) not found, so new default physics manager attributes created and registered.
[06:54:19:303837]:[Metadata] MetadataMediator.cpp(66)::setSimulatorConfiguration : Set new simulator config for scene/stage : data/scene_datasets/habitat-test-scenes/skokloster-castle.glb and dataset : default which is currently active dataset.
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=0): EGL_EXT_device_drm
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=1): EGL_EXT_device_drm
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=2): EGL_MESA_device_software
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
Platform::WindowlessEglApplication::tryCreateContext(): unable to find EGL device for CUDA device 0
WindowlessContext: Unable to create windowless context
 ⚡ root  examples  fffa5437 v0.2.3 ?  ✘  

```

I read some introductions to GL, Mesa etc.  (e.g. https://dzone.com/articles/hardware-accelerated-opengl-rendering-in-a-linux-container) but I still dont fully get why it isnt working, particularly when I specified to use X-server on DISPLAY 2 (which is intel graphics)

Thanks for any help : - )

0mdc, 2023-02-20:
Hey @uahic,

The logs might provide more information on the issue when running with the environment variable `MAGNUM_LOG=verbose`. Hopefully this will give some pointers.

>I had habitat once running on my machine using Docker

Could test an older version of Habitat? This would rule out whether the issue comes from a regression or a new environment issue.

uahic, 2023-02-24:
Hi @0mdc sure! I will try that out today or tomorrow and come back.