facebookresearch/habitat-sim #2005: Stop print the step up log

GuoPingPan, 2023-02-13:
I want  to know how to stop the output log as below?
```
Using optional features:
    GL_ARB_vertex_array_object
    GL_ARB_ES2_compatibility
    GL_ARB_separate_shader_objects
    GL_ARB_robustness
    GL_ARB_texture_storage
    GL_ARB_invalidate_subdata
    GL_ARB_texture_storage_multisample
    GL_ARB_multi_bind
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_texture_filter_anisotropic
    GL_KHR_debug
    GL_KHR_parallel_shader_compile
Using driver workarounds:
    no-forward-compatible-core-context
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
```
I find some tips is to add `os.environ['HABITAT_SIM_LOG'] = "quiet"` into the code, it actually work to ignore some log, but not completely.

I want to know all the setting about the logging about habitat.

aclegg3, 2023-02-13:
Those look like Magnum logging statements. As described [here](https://aihabitat.org/docs/habitat-sim/logging.html), you may also try `export MAGNUM_LOG=quiet`.

GuoPingPan, 2023-02-14:
That is a great method, all logging  kept slient!

closed 2023-02-14 by GuoPingPan