facebookresearch/habitat-sim #2006: Adding custom dataset

YESAndy, 2023-02-13:
Hi,

Is it possible to add custom datasets? I looked through the existing datasets which contain scene data (.glb file) and navigation data (navmesh file). I also looked through the habitat-sim docs that do not have a function for importing custom datasets. Many thanks!

aclegg3, 2023-02-13:
Hey @YESAndy,

You absolutely can add custom scene asset datasets for use in Habitat-sim.
All you really need to do is write the JSON config files which reference all of your assets so the simulator can find them.
See the [SceneDataset JSON documentation](https://aihabitat.org/docs/habitat-sim/attributesJSON.html) for details on this and [ReplicaCAD](https://aihabitat.org/datasets/replica_cad/) for an example of a complete SceneDataset with synthetic assets.

closed 2023-02-13 by aclegg3

magistri, 2023-07-27:
Hi!
Is there a specific format for importing a custom dataset with semantic classes? I tried to render the semantic of my dataset but while the geometry of the scene is rendered in the right way the semantic sensor gives me a completely black image. I tried to replicate the structure from the matterport dataset but no luck