facebookresearch/habitat-sim #2008: Habitat-Sim crashes when initialised with rgb sensor

YYDreamzure, 2023-02-15:
## Habitat-Sim version
v0.2.3

I am following the Google Colab scripts on Jupyter Notebook.

However, the jupyter kernel dies everytime the rgb sensors are mounted onto the agent before initialising the simulator. However, when I **run it as a python script in the terminal, there is no error.**

_rgb_sensor_spec = habitat_sim.CameraSensorSpec()
rgb_sensor_spec.uuid = "color_sensor"
rgb_sensor_spec.sensor_type = habitat_sim.SensorType.COLOR
rgb_sensor_spec.resolution = [sim_settings["height"], sim_settings["width"]]
rgb_sensor_spec.position = [0.0, sim_settings["sensor_height"], 0.0]
    
agent_config.sensor_specifications = [rgb_sensor_spec]

cfg = habitat_sim.Configuration(backend_sim_cfg, [agent_config])

sim = habitat_sim.Simulator(cfg)_ **<----------------crashes here**

However when I tried to add the sensor at a later stage after the initialisation of simulator, habitat-sim does not allow me to do it, the code I used: 

_agent_config.sensor_specifications = [rgb_sensor_spec]
sim.add_sensor(rgb_sensor_spec)
rgb_sensor = sim.get_agent(0)._sensors["color_sensor"]_

citing "**Data for SensorType.COLOR sensor was not loaded during Simulator init. Cannot dynamically add a SensorType.COLOR sensor unless one already exists.**"

Please advice on how to ensure the script does not crash the jupyter kernel so that I can view the scene in jupyter notebook.

Thank you!
## ❓ Questions and Help

aclegg3, 2023-03-01:
Hey, @YYDreamzure.

>when I tried to add the sensor at a later stage after the initialisation of simulator, habitat-sim does not allow me to do 

The issue with the is that when you initialize Habitat-sim without an RGB sensor, there are efficiency shortcuts (e.g. no loading of texture images or materials) which make the simulator instance incompatible with color rendering later.

> sim = habitat_sim.Simulator(cfg) <----------------crashes here
Can you provide any more information? A stack trace, error log, etc.. This may be a GL config issue in the Notebook environment. :thinking:

aclegg3, 2023-03-01:
@0mdc FYI

IbrahimMoussa123, 2023-03-03:
I am facing the same problem, the google colab session always crashes whenever it creates simulator instances sim = habitat_sim.Simulator(cfg)

This is the Run time logs Table: 

|Timestamp|Level|Message|
|---|---|---|
|Mar 3, 2023, 11:10:55 PM|WARNING|WARNING:root:kernel 25189fe2-8815-4097-968f-752a9f7cb034 restarted|
|Mar 3, 2023, 11:10:55 PM|INFO|KernelRestarter: restarting kernel \(1/5\), keep random ports|
|Mar 3, 2023, 11:10:52 PM|WARNING|WindowlessContext: Unable to create windowless context|
|Mar 3, 2023, 11:10:52 PM|WARNING|Platform::WindowlessEglApplication::tryCreateContext\(\): no EGL devices found, likely a driver issue; enable --magnum-gpu-validation to see additional info|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:638198\]:\[Metadata\] MetadataMediator\.cpp\(66\)::setSimulatorConfiguration : Set new simulator config for scene/stage : /content/data/scene\_datasets/habitat-test-scenes/apartment\_1\.glb and dataset : default which is currently active dataset\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:638183\]:\[Metadata\] MetadataMediator\.cpp\(175\)::setCurrPhysicsAttributesHandle : Old physics manager attributes  changed to \./data/default\.physics\_config\.json successfully\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:638170\]:\[Metadata\] MetadataMediator\.cpp\(203\)::setActiveSceneDatasetName : Previous active dataset  changed to default successfully\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:638152\]:\[Metadata\] PhysicsAttributesManager\.cpp\(26\)::createObject : JSON Configuration File \(\./data/default\.physics\_config\.json\) based physics manager attributes created and registered\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:638041\]:\[Metadata\] AttributesManagerBase\.h\(380\)::createFromJsonOrDefaultInternal : \<Physics Manager\>: Proposing JSON name : \./data/default\.physics\_config\.json from original name : \./data/default\.physics\_config\.json&#124; This file exists\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:638020\]:\[Metadata\] MetadataMediator\.cpp\(120\)::createSceneDataset : Dataset default successfully created\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:638001\]:\[Metadata\] SceneDatasetAttributesManager\.cpp\(37\)::createObject : File \(default\) not found, so new default dataset attributes created  and registered\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:637528\]:\[Metadata\] AssetAttributesManager\.cpp\(112\)::AssetAttributesManager : Built default primitive asset templates : 12|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:637505\]:\[Metadata\] AssetAttributesManager\.cpp\(123\)::createObject : Asset attributes \(uvSphereWireframe:uvSphereWireframe\_rings\_16\_segments\_32\) created and registered\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:637471\]:\[Metadata\] AssetAttributesManager\.cpp\(123\)::createObject : Asset attributes \(uvSphereSolid:uvSphereSolid\_rings\_8\_segments\_16\_useTexCoords\_false\_useTangents\_false\) created and registered\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:637437\]:\[Metadata\] AssetAttributesManager\.cpp\(123\)::createObject : Asset attributes \(icosphereWireframe:icosphereWireframe\_subdivs\_1\) created and registered\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:637406\]:\[Metadata\] AssetAttributesManager\.cpp\(123\)::createObject : Asset attributes \(icosphereSolid:icosphereSolid\_subdivs\_1\) created and registered\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:637376\]:\[Metadata\] AssetAttributesManager\.cpp\(123\)::createObject : Asset attributes \(cylinderWireframe:cylinderWireframe\_rings\_1\_segments\_32\_halfLen\_1\) created and registered\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:637338\]:\[Metadata\] AssetAttributesManager\.cpp\(123\)::createObject : Asset attributes \(cylinderSolid:cylinderSolid\_rings\_1\_segments\_12\_halfLen\_1\_useTexCoords\_false\_useTangents\_false\_capEnds\_true\) created and registered\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:637296\]:\[Metadata\] AssetAttributesManager\.cpp\(123\)::createObject : Asset attributes \(cubeWireframe:cubeWireframe\) created and registered\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:637263\]:\[Metadata\] AssetAttributesManager\.cpp\(123\)::createObject : Asset attributes \(cubeSolid:cubeSolid\) created and registered\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:637233\]:\[Metadata\] AssetAttributesManager\.cpp\(123\)::createObject : Asset attributes \(coneWireframe:coneWireframe\_segments\_32\_halfLen\_1\.25\) created and registered\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:637198\]:\[Metadata\] AssetAttributesManager\.cpp\(123\)::createObject : Asset attributes \(coneSolid:coneSolid\_segments\_12\_halfLen\_1\.25\_rings\_1\_useTexCoords\_false\_useTangents\_false\_capEnd\_true\) created and registered\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:637153\]:\[Metadata\] AssetAttributesManager\.cpp\(123\)::createObject : Asset attributes \(capsule3DWireframe:capsule3DWireframe\_hemiRings\_8\_cylRings\_1\_segments\_16\_halfLen\_1\) created and registered\.|
|Mar 3, 2023, 11:10:52 PM|WARNING|\[21:10:52:637079\]:\[Metadata\] AssetAttributesManager\.cpp\(123\)::createObject : Asset attributes \(capsule3DSolid:capsule3DSolid\_hemiRings\_4\_cylRings\_1\_segments\_12\_halfLen\_0\.75\_useTexCoords\_false\_useTangents\_false\) created and registered\.|
[... cut ...]
|Mar 3, 2023, 8:51:17 PM|WARNING|    	/root/\.jupyter/jupyter\_notebook\_config\.json|
|Mar 3, 2023, 8:51:17 PM|WARNING|    	/root/\.local/etc/jupyter/jupyter\_notebook\_config\.json|
|Mar 3, 2023, 8:51:17 PM|WARNING|    	/usr/etc/jupyter/jupyter\_notebook\_config\.json|
|Mar 3, 2023, 8:51:17 PM|WARNING|    	/usr/local/etc/jupyter/jupyter\_notebook\_config\.json|
|Mar 3, 2023, 8:51:17 PM|WARNING|    	/usr/local/etc/jupyter/jupyter\_notebook\_config\.d/panel-client-jupyter\.json|
|Mar 3, 2023, 8:51:17 PM|WARNING|    	/etc/jupyter/jupyter\_notebook\_config\.json|

aclegg3, 2024-01-16:
Note we have stopped supporting colab and converted tutorials to local Jupyter notebooks instead. The logs above seem to point to EGL issues on Colab. If similar issues appear elsewhere, please open a new issue.

closed 2024-01-16 by aclegg3