facebookresearch/habitat-sim #2025: Replica-CAD QuickStart Collab Not Working

ykaganov, 2023-03-09:
Hi,

I ran your original quickstart demo for replica-cad from here: https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/main/examples/tutorials/colabs/ReplicaCAD_quickstart.ipynb#scrollTo=6AS9PPsjfB4z inside **google collab** (on google cloud) and got the following error when executing first installation cell:

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2732  100  2732    0     0   8645      0 --:--:-- --:--:-- --:--:--  8645
--2023-03-09 14:44:54--  https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh
Resolving repo.continuum.io (repo.continuum.io)... 104.18.201.79, 104.18.200.79, 2606:4700::6812:c84f, ...
Connecting to repo.continuum.io (repo.continuum.io)|104.18.201.79|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh [following]
--2023-03-09 14:44:54--  https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh
Resolving repo.anaconda.com (repo.anaconda.com)... 104.16.130.3, 104.16.131.3, 2606:4700::6810:8203, ...
Connecting to repo.anaconda.com (repo.anaconda.com)|104.16.130.3|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 74403966 (71M) [application/x-sh]
Saving to: ‘Miniconda3-latest-Linux-x86_64.sh’

Miniconda3-latest-L 100%[===================>]  70.96M   359MB/s    in 0.2s    

2023-03-09 14:44:54 (359 MB/s) - ‘Miniconda3-latest-Linux-x86_64.sh’ saved [74403966/74403966]

PREFIX=/usr/local
Unpacking payload ...
                                                                                         
Installing base environment...

Downloading and Extracting Packages

Downloading and Extracting Packages

Preparing transaction: done
Executing transaction: done
installation finished.
WARNING:
    You currently have a PYTHONPATH environment variable set. This may cause
    unexpected behavior when running the Python interpreter in Miniconda3.
    For best results, please verify that your PYTHONPATH only points to
    directories of packages that are compatible with the Python interpreter
    in Miniconda3: /usr/local
Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed
Solving environment: / 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abortfailed

UnsatisfiableError: The following specifications were found to be incompatible with a past
explicit spec that is not an explicit spec in this operation (python):

  - cffi=1.15.1 -> pycparser -> python[version='2.7.*|2.7.*|>=3.4|>=3.6,<3.7.0a0|>=2.7,<2.8.0a0|3.6.*|3.5.*|3.4.*|>=3.6|>=3.5,<3.6.0a0|3.11.*|3.9.*|3.8.*|3.10.*|3.7.*']
  - cffi=1.15.1 -> python[version='3.8.16|3.9.16|>=3.10,<3.11.0a0|>=3.11,<3.12.0a0|>=3.9,<3.10.0a0|>=3.8,<3.9.0a0|>=3.7,<3.8.0a0',build=0_73_pypy]
  - habitat-sim -> attrs[version='>=19.1.0'] -> python[version='2.7.*|3.5.*|3.6.*|>=2.7,!=3.0.*,!=3.1.*,!=3.2.*,!=3.3.*,!=3.4.*|>=3.5|>=3.10,<3.11.0a0|>=3.9,<3.10.0a0|>=3.6,<3.7.0a0|>=2.7,<2.8.0a0|>=3.7|>=3.4|>=3|3.4.*|>=3.5,<3.6.0a0|>=3.11,<3.12.0a0|3.8.*|>=2.7|3.7.*']
  - habitat-sim -> imageio-ffmpeg -> setuptools
  - habitat-sim -> python[version='>=3.7,<3.8.0a0|>=3.8,<3.9.0a0']
  - numba=0.56.4 -> numpy[version='>=1.21.6,<2.0a0'] -> python[version='2.7.*|3.10.*|>=3.11,<3.12.0a0|>=3.7|>=3.8|>=3.6,<3.7.0a0|>=2.7,<2.8.0a0|>=3.5,<3.6.0a0|3.6.*|3.5.*|3.4.*|3.9.*|>=3.6|>=3|3.8.*']
  - numba=0.56.4 -> python[version='>=3.10,<3.11.0a0|>=3.9,<3.10.0a0|>=3.8,<3.9.0a0|>=3.7,<3.8.0a0']
  - numba=0.56.4 -> setuptools
  - numpy=1.22.4 -> python[version='>=3.10,<3.11.0a0|>=3.9,<3.10.0a0|>=3.8,<3.9.0a0']
  - numpy=1.22.4 -> python_abi=3.9[build=*_cp39] -> python[version='3.10.*|3.9.*|3.8.*']
  - python=3.9 -> pip -> setuptools
  - python=3.9 -> pip -> wheel
  - scipy=1.10.1 -> numpy[version='>=1.20.3,<1.27'] -> python[version='3.10.*|>=2.7,<2.8.0a0|>=3.5|>=3.6|>=3.7|>=3.7,<3.8.0a0|>=3.6,<3.7.0a0|>=3.5,<3.6.0a0|3.8.*|3.11.*|3.9.*']
  - scipy=1.10.1 -> python[version='>=3.10,<3.11.0a0|>=3.8,<3.9.0a0|>=3.9,<3.10.0a0|>=3.11,<3.12.0a0']

The following specifications were found to be incompatible with each other:

[... 227 lines cut ...]
numpy=1.22.4 -> libblas[version='>=3.8.0,<4.0a0'] -> openblas[version='0.3.5.*|>=0.3.6,<0.3.7.0a0']
habitat-sim -> numpy[version='>=1.13'] -> openblas[version='0.2.19|0.2.19.*|0.2.20|0.2.20.*|>=0.2.20,<0.2.21.0a0|>=0.3.3,<0.3.4.0a0']

Package libgfortran5 conflicts for:
habitat-sim -> numpy[version='>=1.13'] -> libgfortran5[version='>=10.3.0|>=10.4.0|>=11.2.0|>=11.3.0|>=9.4.0|>=9.3.0']
numpy=1.22.4 -> libblas[version='>=3.8.0,<4.0a0'] -> libgfortran5[version='>=9.3.0']
scipy=1.10.1 -> libblas[version='>=3.9.0,<4.0a0'] -> libgfortran5[version='10.3.0|10.3.0|10.3.0|10.3.0|10.3.0|10.3.0|10.3.0|10.3.0|10.3.0|10.3.0|10.3.0|10.3.0|10.3.0|10.4.0|10.4.0|10.4.0|10.4.0|11.1.0|11.1.0|11.1.0|11.1.0|11.1.0|11.2.0|11.2.0|11.2.0|11.2.0|11.2.0|11.2.0|11.2.0|11.2.0|11.2.0|11.3.0|11.3.0|11.3.0|11.3.0|12.1.0|12.1.0|12.2.0|12.2.0|>=9.3.0|9.5.0|9.5.0|9.5.0|9.5.0|9.4.0|9.4.0|9.4.0|9.4.0|9.4.0|9.4.0|9.4.0|9.4.0|9.4.0|9.4.0|9.4.0|9.4.0|9.4.0|9.3.0.*|8.5.0|8.5.0|8.5.0|8.5.0|8.5.0|8.5.0|8.5.0|8.5.0|8.5.0|8.5.0|8.5.0|8.5.0|8.4.0.*|11.2.0.*',build='h0ffbd86_8|h0ffbd86_9|h0ffbd86_12|h0ffbd86_13|h0ffbd86_14|h62347ff_5|h62347ff_6|h62347ff_8|h62347ff_12|h62347ff_15|h6e911d1_16|hab08dfb_18|hab08dfb_19|hb56cab1_4|hb56cab1_5|hb56cab1_6|hb56cab1_8|hb56cab1_9|hb56cab1_13|hb56cab1_15|hb56cab1_16|he3294f5_16|h6c583b3_5|h6c583b3_7|h6c583b3_8|h5c6108e_8|h5c6108e_9|h5c6108e_12|h5c6108e_13|h5c6108e_15|h39d6296_19|h337968e_19|h337968e_18|hdcd56e2_17|hdcd56e2_16|h39d6296_18|h6a973e8_17|h6a973e8_16|h5c6108e_16|h5c6108e_14|h5c6108e_11|h5c6108e_10|h6c583b3_6|h6c583b3_4|hfbd5096_19|hfbd5096_18|he3294f5_17|hb56cab1_14|hb56cab1_12|hb56cab1_11|hb56cab1_10|hb56cab1_7|h6e911d1_17|h62347ff_16|h62347ff_14|h62347ff_13|h62347ff_11|h62347ff_10|h62347ff_9|h62347ff_7|h62347ff_4|h42c683c_19|h42c683c_18|h0ffbd86_17|h0ffbd86_16|h0ffbd86_15|h0ffbd86_11|h0ffbd86_10']
scipy=1.10.1 -> libgfortran5[version='>=11.3.0']

Package habitat-sim-mutex conflicts for:
withbullet -> habitat-sim-mutex==1.0[build=*_bullet]
headless -> habitat-sim-mutex==1.0[build=headless_*]
habitat-sim -> habitat-sim-mutex==1.0[build='display_bullet|headless_nobullet|display_nobullet|headless_bullet']The following specifications were found to be incompatible with your system:

  - feature:/linux-64::__glibc==2.31=0
  - feature:|@/linux-64::__glibc==2.31=0
  - cffi=1.15.1 -> libgcc-ng[version='>=11.2.0'] -> __glibc[version='>=2.17']
  - numba=0.56.4 -> libgcc-ng[version='>=11.2.0'] -> __glibc[version='>=2.17']
  - openssl -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - python=3.9 -> libgcc-ng[version='>=10.3.0'] -> __glibc[version='>=2.17']
  - scipy=1.10.1 -> libgfortran-ng -> __glibc[version='>=2.17']

Your installed version is: 2.31

**An error occured during the installation of Habitat-sim or Habitat-Lab.**

aclegg3, 2023-03-22:
Hey @ykaganov,

Thanks for reporting this issue. The solution from #1818 doesn't seem to work in this case. Perhaps some recent changes have resulted in additional issues.
@Skylion007 thoughts on this?

cdor8, 2023-04-04:
Are there any updates on this? I've tried other Habitat Colabs as well and got a similar issue.

aclegg3, 2024-01-16:
Note we have stopped supporting colab and converted tutorials to local Jupyter notebooks instead.

closed 2024-01-16 by aclegg3