facebookresearch/habitat-sim #2027: `Datatool create_navmesh` does not work

JayeFu, 2023-03-09:
## Habitat-Sim version
v0.2.3

## 🐛 Bug

Under the root directory of `habitat-sim`, I run `./build/utils/datatool/Datatool create_navmesh ./abandoned.glb ./abandoned.navmesh`, the error shown is that
```text
ESP_CHECK failed: esp::logging::LoggingContext: No current logging context.
Aborted (core dumped)
```

## Steps to Reproduce

Steps to reproduce the behavior:

```bash
$ ./build/utils/datatool/Datatool create_navmesh ./abandoned.glb ./abandoned.navmesh
ESP_CHECK failed: esp::logging::LoggingContext: No current logging context.
Aborted (core dumped)
```

The `abandoned.glb` is upload in the attachment ([abandoned.zip](https://github.com/facebookresearch/habitat-sim/files/10933851/abandoned.zip)) and is publicly available here [[Sketchfab](https://sketchfab.com/3d-models/abandoned-room-1b0b9dcfffd04e50ad107409d9b59642)]

## Expected behavior

It creates a `navmesh` file without reporting the error.

## System Info
The script is run in a docker container.

```
ENVIRONMENT INFO:
Platform: Linux-5.4.0-132-generic-x86_64-with-debian-bullseye-sid
Machine: x86_64
Processor: x86_64
Libc version: glibc 2.17
Mac version: 
Python version: 3.7.16
Architecture: 64bit 
Win version:    
System OS: Linux
Release: 5.4.0-132-generic
Version: #148-Ubuntu SMP Mon Oct 17 16:02:06 UTC 2022
Operational System: linux
GCC version: b'gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0\n'
CMAKE version: b'cmake version 3.14.0\n'
NVIDIA-SMI: b'Thu Mar  9 16:57:20 2023       \n+-----------------------------------------------------------------------------+\n| NVIDIA-SMI 515.86.01    Driver Version: 515.86.01    CUDA Version: 11.7     |\n|-------------------------------+----------------------+----------------------+\n| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |\n| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |\n|                               |                      |               MIG M. |\n|===============================+======================+======================|\n|   0  NVIDIA A100-SXM...  On   | 00000000:0B:00.0 Off |                    0 |\n| N/A   33C    P0    55W / 400W |      2MiB / 40960MiB |      0%      Default |\n|                               |                      |             Disabled |\n+-------------------------------+----------------------+----------------------+\n'
Pip packages versions:
b'imageio-ffmpeg==0.4.8\nmatplotlib==3.5.3\nnumba==0.56.4\nnumpy-quaternion==2022.4.3\ntqdm==4.65.0\n'
Conda packages versions:
b'attrs                     22.2.0                   pypi_0    pypi  19.1.0\ngitpython                 3.1.31                   pypi_0    pypi\nimageio                   2.26.0                   pypi_0    pypi\nimageio-ffmpeg            0.4.8                    pypi_0    pypi\nmatplotlib                3.5.3                    pypi_0    pypi\nnumba                     0.56.4                   pypi_0    pypi\nnumpy                     1.21.6                   pypi_0    pypi\nnumpy-quaternion          2022.4.3                 pypi_0    pypi\npillow                    9.4.0                    pypi_0    pypi\nscipy                     1.7.3                    pypi_0    pypi  1.3.0\ntqdm                      4.65.0                   pypi_0    pypi\n'
```

JayeFu, 2023-03-10:
The command used to install `Habitat-Sim` is
```
$ git clone --branch stable https://github.com/facebookresearch/habitat-sim.git
$ cd habitat-sim
$ pip install -r requirements.txt; python setup.py install --headless --bullet --with-cuda --build-datatool
```

JayeFu, 2023-03-10:
Found a workaround.

Add a line at in `main` function in **Datatool.cpp** [[Link](https://github.com/facebookresearch/habitat-sim/blob/v0.2.3/src/utils/datatool/Datatool.cpp#L157)].
```
esp::logging::LoggingContext current_context(Corrade::Containers::StringView("quiet"));
```
Then `currentLoggingContext` in Logging.cpp [[Link](https://github.com/facebookresearch/habitat-sim/blob/v0.2.3/src/esp/core/Logging.cpp#L87)] is initialized.

But I would assume there is a more elegant solution.

aclegg3, 2023-03-31:
Thanks for identifying a workaround @JayeFu :+1: 
Please feel free to submit a PR with the fix and we can review or discuss alternatives there.
Admittedly this part of the code is not CI tested and rarely run, so we rely on reports and fixes from users to keep it maintained.