facebookresearch/habitat-sim #2042: Object contact_test() Segmentation Fault when colliding with non-floor object in Replica

denfed, 2023-03-21:
Hello!

I am having an issue with the replica dataset and some custom code I have written to insert human objects into the scene. Piecing together some of the code from the ECCV 2020 tutorial, I got to the point where I have an agent at some random point in the scene, and I place a human object in front of the agent. I am currently trying to make a robust way of placing that object on the floor, or on top of whatever object it is over currently (so there isn't any clipping). Looking at tutorials, I can use an Object's contact_test() function, and it works well when that inserted object is clipping through the floor (i.e. contact_test returns true when it is clipping, and False when it is not. I can use this to iteratively lower an object until it starts clipping, which would result in the object being on the floor). 

However, I noticed the simulator running into a segmentation fault when running the contact_test when the object is clipping with a _non-floor_ object, such as a table. Here is a picture of an inserted object, that is clipping, and contact_test is returned true: 

[image]

And here is an example where contact_test() results in a segmentation fault with the simulator:

[image]

I can provide the code that I created to do all this if needed, but maybe someone has a fast answer as to why this is happening. 

Any help with this would be great!

aclegg3, 2023-03-22:
Hey @denfed,

This is a known issue with Replica and Bullet phyaics integration. You are correct that the geometry density is the root cause here.

I believe the best immediate solution is to use decimated meshes for collision instead of the render mesh. For example, you could use the [MeshOptimizer ](https://meshoptimizer.org/) project to generate a reduced-geometry mesh. The resulting mesh can be configured for use in Habitat by referencing it in the [stage_config.json](https://aihabitat.org/docs/habitat-sim/attributesJSON.html#stageattributes) file as the `"collision_asset"`.

Regarding your application of finding the correct placement for the object once collisions are working, you should check out the [snap_down](https://github.com/facebookresearch/habitat-lab/blob/35f92bf5492da99e4aa90e71c0c80a8db338c865/habitat-lab/habitat/sims/habitat_simulator/sim_utilities.py#L231) utility in Habitat-lab. Essentially this uses the object's bounding box and a raycast against the scene geometry to determine the best placement height for an object to rest on the support surface below it.

denfed, 2023-03-22:
Oh wow, that snap_down function is perfect! I wrote some code to iteratively lower an object until it collides, but this seems to be a better way of doing it. Thank you!

aclegg3, 2023-03-31:
Glad that helped. Let me know if you have success with the decimation. I don't think we'll have cycles for this in the near term.
Note: related to #613