facebookresearch/habitat-sim #2046: problem: deleting objects

Ukias, 2023-03-27:
In the Habitat simulation, I want to delete objects. It is not possible with the rigid object manager. Link to repository with problem: https://github.com/Ukias/habitat_example.

jturner65, 2023-03-27:
Hi!
What is the error you are getting? Can you copy the last few console lines and attach them in a text file?

I am running your repo locally and wish to make sure I see what you see.

jturner65, 2023-03-27:
Actually I found the problem. You need to use the internal handle of the object to delete the object.  'banana' is not the handle that references this object, '011_banana_:0000' is (look in `addYcbObjOnTbl` method, you get 'obj' after you create an object based on the first entry in the list of object templates with the substring 'banana')

If you add print statements in your code similar to what I have added here : 
```

    def addYcbObjOnTbl(self, tmplHName, transl):
        # tmplHName: str
        # transl: mn.Vector3
        obj_mgr = self.sim.get_object_template_manager()
        print(f"Attempt to load : {tmplHName}")
        obj = self.sim.get_rigid_object_manager().add_object_by_template_handle(
            obj_mgr.get_template_handles(tmplHName)[0]
        )
        print(f"Built object named : {obj.handle}")
```

you will see output that looks something like this : 

```
Attempt to load : banana
[18:20:42:430363]:[Assets] ResourceManager.cpp(866)::loadRenderAsset : Loading general asset named: /home/john/Facebook/habitat_example/data/hab_ycb_v1.2/configs/../meshes/011_banana/google_16k/textured.glb
[18:20:42:430372]:[Assets] ResourceManager.cpp(1666)::configureImporterManagerGLExtensions : Importing Basis files as BC7.
[18:20:42:479844]:[Assets] ResourceManager.cpp(2232)::getMaterialShaderType : Shadertype being used for file : textured.glb | shadertype name : material
[18:20:42:479872]:[Assets] ResourceManager.cpp(2105)::loadMaterials : Building 1 materials for asset named 'textured.glb' : 
[18:20:42:479908]:[Assets] ResourceManager.cpp(2210)::loadMaterials : Idx 00:PBR.
Built object named : 011_banana_:0000
```

In the final printout, you see the object's actual handle.

jturner65, 2023-03-27:
Removing the object using the correct handle ...: 

```
self.sim.get_rigid_object_manager().remove_object_by_handle(
                "011_banana_:0000"
            )
```
Yields before and after pictures for me as follows (before and after) : 

[image]
[image]

Ukias, 2023-03-28:
Thanks, this solved it for me

closed 2023-03-28 by jturner65