facebookresearch/habitat-sim #2048: The depth information in the depth map is non-linear

oskar0812, 2023-03-28:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

When I generated the depth map, I found that there are some stripes on the left wall.This is the performance of depth discontinuity or non-linearity.
Is there any way to solve this problem? Make the depth map smooth and continuous
[image]

oskar0812, 2023-03-28:
I still have the problem that the output camera Intrinsic parameters and extrinsic parameters cannot match the picture.And i  used Replica Dataset.I want to use this data to do some novel viewpoint synthesis tasks.
I think I need some help, such as how to render the Replica image while outputting its camera intrinsic and extrinsic parameters.I would be grateful if someone could provide some help.

aclegg3, 2023-03-31:
@0mdc thoughts on this?