facebookresearch/habitat-sim #2049: ImportError: cannot import name 'text' from 'magnum'

YuquanDeng, 2023-03-28:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
I setup the conda environment by running `conda create -n habitat python=3.8` and install habitat-sim from night build by running `conda install habitat-sim=0.2.2 withbullet -c conda-forge -c aihabitat-nightly`. As I followed the solution from #1909 , I also pull the stable tag. But when I run `python examples/viewer.py --scene ./data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`, I still met the error:
<pre><code>
Traceback (most recent call last):
  File "examples/viewer.py", line 19, in <module>
    from magnum import shaders, text
ImportError: cannot import name 'text' from 'magnum' 
</code></pre>

- habitat-sim version and current branch of the repo
[image]
[image]

- If I run the viewer with habitat-sim in c++, it works fine.
[image]

___

I don't know what exactly goes wrong. Thank you so much in advance!

KyleHuo, 2023-03-29:
I met the same problem with you. I also use python 3.8 instead. In addition, I cannot run with C++ either. Hope someone can help.

jemo-mh, 2023-03-29:
Same issue, can't solve yet. 

Currently available  C++  test `habitat-viewer /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`

I tried 0.2.2 and 0.2.3 but both encountered same issue.
Env setting is UIbuntu 20.04, nvidia-470 graphic driver with RTX 2080Ti

aclegg3, 2023-03-30:
Hey folks,

To run the viewer application, you'll need a matching and recent version of the conda packages and cloned local repo code.

I tested today on a fresh conda env following build instructions:
```
conda create -n habitat python=3.9 cmake=3.14.0
conda activate habitat
conda install habitat-sim withbullet -c conda-forge -c aihabitat-nightly
git clone https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim
```
Note that I installed the nightly build.
Then (assuming you already have `data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`):
```
python examples/viewer.py --scene data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
habitat-viewer data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
```
Both of these work for me.

jemo-mh, 2023-03-31:
> Hey folks,
>
> To run the viewer application, you'll need a matching and recent version of the conda packages and cloned local repo code.
>
> I tested today on a fresh conda env following build [...]

HI there!
Thanks for your help.
As follow ur instructions (also same with documentation says)
c++ code for 
`habitat-viewer data/scene_datasets/habitat-test-scenes/skokloster-castle.glb` works well

but using python example, 
`python examples/viewer.py --scene data/scene_datasets/habitat-test-scenes/skokloster-castle.glb` print out RenderTarget Error which is 
`RenderTarget::Impl::blitRgbaTo(): target framebuffer has a size of Vector(1600, 1200) but expected Vector(800, 600)
Aborted (core dumped)`

I think a defined buffer size is different. Is there any solution to fix as setting configuration?
Thanks for your help in advance.

Have a nice day :)

aclegg3, 2023-03-31:
Hey @jemo-mh, this sounds like a different issue.
It reminds me of system DPI scaling issues we've seen in the past.
@0mdc @mosra thoughts here?

0mdc, 2023-04-01:
I was able to reproduce locally. I issued a fix here: #2055

@jemo-mh Can you try out these changes?

YuquanDeng, 2023-04-02:
> 

Sorry for the late reply! Thank you so much for your help!!! That worked!

YuquanDeng, 2023-04-02:
> Hey folks,
>
> To run the viewer application, you'll need a matching and recent version of the conda packages and cloned local repo code.
>
> I tested today on a fresh conda env following build [...]

Sorry for the late reply! Thank you so much for your help!!! That worked for me!

jemo-mh, 2023-04-03:
> I was able to reproduce locally. I issued a fix here: #2055
> 
> @jemo-mh Can you try out these changes?

Hello there!
Cloning your recent commit, It works now.
Thanks for your help and have a nice day :)

closed 2023-04-03 by 0mdc

chenhaomingbob, 2024-06-15:
> > I was able to reproduce locally. I issued a fix here: #2055
> > @jemo-mh Can you try out these changes?
> 
> Hello there! Cloning your recent commit, It works now. Thanks for your help and have a nice day :) 

Hi. I encountered the ImportError in tag v0.3.1. 
Can you remember the version you used that is successfully running?

0mdc, 2024-06-16:
That version works for me. What's your environment?

You might want to try re-creating your conda environment before debugging this further.

chenhaomingbob, 2024-06-17:
> That version works for me. What's your environment?
> 
> You might want to try re-creating your conda environment before debugging this further.

I find that this error is related to the `headless`.
If I use `conda install habitat-sim withbullet -c conda-forge -c aihabitat`  instead of `conda install habitat-sim withbullet headless -c conda-forge -c aihabitat`, the error does not exist.

0mdc, 2024-06-17:
I see. In the case of `viewer.py`, this makes sense as `headless` is meant to be used on systems without displays. Therefore, it excludes dependencies like GUI text.

Did you get this error by running something else?

chenhaomingbob, 2024-06-17:
Thanks for your explanation. When I run `python examples/viewer.py --scene data`, I encounter this error.

2927803072, 2025-04-01:
> Hey folks,
>
> To run the viewer application, you'll need a matching and recent version of the conda packages and cloned local repo code.
>
> I tested today on a fresh conda env following build [...]
Hello，I followed your procedure and this error occurred

Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to physical DPI
EGL: Failed to get EGL display: Success
Platform::GlfwApplication::tryCreate():  cannot create a window with core OpenGL context, falling back to compatibility context
EGL: Failed to get EGL display: Success
Platform::GlfwApplication::tryCreate(): cannot create a window with OpenGL context

I did this on a remote server, and then I tried to add headless when installing habitat, but there was this problem

from magnum import shaders, text
ImportError: cannot import name 'text' from 'magnum'

Do you know the reason for this? Hope to get your reply, thank you

2927803072, 2025-04-01:
> Thanks for your explanation. When I run , I encounter this error.`python examples/viewer.py --scene data`

@chenhaomingbob  Hello, I met the same problem, do you solve it?

PR #2055 (2023-04-03): Fix viewer.py framebuffer size mismatch.
## Motivation and Context

Upon initializing the python viewer, the framebuffer size may differ from the application config size due to environment factors.
This change ensures that the appropriate size is used for rendering.

Fixes: #2049

## How Has This Been Tested

Tested locally.

## Types of changes

- [ ] Docs change / refactoring / dependency upgrade
- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionalit [... cut]