facebookresearch/habitat-sim #2063: ModuleNotFoundError: No module named 'habitat_sim.robots'

YYDreamzure, 2023-04-06:
## Habitat-Sim version
[v0.2.4](https://github.com/facebookresearch/habitat-sim/releases/tag/v0.2.4)

I have used Habitat-sim before with no problem. However as I reinstalled habitat-sim and when I ran my old script, the problem (title) happened.

Below is my code:
if display:
    from habitat.utils.visualizations import maps

Errors:
ModuleNotFoundError Traceback (most recent call last)
Cell In[4], line 52
     50 # import the maps module alone for topdown mapping
     51 if display:
---> 52     from habitat.utils.visualizations import maps

File ~/sound/habitat-lab/habitat/__init__.py:9
      7 from habitat.config import Config, get_config
      8 from habitat.core.agent import Agent
----> 9 from habitat.core.benchmark import Benchmark
     10 from habitat.core.challenge import Challenge
     11 from habitat.core.dataset import Dataset

File ~/sound/habitat-lab/habitat/core/benchmark.py:18
     16 from habitat.config.default import get_config
     17 from habitat.core.agent import Agent
---> 18 from habitat.core.env import Env
     21 class Benchmark:
     22     r"""Benchmark for evaluating agents in environments."""

File ~/sound/habitat-lab/habitat/core/env.py:22
     20 from habitat.datasets import make_dataset
     21 from habitat.sims import make_sim
---> 22 from habitat.tasks import make_task
     23 from habitat.utils import profiling_wrapper
     26 class Env:

File ~/sound/habitat-lab/habitat/tasks/__init__.py:7
      1 #!/usr/bin/env python3
      2 
      3 # Copyright (c) Facebook, Inc. and its affiliates.
      4 # This source code is licensed under the MIT license found in the
      5 # LICENSE file in the root directory of this source tree.
----> 7 from habitat.tasks.registration import make_task  # noqa: F401

File ~/sound/habitat-lab/habitat/tasks/registration.py:28
     26 _try_register_nav_task()
     27 _try_register_vln_task()
---> 28 _try_register_rearrange_task()

File ~/sound/habitat-lab/habitat/tasks/rearrange/__init__.py:13, in _try_register_rearrange_task()
     12 def _try_register_rearrange_task():
---> 13     import habitat.tasks.rearrange.actions
     14     import habitat.tasks.rearrange.grip_actions
     15     import habitat.tasks.rearrange.multi_task.composite_sensors

File ~/sound/habitat-lab/habitat/tasks/rearrange/actions.py:21
     16 from habitat.sims.habitat_simulator.actions import HabitatSimActions
     18 # flake8: noqa
     19 # These actions need to be imported since there is a Python evaluation
     20 # statement which dynamically creates the desired grip controller.
---> 21 from habitat.tasks.rearrange.grip_actions import (
     22     GripSimulatorTaskAction,
     23     MagicGraspAction,
     24     SuctionGraspAction,
     25 )
     26 from habitat.tasks.rearrange.rearrange_sim import RearrangeSim
     27 from habitat.tasks.rearrange.utils import rearrange_collision, rearrange_logger

File ~/sound/habitat-lab/habitat/tasks/rearrange/grip_actions.py:14
     12 from habitat.core.embodied_task import SimulatorTaskAction
     13 from habitat.core.registry import registry
---> 14 from habitat.tasks.rearrange.rearrange_sim import RearrangeSim
     15 from habitat.tasks.rearrange.utils import (
     16     coll_link_name_matches,
     17     coll_name_matches,
     18 )
     21 class GripSimulatorTaskAction(SimulatorTaskAction):

File ~/sound/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py:36
     33 from habitat_sim.physics import JointMotorSettings, MotionType
     35 # flake8: noqa
---> 36 from habitat_sim.robots import FetchRobot, FetchRobotNoWheels
     37 from habitat_sim.sim import SimulatorBackend
     40 @registry.register_simulator(name="RearrangeSim-v0")
     41 class RearrangeSim(HabitatSim):

ModuleNotFoundError: No module named 'habitat_sim.robots'

Please advice how should I deal with this. Thanks!

## ❓ Questions and Help

aclegg3, 2023-04-06:
Hey @YYDreamzure 
The robots module has been moved out of habitat-sim into habitat-lab. The same functionality is available there plus some new features. You'll need to refactor your code to pull that module in from [the new source](https://github.com/facebookresearch/habitat-lab/tree/main/habitat-lab/habitat/articulated_agents/robots).

closed 2023-04-06 by aclegg3

YYDreamzure, 2023-04-07:
Hi thank you for the solution on habitat-lab. However, I have more issues now. 

I am unable to run the following in Jupyter notebook after specifying RGB configuration:
**habitat_sim.Simulator(cfg)**
Only the simplest code without any agent specification is able to run in Jupyter notebook without dead kernel.

The error given is:
libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
Platform::WindowlessGlxContext: cannot create core context, falling back to compatibility context: GLXBadFBConfig
Platform::WindowlessGlxContext: cannot create context: BadValue (integer parameter out of range for operation)
WindowlessContext: Unable to create windowless context

I am able to run the code in the terminal as a .py file.
I wonder if this is an anaconda issue (I have tried [this](https://stackoverflow.com/questions/72110384/libgl-error-mesa-loader-failed-to-open-iris) or a habitat-sim one.