facebookresearch/habitat-sim #2072: SoundSpaces 2.0

denfed, 2023-04-13:
Hello! I am having an issue using SS2.0 with materials enabled in HM3D environments. For some reason, when I run the audio sensor observation, I get these errors:

[13:07:34:776730]:[Assets] ResourceManager.cpp(2941)::joinSemanticHierarchy : Could not get the GenericSemanticMeshData
File: arvr/libraries/audio/AudioSDK/Research/Source/Wrapper/PropagationWrapper.cpp
Function: ovrResult PropagationWrapper::AddMeshVertices(const float *, ovra::Size), Line 926
Mesh vertices were NULL
[13:07:34:777420]:[Sensor] AudioSensor.cpp(394)::loadSemanticMesh : Couldn't add vertices to audio mesh, error:  2001
I am assuming this is something to do with the configurations of the hm3d dataset. However, I saw a previous issue where it was mentioned that materials seems to work fine for hm3d, along with mp3d and gibson.

The hm3d scene I am trying to run is :
hm3d/minival/00802-wcojb4TFT35/wcojb4TFT35.basis.glb
With the scene dataset config:
hm3d/minival/hm3d_annotated_minival_basis.scene_dataset_config.json

Any help would be greatly appreciated.

iyeon915, 2023-04-21:
Hi @denfed, 
Did you solve this issue? I got the same issue when I ran SS2.0 with HM3D environments.

denfed, 2023-04-21:
I have not solved it yet, hoping others can chime in and help.

denfed, 2023-06-21:
Has any progress been made on this issue? Thank you!

Sean1572, 2025-01-22:
@denfed I encountered the same issue and managed to figure out something funny that may just work. Some disclaimers, the error is gone but I haven't done unit tests and I'm using a custom 3d model that is formatted like HM3d (with the same kinds of glb and txt's). 

The issue is the audio propagation engine relies on vertices metadata for its material computation but HM3d only uses textures and doesn't encoded metadata about the vertices. Hence the error that there are no vertices, because the only semantic information habitat-sim gives the audio-prop engine is a uv texture. 

 I noticed that the `scene_dataset.json` had a `has_semantic_textures boolean` flag and set to true for HM3d like datasets (but false for other datasets). When I changed it to false I got a diffrent error saying the mesh didn't have vertex colors for the semantic annotation in `*.semantic.txt`.  So from the sounds of it, the semantic mesh's vertices is at least being looked at now!

So in blender I loaded up the `*.semantic.glb` and set the texture color to the vertices in the image (for complex textures you could bake it, I just made every vertex the same color for each object manually), exported the newly updated glb file (making sure that vertex colors were being exported with everything else in the glb) and gave it another try. 

No more error and it seemed to work fine. IRs contains some reasonable values when before it was basically 0. 

I have `backend_cfg.load_semantic_mesh = True` and `acoustics_config.enableMaterials = True`. I didn't make changes to the `*.semantic.txt` so it should just work assuming the color matches correctly (key word assuming).