facebookresearch/habitat-sim #2074: Python 3.9 on RHEL 7 python setup.py install --user  --headless fails with returned non-zero exit status 1.

RobbieTheK, 2023-04-17:
Running `python setup.py install --user  --headless` on RHEL 7 with with Python 3.9, fails with the following:
```
[124/868] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/LWO/LWOMaterial.cpp.o
/path/to/me/habitat-sim/src/deps/assimp/code/LWO/LWOMaterial.cpp: In member function ‘bool Assimp::LWOImporter::HandleTextures(aiMaterial*, const TextureList&, aiTextureType)’:
/path/to/me/habitat-sim/src/deps/assimp/code/LWO/LWOMaterial.cpp:141:9: warning: ‘mapping’ may be used uninitialized in this function [-Wmaybe-uninitialized]
         if (mapping != aiTextureMapping_UV) {
         ^
[240/868] Building CXX object _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathRandom.cpp.o
FAILED: _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathRandom.cpp.o 
/usr/bin/c++  -I/path/to/me/habitat-sim/build/_deps/imath-build/src/Imath -I/path/to/me/habitat-sim/build/_deps/imath-src/src/Imath -I/path/to/me/habitat-sim/src/deps -I/path/to/me/habitat-sim/src/cmake/../deps/tinyxml2 -I/path/to/me/habitat-sim/build/_deps/imath-build/config -isystem /path/to/me/habitat-sim/src/cmake/../deps/eigen -isystem /path/to/me/habitat-sim/src/cmake/../deps/rapidjson/include -DVERSION_INFO=\"0.2.4\" -fvisibility=hidden -DBASISD_SUPPORT_BC7_MODE6_OPAQUE_ONLY=0 -DBASISD_SUPPORT_ATC=0 -DBASISD_SUPPORT_FXT1=0 -DBASISD_SUPPORT_PVRTC2=0 -O2 -g -DNDEBUG -fPIC -fvisibility=hidden -fvisibility-inlines-hidden -std=c++1y -MD -MT _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathRandom.cpp.o -MF _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathRandom.cpp.o.d -o _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathRandom.cpp.o -c /path/to/me/habitat-sim/build/_deps/imath-src/src/Imath/ImathRandom.cpp
In file included from /path/to/me/habitat-sim/build/_deps/imath-src/src/Imath/ImathExport.h:9:0,
                 from /path/to/me/habitat-sim/build/_deps/imath-src/src/Imath/ImathRandom.h:21,
                 from /path/to/me/habitat-sim/build/_deps/imath-src/src/Imath/ImathRandom.cpp:13:
/path/to/me/habitat-sim/build/_deps/imath-build/config/ImathConfig.h:131:22: error: missing binary operator before token "("
 #  if __has_attribute(__type_visibility__)
                      ^
[248/868] Building CXX object _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathColorAlgo.cpp.o
FAILED: _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathColorAlgo.cpp.o 
/usr/bin/c++  -I/path/to/me/habitat-sim/build/_deps/imath-build/src/Imath -I/path/to/me/habitat-sim/build/_deps/imath-src/src/Imath -I/path/to/me/habitat-sim/src/deps -I/path/to/me/habitat-sim/src/cmake/../deps/tinyxml2 -I/path/to/me/habitat-sim/build/_deps/imath-build/config -isystem /path/to/me/habitat-sim/src/cmake/../deps/eigen -isystem /path/to/me/habitat-sim/src/cmake/../deps/rapidjson/include -DVERSION_INFO=\"0.2.4\" -fvisibility=hidden -DBASISD_SUPPORT_BC7_MODE6_OPAQUE_ONLY=0 -DBASISD_SUPPORT_ATC=0 -DBASISD_SUPPORT_FXT1=0 -DBASISD_SUPPORT_PVRTC2=0 -O2 -g -DNDEBUG -fPIC -fvisibility=hidden -fvisibility-inlines-hidden -std=c++1y -MD -MT _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathColorAlgo.cpp.o -MF _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathColorAlgo.cpp.o.d -o _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathColorAlgo.cpp.o -c /path/to/me/habitat-sim/build/_deps/imath-src/src/Imath/ImathColorAlgo.cpp
In file included from /path/to/me/habitat-sim/build/_deps/imath-src/src/Imath/ImathNamespace.h:49:0,
                 from /path/to/me/habitat-sim/build/_deps/imath-src/src/Imath/ImathColorAlgo.h:13,
                 from /path/to/me/habitat-sim/build/_deps/imath-src/src/Imath/ImathColorAlgo.cpp:12:
/path/to/me/habitat-sim/build/_deps/imath-build/config/ImathConfig.h:131:22: error: missing binary operator before token "("
 #  if __has_attribute(__type_visibility__)
                      ^
[265/868] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/Importer/StepFile/StepFileGen1.cpp.o
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/path/to/me/habitat-sim/setup.py", line 459, in <module>
    setup(
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/__init__.py", line 108, in setup
    return distutils.core.setup(**attrs)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 185, in setup
    return run_commands(dist)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
    dist.run_commands()
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
    self.run_command(cmd)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/dist.py", line 1221, in run_command
    super().run_command(command)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/command/install.py", line 74, in run
    self.do_egg_install()
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/command/install.py", line 123, in do_egg_install
    self.run_command('bdist_egg')
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/dist.py", line 1221, in run_command
    super().run_command(command)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/command/bdist_egg.py", line 164, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/command/bdist_egg.py", line 150, in call_command
    self.run_command(cmdname)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/dist.py", line 1221, in run_command
    super().run_command(command)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/command/install_lib.py", line 111, in build
    self.run_command('build_ext')
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/dist.py", line 1221, in run_command
    super().run_command(command)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/path/to/me/habitat-sim/setup.py", line 271, in run
    self.build_extension(ext)
  File "/path/to/me/habitat-sim/setup.py", line 375, in build_extension
    subprocess.check_call(
  File "/path/to//anaconda3-2021.11/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/path/to/me/.local/lib/python3.9/site-packages/cmake/data/bin/cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--']' returned non-zero exit status 1.
```

Also this warning:
```
CMake Warning (dev) at /path/to/me/.local/lib/python3.9/site-packages/cmake/data/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (rt) does
  not match the name of the calling package (RT).  This can lead to problems
  in calling code that expects `find_package` result variables (e.g.,
  `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  deps/assimp/cmake-modules/FindRT.cmake:19 (find_package_handle_standard_args)
  deps/assimp/code/CMakeLists.txt:1013 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.
```

Any solution for this?

aclegg3, 2023-05-18:
Interesting. I'm not sure where to start digging here. Likely we need more system information. Could be a system package or library version issue somewhere.

RobbieTheK, 2023-05-18:
I tried again making sure to do the pip install -r requirements. Python 3.9.7, cmake 3.17.4. This is a RHEL 7 cluster where we can load various versions of software via modules.

```
check.warn(importable)
running build_ext
[3/605] Building CXX object _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathRandom.cpp.o
FAILED: _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathRandom.cpp.o 
/usr/bin/c++  -I/path/to/me/habitat-sim/build/_deps/imath-build/src/Imath -I/path/to/me/habitat-sim/build/_deps/imath-src/src/Imath -I/path/to/me/habitat-sim/src/deps -I/path/to/me/habitat-sim/src/cmake/../deps/tinyxml2 -I/path/to/me/habitat-sim/build/_deps/imath-build/config -isystem /path/to/me/habitat-sim/src/cmake/../deps/eigen -isystem /path/to/me/habitat-sim/src/cmake/../deps/rapidjson/include -DVERSION_INFO=\"0.2.4\" -fvisibility=hidden -DBASISD_SUPPORT_BC7_MODE6_OPAQUE_ONLY=0 -DBASISD_SUPPORT_ATC=0 -DBASISD_SUPPORT_FXT1=0 -DBASISD_SUPPORT_PVRTC2=0 -O2 -g -DNDEBUG -fPIC -fvisibility=hidden -fvisibility-inlines-hidden -std=c++1y -MD -MT _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathRandom.cpp.o -MF _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathRandom.cpp.o.d -o _deps/imath-build/src/Imath/CMakeFiles/Imath.dir/ImathRandom.cpp.o -c /path/to/me/habitat-sim/build/_deps/imath-src/src/Imath/ImathRandom.cpp
In file included from /path/to/me/habitat-sim/build/_deps/imath-src/src/Imath/ImathExport.h:9:0,
                 from /path/to/me/habitat-sim/build/_deps/imath-src/src/Imath/ImathRandom.h:21,
                 from /path/to/me/habitat-sim/build/_deps/imath-src/src/Imath/ImathRandom.cpp:13:
/path/to/me/habitat-sim/build/_deps/imath-build/config/ImathConfig.h:131:22: error: missing binary operator before token "("
 #  if __has_attribute(__type_visibility__)
                      ^
[4/605] Linking CXX static library deps/assimp/code/libassimp.a
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/path/to/me/habitat-sim/setup.py", line 459, in <module>
    setup(
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/__init__.py", line 108, in setup
    return distutils.core.setup(**attrs)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 185, in setup
    return run_commands(dist)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
    dist.run_commands()
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
    self.run_command(cmd)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/dist.py", line 1221, in run_command
    super().run_command(command)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/command/install.py", line 74, in run
    self.do_egg_install()
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/command/install.py", line 123, in do_egg_install
    self.run_command('bdist_egg')
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/dist.py", line 1221, in run_command
    super().run_command(command)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/command/bdist_egg.py", line 164, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/command/bdist_egg.py", line 150, in call_command
    self.run_command(cmdname)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/dist.py", line 1221, in run_command
    super().run_command(command)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/command/install_lib.py", line 111, in build
    self.run_command('build_ext')
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/dist.py", line 1221, in run_command
    super().run_command(command)
  File "/path/to/me/.local/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/path/to/me/habitat-sim/setup.py", line 271, in run
    self.build_extension(ext)
  File "/path/to/me/habitat-sim/setup.py", line 375, in build_extension
    subprocess.check_call(
  File "/rigel/opt/anaconda3-2021.11/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/path/to/me/.local/lib/python3.9/site-packages/cmake/data/bin/cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--']' returned non-zero exit status 1.
```

Moon-heart, 2023-06-27:
I also want to know how to solve this problem

chenjiahes, 2024-04-09:
(habitat) sz@sz-pr4908p:~/extra/JH_lzl/intro_package/habitat-sim$ python setup.py install
/home/sz/extra/JH_lzl/intro_package/habitat-sim/setup.py:297: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  assert StrictVersion(
/home/sz/extra/JH_lzl/intro_package/habitat-sim/setup.py:299: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  ) >= StrictVersion("3.6"), "Must use python3.6 or newer"
running install
/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/_distutils/cmd.py:66: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  self.initialize_options()
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
CMake Error: The current CMakeCache.txt directory /home/sz/extra/JH_lzl/intro_package/habitat-sim/build/CMakeCache.txt is different than the directory /home/aaai/extra/extra1/JH_lzl/intro-package/habitat-sim/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
CMake Error: The source "/home/sz/extra/JH_lzl/intro_package/habitat-sim/src/CMakeLists.txt" does not match the source "/home/aaai/extra/extra1/JH_lzl/intro-package/habitat-sim/src/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
Traceback (most recent call last):
  File "/home/sz/extra/JH_lzl/intro_package/habitat-sim/setup.py", line 313, in <module>
    setup(
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/__init__.py", line 104, in setup
    return distutils.core.setup(**attrs)
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 185, in setup
    return run_commands(dist)
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
    dist.run_commands()
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
    self.run_command(cmd)
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/dist.py", line 967, in run_command
    super().run_command(command)
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/command/install.py", line 87, in run
    self.do_egg_install()
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/command/install.py", line 139, in do_egg_install
    self.run_command('bdist_egg')
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/dist.py", line 967, in run_command
    super().run_command(command)
[... 6 lines cut ...]
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/dist.py", line 967, in run_command
    super().run_command(command)
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/_distutils/command/install_lib.py", line 111, in build
    self.run_command('build_ext')
  File "/home/sz/anaconda3/envs/habitat/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
[... cut ...]
subprocess.CalledProcessError: Command '['cmake', '-H/home/sz/extra/JH_lzl/intro_package/habitat-sim/src', '-Bbuild', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/sz/extra/JH_lzl/intro_package/habitat-sim/build/lib.linux-x86_64-cpython-39/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/home/sz/anaconda3/envs/habitat/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-DBUILD_GUI_VIEWERS=ON', '-DBUILD_TESTS=OFF']' returned non-zero exit status 1.

How to solve this when python setup.py install