facebookresearch/habitat-sim #2089: Pass 'Combine redundant instructions' is not initialized.

YuminosukeSato, 2023-04-30:
python examples/viewer.py --dataset data/replica_cad/replicaCAD.scene_dataset_config.json --scene apt_1
I did this .
```

[15:13:58:654358]:[Assets] ResourceManager.cpp(984)::loadStageInternal : Attempting to load stage data/replica_cad/configs/stages/../../stages/frl_apartment_stage.glb
[15:13:58:655350]:[Assets] ResourceManager.cpp(866)::loadRenderAsset : Loading general asset named: data/replica_cad/configs/stages/../../stages/frl_apartment_stage.glb
[15:13:58:655680]:[Assets] ResourceManager.cpp(1655)::configureImporterManagerGLExtensions : Importing Basis files as ASTC 4x4.
Pass 'Combine redundant instructions' is not initialized.
Verify if there is a pass dependency cycle.
Required Passes:
Segmentation fault (core dumped)
```
I got this error . How I do solve this error?

Skylion007, 2023-05-24:
@YuminosukeSato What hardware / OS version are you using?

YuminosukeSato, 2023-05-29:
I use windows 10.

blu666, 2023-06-30:
Hi, I'm running into the exact same issue. I'm running on Ubuntu 20.04 on a WSL2 (Windows 11). My GPU is RTX A2000 8GB Notebook

Running the example with skokloster-castle
`python examples/viewer.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`
is fine. Except when I click ESC to close the application, it always result in segmentation fault.

Any advice on how to resolve this would be greatly appreciated.

blu666, 2023-06-30:
@Skylion007 @jturner65