facebookresearch/habitat-sim #2093: Platform::GlfwApplication::tryCreate(): cannot create a window with core OpenGL context, falling back to compatibility context

YuminosukeSato, 2023-05-03:
Device Specifications
OS: Windows 10
WSL: ubuntu 18.04.6LTS
GPU: NVIDIA Geforce 3090, Driver Version: 4

Steps to reproduce the error
Followed the instructions detailed in the README

To test the interactive viewer I ran the commandpython examples/viewer.py --scene ./data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to physical DPI
GLX: Failed to create context: GLXBadFBConfig
Platform::GlfwApplication::tryCreate(): cannot create a window with core OpenGL context, falling back to compatibility context
GLX: Failed to create context: GLXBadFBConfig
Platform::GlfwApplication::tryCreate(): cannot create a window with OpenGL context
malloc_consolidate(): invalid chunk size
Aborted (core dumped)

aclegg3, 2023-05-18:
@0mdc thoughts?

0mdc, 2023-05-25:
@YuminosukeSato What version of Habitat-Sim are you using? I recall a similar issue that was fixed here: #2055

YuminosukeSato, 2023-05-29:
@0mdcI use git cloned habitat sim and conda install habitat-sim -c conda-forge -c aihaveat in last month.

0mdc, 2023-07-04:
@YuminosukeSato Did you manage to find a solution for this?

YuminosukeSato, 2023-07-05:
No.I think that wsl is problem.

0mdc, 2023-07-05:
It's possible. I am not set up to reproduce on WSL at the moment.

Could you try if this solution addresses the problem: https://stackoverflow.com/questions/66497147/cant-run-opengl-on-wsl2

wzjscut, 2023-12-25:
i have the same problem in Ubuntu20.04, RTX3090

guchi-ac, 2024-10-19:
I have the same problem，And I have made sure that I have configured X11 forwarding and openGL, and successfully run the ‘glxgears’ command, but I still don't know what the problem is. Has anyone configured it successfully on WSL2?