facebookresearch/habitat-sim #2095: Physics simulation is unstable and different from that of pybullet

rishiagarwal2000, 2023-05-03:
## Habitat-Sim version
main

## 🐛 Bug

I tried simulating a urdf from [here](https://github.com/Stanford-TML/circle_dataset/tree/main/src/viewer#:~:text=scene%20and%20the%20URDF%20skeletons) in habitat-sim and pybullet. The outputs for the simulation are attached. Apparently, the simulations differ significantly which seems unexpected to me because habitat-sim uses the same pybullet simulation internally. Also, the simulation is highly unstable. I am not sure if this is a bug, or if I am using it incorrectly. It would be great if you could help me with this.

## Steps to Reproduce

Pybullet snippet
```python
import pybullet as p
import time
import pybullet_data
from scipy.spatial.transform import Rotation as R
import sys
import imageio

physicsClient = p.connect(p.GUI)#or p.DIRECT for non-graphical version
p.setAdditionalSearchPath(pybullet_data.getDataPath()) #optionally
p.setGravity(0,0,-9.8)
planeId = p.loadURDF("plane.urdf")
startPos = [0,0,1.1]
startOrientation = p.getQuaternionFromEuler([0,0,0])

humanId = p.loadURDF("CIRCLE_assets/subjects/2.urdf", startPos, startOrientation) #R.from_euler('xyz', [0, 0, 0], degrees=True).as_quat())
jointIndices = [i for i in range(p.getNumJoints(humanId)) if p.getJointInfo(humanId, i)[2] == p.JOINT_SPHERICAL]
print([p.changeDynamics(humanId, i, linearDamping=0, angularDamping=0, jointDamping=0.0) for i in jointIndices])

print("starting sim loop")

for i in range (100):
    p.stepSimulation()
    time.sleep(1./240.)
    
p.disconnect()
```

Habitat-sim snippet
```python
import habitat_sim
from habitat_sim.logging import logger
from settings import default_sim_settings, make_cfg

import habitat_sim.physics as phy
from habitat_sim.utils.common import quat_from_magnum, quat_to_magnum
from scipy.spatial.transform import Rotation as R

import magnum as mn
import numpy as np
from typing import Any, Callable, Dict, List, Optional, Tuple
import pathlib
import json

import os
import imageio

def global_correction_quat(up_v: mn.Vector3, forward_v: mn.Vector3) -> mn.Quaternion:
    """
    Given the upward direction and the forward direction of a local space frame, this methd produces
    the correction quaternion to convert the frame to global space (+Y up, -Z forward).
    """
    if up_v.normalized() != mn.Vector3.y_axis():
        angle1 = mn.math.angle(up_v.normalized(), mn.Vector3.y_axis())
        axis1 = mn.math.cross(up_v.normalized(), mn.Vector3.y_axis())
        rotation1 = mn.Quaternion.rotation(angle1, axis1)
        forward_v = rotation1.transform_vector(forward_v)
    else:
        rotation1 = mn.Quaternion()

[... 374 lines cut ...]

Bullet simulation (expected)
[image]

Habitat sim (unstable and way different)
[image]

## Additional context

## System Info

You can run the script with:
```
# For security purposes, please check the contents of collect_env.py before running it.
python src_python/habitat_sim/utils/collect_env.py
```

 - OS (e.g., Linux): Linux
 - How you installed PyTorch (`conda`, `pip`, source): pip
 - Build command you used (if compiling from source): python setup.py develop --bullet
 - Python version: 3.9.16
 - GPU models and configuration: P8
 - CUDA version: 11.7
 - PIP/conda dependencies packages versions
 - Any other relevant information:

aclegg3, 2023-05-04:
@rishiagarwal2000 Thanks for the bug report.

First, it is important to note that internally Habitat connects to the Bullet C++ API, not PyBullet (which embeds much of its multi-body dynamics support directly into bindings layers and C Bullet integration). Therefore, the expectation of similar default behavior is somewhat unrealistic. 

I've looked into this and noticed a couple of things.
1. These examples appear to be running at very different frame-rates. Your call to `sim.step_world()` excludes a time-step so uses the default of 1/60. I assume pybullet steps at the physics timestep of 1/240, so this would explain the first inconsistency. I'm not sure what pybullet sim would look like if it continued simulation. :thinking: 
2. PyBullet adds joint motors by default (I'm not sure if this uses SPD by default) while Habitat only adds damping motors when friction is defined. You can remove any automatically created motors and add one full set of new motors in Habitat like: 
```
                # check removal and auto-creation
                joint_motor_settings = habitat_sim.physics.JointMotorSettings(
                    position_target=0.0,
                    position_gain=1.0,
                    velocity_target=0.0,
                    velocity_gain=1.0,
                    max_impulse=1000.0,
                )
                existing_motor_ids = ao.existing_joint_motor_ids
                for motor_id in existing_motor_ids:
                    ao.remove_joint_motor(motor_id)
                ao.create_all_motors(joint_motor_settings)
```
3. Habitat does not have SPD implemented for joint motors currently, instead using the Bullet C++ `btMultiBodyJointMotor` interface. I'm not completely sure how this difference will affect stability.
4. After playing with these skeletons a bit, I think a large contributing factor to the instability is that inertia matrices are computed from the collision shapes automatically. These are fairly small and spread out possibly contributing to the unstable behavior. For example, see the [AMASS test urdf](https://github.com/facebookresearch/habitat-sim/blob/main/data/test_assets/urdf/amass_male.urdf) included with Habitat. It has a very similar structure, but much larger collision shapes and much more stable behavior in-engine. 
If I manually force link inertia to be loaded from the file, I get a much more stable simulation. I could consider adding this option to the API.

Finally, I suggest you try out your simulations interactively in the python viewer. 

Make the following small edit to the examples/viewer.py file at line 546:
```
elif key == pressed.T:
    # load URDF
    fixed_base = alt_pressed
    urdf_file_path = "data/test_assets/urdf/amass_male.urdf" #or your own filepath
    
    aom = self.sim.get_articulated_object_manager()
    ao = aom.add_articulated_object_from_urdf(
        urdf_file_path, fixed_base, 1.0, 1.0, True
    )
    ao.translation = (
        self.default_agent.scene_node.transformation.transform_point(
            [0.0, 1.0, -1.5]))
    # check removal and auto-creation
    joint_motor_settings = habitat_sim.physics.JointMotorSettings(
        position_target=0.0,
        position_gain=1.0,
        velocity_target=0.0,
        velocity_gain=1.0,
        max_impulse=1000.0,
    )
    existing_motor_ids = ao.existing_joint_motor_ids
    for motor_id in existing_motor_ids:
        ao.remove_joint_motor(motor_id)
    ao.create_all_motors(joint_motor_settings)
```

Then run it like:
`python examples/viewer.py --stage-requires-lighting --scene data/test_assets/scenes/simple_room.glb `
Press `'m'` to enter GRAB mouse mode and then click and drag the skeleton to play around with it like so:

https://user-images.githubusercontent.com/1445143/236301486-70676f66-5754-4c14-8b86-8c1769c0e033.mp4

With your file (default) simulation is unstable :frowning_face: :

https://user-images.githubusercontent.com/1445143/236301559-2f9f074b-e80c-4c38-aa42-b23ac956b374.mp4

With your file inertia loaded from URDF it is very stiff, but stable:

https://user-images.githubusercontent.com/1445143/236301684-ccd60e89-a9c2-489a-ae36-b5e03f24ba18.mp4

With your file inertia loaded from URDF but also scaled to 0.001 (from 1) we get what we might expect:

https://user-images.githubusercontent.com/1445143/236301912-cd326ecb-d32b-4454-9a04-9ea83cf7ac4a.mp4

rishiagarwal2000, 2023-05-08:
Thank you so much @aclegg3. This is super helpful and clarifies a lot of questions. I appreciate your elaborate reply. I will try to incorporate these suggestions and if I face any further issues, I will post them here.

closed 2023-07-31 by aclegg3