facebookresearch/habitat-sim #2099: Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 1 EGL devices in total WindowlessContext: Unable to create windowless context

YuminosukeSato, 2023-05-08:
## ❓ Questions and Help
Hello,

I am trying to set up habitat-sim on a wsl remote Ubuntu-22.04 machine, but when I try to run the simulation I encounter the following error Any help on installation please:

````
eglQueryDevicesEXT(): EGL_BAD_ALLOC error: In eglQueryDevicesEXT: Failed to allocate device list.
eglQueryDevicesEXT(): EGL_BAD_ALLOC error: In eglQueryDevicesEXT: Failed to allocate device list.
Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=0): EGL_MESA_device_software EGL_EXT_device_drm_render_node
eglQueryDeviceAttribEXT(): eglQueryDeviceAttribEXT
Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 1 EGL devices in total
WindowlessContext: Unable to create windowless context
````
Ubuntu-22.04 with NVIDIA GPU3090. about via wsl in Windows 10

Steps to reproduce
Install habitat-sim:
conda env create -n habitat python=3.8 cmake=3.14.0
conda start habitat
conda install -n habitat habitat-sim -c conda-forge -c aihabitat
Download and run example: cloned the examples/example.py repo and downloaded the test data. Execute
python examples/example.py --scene . /test_data/scene_datasets/habitat-test-scenes/skokloster-castle.glb

After running nvidia-smi.
```
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 530.41.03 Driver Version: 531.41 CUDA Version: 12.1 |
|-----------------------------------------+----------------------+----------------------+
| GPU Name Persistence-M| Bus-Id Disp.A | Volatile Uncorr.
| Fan Temp Perf Pwr:Usage/Cap| Memory-Usage | GPU-Util Compute M. | | | MIG M. | GPU Name Persistence-M| Bus-Id Disp.A | Volatile Uncorr.
| MIG M.
| =========================================+======================+======================|
| 0 NVIDIA GeForce RTX 3090 On | 00000000:02:00.0 On | N/A |
| 0% 47C P8 26W / 370W| 774MiB / 24576MiB | 1% Default |
| | N/A |
+-----------------------------------------+----------------------+----------------------+

+---------------------------------------------------------------------------------------+
| Processes: |
| GPU GI CI PID Type Process name GPU Memory |
| ID ID Usage |
|=======================================================================================|
| No running processes found |
+---------------------------------------------------------------------------------------+
```

0mdc, 2023-07-04:
@YuminosukeSato 

Have you managed to work around this issue?

`eglQueryDeviceStringEXT(EGLDevice=0): EGL_MESA_device_software EGL_EXT_device_drm_render_node` hints that EGL is not finding your NVIDIA driver. This may be a problem with your `glvnd` installation.

friolero, 2023-07-19:
Not sure if it is the same case with yours. Mine turns out to miss nvidia gl packages from the nvidia cuda docker environment and installing it resolves my issue:

```
sudo apt-get install libnvidia-gl-535 (replace with your nvidia driver version)
sudo apt-get install --reinstall libglvnd-dev
```

0mdc, 2023-10-25:
I'll close this as stale. Feel free to file another issue.

closed 2023-10-25 by 0mdc

programingisgoddamndifficult, 2024-01-11:
> I'll close this as inactive. Feel free to file another issue.

> Not sure if it is the same case with yours. Mine turns out to miss nvidia gl packages from the nvidia cuda docker environment and installing it resolves my issue:
> 
> ```
> sudo apt-get install libnvidia-gl-535 (replace with your nvidia driver version)
> sudo apt-get install --reinstall libglvnd-dev
> ```

thanks a lot! it works

goodstudent9, 2024-02-19:
> thanks a lot! it works

E: Unable to locate package libnvidia-gl-535
That seems like I cannot install this package

EftePA, 2024-03-06:
I have the same problem, and used `nvidia-smi` to check my nvidia driver version, it is 535. But installing `libnvidia-gl-535` and `libglvnd-dev` didn't solve it.

goodstudent9, 2024-03-09:
> I have the same problem, and used `nvidia-smi` to check my nvidia driver version, it is 535. But installing `libnvidia-gl-535` and `libglvnd-dev` didn't solve it.
 DO NOT TRY to install habitat if you can not do that!
Using docker image provided. I use it smoothly!
[image]

Good luck!

betray12138, 2025-08-21:
Hi,  I meet the same problem here. My nvidia-smi information is:

Thu Aug 21 16:49:43 2025
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.163.01 Driver Version: 550.163.01 CUDA Version: 12.4 |
|-----------------------------------------+------------------------+----------------------+
| GPU Name Persistence-M | Bus-Id Disp.A | Volatile Uncorr. ECC |
| Fan Temp Perf Pwr:Usage/Cap | Memory-Usage | GPU-Util Compute M. |
| | | MIG M. |
|=========================================+========================+======================|
| 0 NVIDIA GeForce RTX 4090 On | 00000000:B8:00.0 Off | Off |
| 59% 27C P8 30W / 450W | 12MiB / 49140MiB | 0% Default |
| | | N/A |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes: |
| GPU GI CI PID Type Process name GPU Memory |
| ID ID Usage |
|=========================================================================================|
+-----------------------------------------------------------------------------------------+

[image]

I tried to install libnvidia-gl-550, reinstall libglvnd-dev and install libglib2.0-0 on top of the CUDA base image, but it didn't work. Any help would be greatly appreciated!!!

Hi @goodstudent9 , could you please tell me how to use the Docker image to run the habitat? Many thanks in advance!