facebookresearch/habitat-sim #2101: Simulation loading indefinitely in Jupyter Notebook

pfmephisto, 2023-05-10:
`habitat-sim == 0.2.4`
installed through conda

I am experiencing a bug when trying to initialise a simulation that contains sematic data in a Jupyter Notebook.
`jupyterhub==2.0.0`
I'm using the HM3D dataset and the issue is that the Jupyter cell will keep running indefinitely, without returning any output or error messages. The issue seems to appear when the `scene_dataset_config_file` is being supplied. If omit or a wrong file is used the simulation loads fine, just without the sematic mesh.

```python

import os 
import habitat_sim
import matplotlib.pyplot as plt

sim_settings = {
     "scene": '/hm3d_fullpath/minival/00808-y9hTuugGdiq/y9hTuugGdiq.basis.glb', 
    "scene_dataset": '/hm3d_fullpath/simple_scene_dataset_config.json',
    "default_agent": 0,
    "sensor_height": 1.5,  # Height of sensors in meters
    "width": 512,  # Spatial resolution of the observations
    "height": 512,
    "hfov": 90,
}

def make_simple_cfg(settings):    
    # simulator backend
    sim_cfg = habitat_sim.SimulatorConfiguration()
    sim_cfg.gpu_device_id = 0
    sim_cfg.create_renderer = True
    sim_cfg.enable_physics = False
    sim_cfg.scene_id = settings["scene"]
    sim_cfg.scene_dataset_config_file = settings["scene_dataset"]
    sim_cfg.load_semantic_mesh = True
    sim_cfg.requires_textures = True

    sensor_position = [0.0, settings["sensor_height"], 0.0]
    sensor_resolution = [settings["height"], settings["width"]]
    
    sensor_specifications = []
    for sensor_name, sensor_type in zip(
        [ "color_sensor", "depth_sensor", "semantic_sensor" ],
        [ habitat_sim.SensorType.COLOR, habitat_sim.SensorType.DEPTH, habitat_sim.SensorType.SEMANTIC ]
    ):
        sensor_spec  = habitat_sim.CameraSensorSpec()
        sensor_spec.uuid = sensor_name
        sensor_spec.hfov = settings["hfov"]
        sensor_spec.sensor_type = sensor_type
        sensor_spec.resolution = sensor_resolution
        sensor_spec.position = sensor_position
        sensor_spec.sensor_subtype = habitat_sim.SensorSubType.PINHOLE
        sensor_spec.noise_model = "None"
        
        sensor_specifications.append(sensor_spec)

    # agent
    agent_cfg = habitat_sim.agent.AgentConfiguration()
    agent_cfg.sensor_specifications = sensor_specifications

    rotation = 60.0
    agent_cfg.action_space = {
        "move_forward": habitat_sim.agent.ActionSpec(
            "move_forward", habitat_sim.agent.ActuationSpec(amount=0.25)
        ),
        "turn_left": habitat_sim.agent.ActionSpec(
            "turn_left", habitat_sim.agent.ActuationSpec(amount=rotation)
        ),
        "turn_right": habitat_sim.agent.ActionSpec(
            "turn_right", habitat_sim.agent.ActuationSpec(amount=rotation)
        ),
    }

[... 373 lines cut ...]
xorg-inputproto           2.3.2             h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.8.4                h0b41bf4_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxcursor           1.2.0                h0b41bf4_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h0b41bf4_2    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxinerama          1.1.5                h27087fc_0    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.10            h7f98852_1003    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-xextproto            7.3.0             h0b41bf4_1003    conda-forge
xorg-xf86vidmodeproto     2.3.1             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xvfbwrapper               0.2.9           pyhd8ed1ab_1005    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zipp                      3.15.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h166bdaf_4    conda-forge
zstd                      1.5.2                h3eb15da_6    conda-forge

frank-gallagher, 2024-02-13:
I am experiencing the same issue with `habitat-sim=0.3.0` installed with conda.

Did you ever find a solution?

pfmephisto, 2024-02-13:
It's a while back and I haven't been working with it since.
But if I recall correctly I gave up on the Jupyter notebook and used a standalone python script.