facebookresearch/habitat-sim #2127: "Could NOT find OpenGL (missing: OPENGL_opengl_LIBRARY OPENGL_glx_LIBRARY)" during installation on headless server

rosanom, 2023-06-09:
Hi, I'm trying to install a specific version of the simulator (v. 0.1.5), as requested by a [third-party project](https://github.com/devendrachaplot/Object-Goal-Navigation). I followed the steps suggested in the [build from source page](https://github.com/facebookresearch/habitat-sim/blob/main/BUILD_FROM_SOURCE.md) but when I run the `python setup.py install --headless` command I experience the following error: 
`Could NOT find OpenGL (missing: OPENGL_opengl_LIBRARY OPENGL_glx_LIBRARY)`.
The machine is a headless server with two nvidia GPUs, I manually installed the nvidia drivers launching a `.run` file. If I run the `echo $CUDA_VISIBLE_DEVICES` command the output is `0,1`, launching the `ldconfig -N -v | grep libEGL` command the output is:
```
/sbin/ldconfig.real: Can't stat /usr/local/lib/x86_64-linux-gnu: No such file or directory
/sbin/ldconfig.real: Path `/usr/lib/x86_64-linux-gnu' given more than once
(from /etc/ld.so.conf.d/x86_64-linux-gnu.conf:4 and /etc/ld.so.conf.d/x86_64-linux-gnu.conf:3)
/sbin/ldconfig.real: Path `/lib/x86_64-linux-gnu' given more than once
(from <builtin>:0 and /etc/ld.so.conf.d/x86_64-linux-gnu.conf:3)
/sbin/ldconfig.real: Path `/usr/lib/x86_64-linux-gnu' given more than once
(from <builtin>:0 and /etc/ld.so.conf.d/x86_64-linux-gnu.conf:3)
/sbin/ldconfig.real: Path `/usr/lib' given more than once
(from <builtin>:0 and <builtin>:0)
        libEGL_mesa.so.0 -> libEGL_mesa.so.0.0.0
        libEGL_nvidia.so.0 -> libEGL_nvidia.so.515.65.01
/sbin/ldconfig.real: /lib/x86_64-linux-gnu/ld-linux-x86-64.so.2 is the dynamic linker, ignoring

        libEGL.so.1 -> libEGL.so.1.1.0
```
With `locate libEGL_nvidia.so ` the output is:
```
/usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.0
/usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.515.65.01
/usr/lib32/libEGL_nvidia.so.0
/usr/lib32/libEGL_nvidia.so.515.65.01
```
I tried to add the path `/usr/lib/x86_64-linux-gnu` to the `LD_LIBRARY_PATH` variable but it had no effect.
Do you have any idea on what the problem could be?
Thank you!

0mdc, 2023-07-04:
@rosanom Did you manage to solve this issue?

circleTreeF, 2024-03-20:
Hello @rosanom @0mdc, had the same issue when building for headless and with audio. Did you eventually have any idea?

zhanglixuan0720, 2024-08-30:
I solve this problem by downgrading the cmake version to 3.14.4
```
pip uninstall cmake
pip install cmake==3.14.4
```