facebookresearch/habitat-sim #2134: Building from source failing on Ubuntu 22.04

suyashkumar2409, 2023-06-20:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
As part of the [Homerobot Challenge](https://aihabitat.org/challenge/2023_homerobot_ovmm/), I am trying to build habit-sim from source, but during the step

`python setup.py install --bullet`

the installation crashes my PC. It always occurs at a very specific step. around 43%. I have a video of the execution around this step.

suyashkumar2409, 2023-06-20:
https://github.com/facebookresearch/habitat-sim/assets/10911412/0e56a6b3-ae5f-4ee4-9373-29c124eb5f08

man469, 2023-07-26:
You may check the Common build issues part in the homepage of habitat:
On some virtual machines, this might result in running out of memory. You can serialize the build process via:
python setup.py build_ext --parallel 1 install
so I think you can try 
python setup.py --bullet --parallel 1 install