facebookresearch/habitat-sim #2147: Conda install Habitat-sim 0.2.3just hangs infinitely

AdriannaGmz, 2023-07-03:
Habitat-Sim version v0.2.3

Steps I'm doing:
```
conda create -n habitat python=3.8 cmake=3.14.0
conda activate habitat
conda install habitat-sim=0.2.3 withbullet -c conda-forge -c aihabitat
```
output from this:
```
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: \  
```

And just hangs in there..  Eventually the terminal window stops responding
could you help?

AdriannaGmz, 2023-07-03:
Changed it to
```
conda install -c aihabitat habitat-sim=0.2.3 withbullet
```
 and now outputs this:

```
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: | 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                            

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package habitat-sim-mutex conflicts for:
habitat-sim=0.2.3 -> habitat-sim-mutex==1.0[build='headless_nobullet|display_nobullet|display_bullet|headless_bullet']
withbullet -> habitat-sim-mutex==1.0[build=*_bullet]The following specifications were found to be incompatible with your system:

  - feature:/linux-64::__glibc==2.27=0
  - python=3.7 -> libgcc-ng[version='>=11.2.0'] -> __glibc[version='>=2.17']

Your installed version is: 2.27

```

can you please help?

0mdc, 2023-07-04:
Quick guess: `habitat-sim=0.2.3` requires `python 3.7`, while `habitat-sim=0.2.4` supports later versions.

Either of those should work:

```
conda create -n habitat python=3.9 cmake=3.14.0
conda activate habitat
conda install habitat-sim=0.2.4 withbullet -c conda-forge -c aihabitat
```
```
conda create -n habitat python=3.7 cmake=3.14.0
conda activate habitat
conda install habitat-sim=0.2.3 withbullet -c conda-forge -c aihabitat
```

taylorlulu, 2023-07-05:
I faced the same problem and I solved it by download the corresponding .tar.gz  file from https://anaconda.org/aihabitat/habitat-sim/files. I'm using python3.9 so I downloaded 
[image]
Then I use the command:
`conda activate habitat
conda install --use-local habitat-sim-0.2.4-py3.9_bullet_linux_f179b584bcd713c5a2a998132211e2cae881d6d1.tar.bz2
`
And it worked. But it still need to install some missing packages using the pip command.

blu666, 2023-07-07:
I tried @0mdc's proposed approach and did not work. I have conda environment with python3.9 and used 

> `conda create -n habitat python=3.9 cmake=3.14.0
conda activate habitat
conda install -c aihabitat habitat-sim=0.2.4 withbullet`
The process still hangs.

Using @taylorlulu's approach worked.

0mdc, 2023-07-07:
Awesome! Thanks @taylorlulu for the solution.

closed 2023-07-07 by 0mdc

francescotaioli, 2024-02-14:
Thanks @taylorlulu , great alternative when conda hangs up!