facebookresearch/habitat-sim #2157: Return Restarting kernel in Spyder IDE when run the code sim = habitat_sim.Simulator(cfg)

man469, 2023-07-14:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

when I run the code sim = habitat_sim.Simulator(cfg) in spyder IDE, it returns Restarting kernel... and stops running, how to cope with this problem?

aclegg3, 2023-08-11:
Hey @man469,

We can't really support IDE specific issues, but it sounds like the process is crashing casuing a kernel restart. Possibly it could be running out of process memory loading the scenes or there may be mis-configured system dependencies. Hard to know without more details.

flxinxout, 2023-10-23:
Hello, I have the same error: my kernel crashes when I call habitat_sim.Simulator(cfg).
I use jupyter notebook on a Ubuntu latest with 60GB of memory and 8GB of RAM

lemonci, 2024-10-22:
Hello, same problem when running habitat-sim 0.3.1 nightly via Jupyter notebook on Ubuntu 22.04 with 64G RAM. Here is the error from terminal:

[W 2024-10-22 16:14:07.246 ServerApp] 404 GET /api/kernels/4dc214dd-8f29-4f94-a3e6-43241d684b99/channels?session_id=3b52f31c-94e3-484a-9a5f-22f6bbab0747 (127.0.0.1): Kernel does not exist: 4dc214dd-8f29-4f94-a3e6-43241d684b99
[W 2024-10-22 16:14:07.249 ServerApp] 404 GET /api/kernels/4dc214dd-8f29-4f94-a3e6-43241d684b99/channels?session_id=3b52f31c-94e3-484a-9a5f-22f6bbab0747 (446d6d9c12104bfc8285bc80d188c382@127.0.0.1) 4.25ms referer=None
[W 2024-10-22 16:14:07.251 ServerApp] 404 GET /api/kernels/4dc214dd-8f29-4f94-a3e6-43241d684b99?1729628047249 (127.0.0.1): Kernel does not exist: 4dc214dd-8f29-4f94-a3e6-43241d684b99
[W 2024-10-22 16:14:07.251 ServerApp] wrote error: 'Kernel does not exist: 4dc214dd-8f29-4f94-a3e6-43241d684b99'
    Traceback (most recent call last):
      File "/home/ml/anaconda3/envs/habitat/lib/python3.9/site-packages/tornado/web.py", line 1790, in _execute
        result = await result
      File "/home/ml/anaconda3/envs/habitat/lib/python3.9/site-packages/jupyter_server/auth/decorator.py", line 73, in inner
        return await out
      File "/home/ml/anaconda3/envs/habitat/lib/python3.9/site-packages/jupyter_server/services/kernels/handlers.py", line 75, in get
        model = await ensure_async(km.kernel_model(kernel_id))
      File "/home/ml/anaconda3/envs/habitat/lib/python3.9/site-packages/jupyter_server/services/kernels/kernelmanager.py", line 501, in kernel_model
        self._check_kernel_id(kernel_id)
      File "/home/ml/anaconda3/envs/habitat/lib/python3.9/site-packages/jupyter_server/services/kernels/kernelmanager.py", line 532, in _check_kernel_id
        raise web.HTTPError(404, "Kernel does not exist: %s" % kernel_id)
    tornado.web.HTTPError: HTTP 404: Not Found (Kernel does not exist: 4dc214dd-8f29-4f94-a3e6-43241d684b99)
[W 2024-10-22 16:14:07.252 ServerApp] 404 GET /api/kernels/4dc214dd-8f29-4f94-a3e6-43241d684b99?1729628047249 (446d6d9c12104bfc8285bc80d188c382@127.0.0.1) 0.93ms referer=http://localhost:8888/notebooks/Habitat/content/habitat-sim/Navigation.ipynb

lemonci, 2024-10-22:
In my case, the problem was solved as shown here: https://github.com/facebookresearch/habitat-sim/issues/1941

OrdealToLive, 2025-06-16:
I encountered the same problem as you and have solved it. I found that when running the latest version (0.3.3) on a virtual machine without a graphics card, the kernel would crash when it reached the line `sim = habitat_sim.Simulator(cfg)`. However, the same code ran successfully on a machine with a graphics card (RTX 2080Ti). I hope this can help you.

lemonci, 2025-06-20:
> I encountered the same problem as you and have solved it. I found that when running the latest version (0.3.3) on a virtual machine without a graphics card, the kernel would crash when it reached [...]

Thanks. The problem was solved.