facebookresearch/habitat-sim #2171: error installing habitat-sim using micromamba on mac os 12.6.7

vmarin3, 2023-08-02:
## Habitat-Sim version
main

## 🐛 Bug

error installing habitat-sim using micromamba on mac os 12.6.7
this command works : `micromamba install withbullet  headless  -c aihabitat`
but this fails:  `micromamba install withbullet habitat-sim   -c aihabitat`

with error

> (habitat) lawn-143-215-101-245:~ vitalymarin$ micromamba install withbullet habitat-sim   -c aihabitat
> aihabitat/noarch                                              No change
> aihabitat/osx-arm64                                           No change
> 
> Pinned packages:
>   - python 3.9.*
> 
> error    libmamba Could not solve for environment specs
>     The following package could not be installed
>     └─ habitat-sim does not exist (perhaps a typo or a missing channel).
> critical libmamba Could not solve for environment specs

## Steps to Reproduce
1.  running Mac OS 12.6.7
2. `micromamba create -n habitat python=3.9 cmake -c conda-forge -c anaconda`
3. ` micromamba install withbullet habitat-sim  -c aihabitat `

 Please note that without a minimal working example to reproduce the bug, we may not be able to help you.

## Expected behavior

## Additional context

## System Info

Please copy and paste the output from the
[environment collection script](https://github.com/facebookresearch/habitat-sim/tree/main/habitat_sim/utils/collect_env.py)
(or fill out the checklist below manually).

You can run the script with:
```
# For security purposes, please check the contents of collect_env.py before running it.
python src_python/habitat_sim/utils/collect_env.py
```

 - OS (e.g., Linux): mac 12.6.7
 - How you installed PyTorch (`conda`, `pip`, source): not installed
 - Build command you used (if compiling from source): ` micromamba install withbullet habitat-sim  -c aihabitat `
 - Python version: 3.9
 - GPU models and configuration: none
 - CUDA version: none
 - PIP/conda dependencies packages versions: micromamba 1.4.9 (tried conda 23.7.2) as well
 - Any other relevant information: -

vmarin3, 2023-08-02:
installing from source works. but the conda and mamba causing errors.

herdrick, 2023-09-09:
There is no build in conda for arm64, i.e. Mac M1 and M2 machines.

aclegg3, 2023-09-11:
Given that we build conda packages on circleCI and that they are sun-setting intel mac machine images, I imagine we'll start building for M1 machines in the near future. Likely starting with the next version.